RobotLocomotion/drake #14601: MySystem::get_input_port(.).HasValue(context) segfault

edrumwri, 2021-01-30:
`MySystem::get_input_port(.).HasValue(const Context& context)` segfaults when MySystem has been built into a Diagram but `context` was created via `MySystem::AllocateContext(.)`.

The use case is virtually cloning the Context from within a System. Note that Context::Clone(.) works only when it is the root context, and I need to be able to "clone" the Context when it is part of a diagram context. 

The virtual in-system cloning strategy I am using appears like thus (cc @sherm1):
1. Allocate a Context for MySystem to "cloned_context"
2. Copy the time, state, and parameters for the current context ("context") to "cloned_context"
3. Fix input ports for "cloned_context" using evaluated input port values for "context" (_only when `MySystem::get_input_port(.).HasValue(context)` is true_)

This virtual cloning strategy works generally but segfaults when MySystem has been built into a Diagram,  `MySystem::get_input_port(.).HasValue(context)` is false, and `MySystem::get_input_port(.).HasValue(cloned_context)` is called. Offending line is here:

https://github.com/RobotLocomotion/drake/blob/ce36a0fba54442d425c8a800638e71355f2e68e4/systems/framework/diagram.cc#L958

To fix, I suggest putting in a line here (https://github.com/RobotLocomotion/drake/blob/ce36a0fba54442d425c8a800638e71355f2e68e4/systems/framework/system_base.cc#L192) that checks whether the Context is the root context, and if so, returning `nullptr`.

EricCousineau-TRI, 2021-02-01:
(As part of platform reviewer's checklist) Assigning @sherm1 for prioritization / work assignment

sherm1, 2021-02-22:
I think that's a good solution -- if the Context doesn't have a parent then the port can't be connected in that Context so should return nullptr as it would for any unconnected port. An easy fix but needs a unit test to verify functionality.

A workaround you probably already thought of: keep a disembodied copy of MySystem around for the purpose of evaluating just a local context.

edrumwri, 2021-02-22:
SGTM- I'll submit a fix when I have some free cycles.

sherm1, 2021-02-22:
cc @rpoyner-tri in case we can get to this before Evan (he says it might be months)

edrumwri, 2021-02-24:
I think I'll be able to get to this within the next week.

rpoyner-tri, 2021-05-03:
Looks like this got fixed, but github magic auto-close of the issue didn't quite work. Closing.

closed 2021-05-03 by rpoyner-tri