RobotLocomotion/drake #14616: sudden movement in an incorrect direction then  kinova arm moves to the desired location

mpatalberta, 2021-02-02:
A problem in arm movement for j2s7s300_sphere_collision.urdf.
You tell the arm to move to location the system goes somewhere else quickly then starts to move towards the desired direction.

sherm1, 2021-02-03:
Assigning to @sammy-tri for disposition.

mpatalberta, 2021-02-04:
The jerk of the endeffector from move_jaco only happens if you terminate move_jaco then move to the same point. My belief is the kinova_simulator is  not returning the proper current position.

jwnimmer-tri, 2021-02-04:
I am closing this issue for two reasons.

First, for similar reasons as #14586 -- this jaco example serves to illustrate the tools.  The example is not intended to be a generic building block for application developers to invoke directly.  It is possible that the problem you are experiencing lies beyond what the example is intended to encompass.

Second, the bug report is so incomplete as to be un-actionable.  There is no way to understand what you have tried, what happened, or how anyone could possibly help to fix it.

Here are some tutorials about writing good bug reports.  If you'd like the Drake community to be able to offer you better advice, you need to provide a better starting point for them to understand your problem:
- [Mozilla](https://developer.mozilla.org/en-US/docs/Mozilla/QA/Bug_writing_guidelines)
- [Ximian](https://homes.cs.washington.edu/~mernst/advice/ximian-bug-guidelines.html)
- [Tatham](https://www.chiark.greenend.org.uk/~sgtatham/bugs.html)
- [Raymond](http://www.catb.org/~esr/faqs/smart-questions.html)
- [Software Testing Help](https://www.softwaretestinghelp.com/how-to-write-good-bug-report/)

closed 2021-02-04 by jwnimmer-tri