RobotLocomotion/drake #14617: kinova arm - the default value for number of joints and number of finger is hardcoded

mpatalberta, 2021-02-02:
> Before you post:
drake/manipulation/kinova_jaco
  jaco_constants.h    

/// proper arguments to the system constructors.
constexpr int kJacoDefaultArmNumJoints = 7;
constexpr int kJacoDefaultArmNumFingers = 3;

This is problem for the case of arm that may only have 6 joints and 2 fingers,
Some of the memory allocations use this number directly..
The urdf files needs to be used and read in early and off of that information.

sherm1, 2021-02-03:
Assigning to @sammy-tri for disposition.

jwnimmer-tri, 2021-02-04:
I am closing this for the same reasons as #14586 -- this is a feature request we'll have to decline.  The example serves to illustrate one sample application.  Making it more configurable would only serve to obscure its tutorial value.  The example is not intended to be a generic building block for application developers to invoke directly.

closed 2021-02-04 by jwnimmer-tri