RobotLocomotion/drake #14638: mbp: Should ensure "world" is an implicitly named body/frame

EricCousineau-TRI, 2021-02-12:
Per https://github.com/RobotLocomotion/drake/pull/10055#issuecomment-777847377, it can be painful to realize that `"world"` may not explicitly exist in MBP, as it's used as an explicit interface element in specs like URDF and SDFormat (e.g. the parent body of a joint). (See parsing code, where it says stuff like `if name == "world": return plant.world_body()`).

At present, the world body/frame are accessible through things like `MultibodyPlant.world_frame()`, `MultibodyPlant.world_body()`, the named instance of `"WorldBody"`, and the relevant geometry info through `SceneGraph::world_frame_id()`.

\cc @naveenoid @amcastro-tri

amcastro-tri, 2021-02-15:
This problem could keep happening if we ever decide to support other file formats with other naming conventions.
IMO, it'd be best if the name for the "world" (as for any other body) matches whatever name that file format specifies.
Therefore a third option would be:
3. Allow users to specify a "default world name", so that it matches whatever parsing utility or code convention they are using.

And along with that, since we do mostly support URDF and SDF, make our Drake convention to match those. i.e. do the renaming WorldBody --> body and deprecate the former.

EricCousineau-TRI, 2021-02-15:
Sounds good to me.

I'm a bit leery of (3), TBH, but I've added it to the summary above.

PR #18158 (2022-10-25): multibody: Change world body name from "WorldBody" to "world"
More in line with conventions in URDF and SDFormat

Resolves #14638

Per #18149 and stability guidelines, shouldn't need deprecation?
(Though Anzu code, and probably other's, will need to change)

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18158)
closed 2022-10-25 by EricCousineau-TRI