RobotLocomotion/drake #14644: LCM Publish Dropout when there are active LCM subscribers

dmcconachie-tri, 2021-02-13:
When publishing LCM data via a periodic update in a simulator, publishing can get skipped if `LcmInterfaceSystem` has data to handle.

The problem appears to be the following sequence:
- The Lcm time triggered publisher adds a publish event to the upcoming event list at some future point
- The `LcmInterfaceSystem` happens to have data to handle
  - Inside `LcmInterfaceSystem::DoCalcNextUpdateTime`, the `next_timed_event_time` is set to the current time, but no event is added to the event list
  - This overrides the next update time as set by the Lcm publisher, clearing all events in the list
- `Simulator::AdvanceTo` gets told that there is an event immediately (`next_timed_event_time == context_.get_time()`, but there are no events, so `next_publish_time` and `next_update_time` are set to `inf`.
- `Simulator::IntegrateContinuousState` ignores `next_timed_event_time`, using only `next_publish_time` and `next_update_time`.
  - Time is advanced far enough that the trigger point of the Lcm publisher gets skipped before `CalcNextUpdateTime` is called again.

See also: https://drakedevelopers.slack.com/archives/C2ETTRXT7/p1613162307003600

An example can be seen in this branch: https://github.com/dmcconachie-tri/drake/tree/lcm-publishing-dropout-debugging. On this branch, run:

```
bazel run //systems/analysis/dev:lcm_bug_example 
```

dmcconachie-tri, 2021-02-13:
A triage of this problem appears to be to added a dummy event to the event list in `LcmInterfaceSystem::DoCalcNextUpdateTime` as is commented out here:
https://github.com/dmcconachie-tri/drake/blob/db4e5408a2c1144a7a0662f4424924aed72b1e95/systems/lcm/lcm_interface_system.cc#L71-L72

dmcconachie-tri, 2021-02-13:
/cc @jwnimmer-tri

jwnimmer-tri, 2021-02-13:
The primary question is whether `LcmInterfaceSystem` is violating the contract of `Simulator` or `System`.

If yes, then we should fix `LcmInterfaceSystem` to comply with whatever contract its breaking, and then also change one or both of `Simulator` or `System` to fail-fast when confronted with this case.

If no, then we should fix the `Simulator` to obey the event contracts.

sherm1, 2021-02-16:
I've taken a look at this now. Good sleuthing, @dmcconachie-tri! You made it easy to investigate, thanks.

There is no defined contract that says you _must_ produce an event to go along with a trigger time returned from a `DoCalcNextUpdateTime()` override. Yet as Dale pointed out, `Simulator::IntegrateContinuousState()` ignores the one piece of information it is provided (`next_timed_event_time`) that could tell it a time-triggered no-op event occurred -- instead it assumes that at least one of `next_publish_time` or `next_update_time` contains the trigger time. I think that's a Simulator bug.

Returning a dummy event (as Dale tried) is a safe workaround but shouldn't be necessary. I'll file a more tightly-focused issue and we'll figure out a fix.

sherm1, 2021-02-16:
Whoops, no need for a new issue -- see #12620!

sherm1, 2021-02-16:
FYI @edrumwri for advice

edrumwri, 2021-02-19:
Just looked at #14663 - I don't agree that this is a bug in Simulator. From the documentation of System::CalcNextUpdateTime(.):

> The System returns the next time at which some discrete action must be taken, and records what those actions
  ought to be in @p events.

That sure sounds like a contract to me- the system is tasked with returning both the time *and* the "discrete action to be taken".

I will argue that the correct course of action is:
1. DoCalcNextUpdateTime documentation should be updated to explicitly state that events are required when time < inf
2. An exception should be thrown when DoCalcNextUpdateTime(.) has created no events when time < inf, since it looks like the derived system just forgot to specify the triggering event
3. Simulator should DRAKE_DEMAND that at least one event was created when the next update time < inf.
4.

sherm1, 2021-02-19:
Thanks, Evan -- that's a very plausible alternative. The overriding code in this case would be required to construct and return a no-op Event object. Is there a downside to treating no-returned-Event as a no-op?

Any thoughts on @edrumwri's alternative, @dmcconachie-tri @jwnimmer-tri ?

edrumwri, 2021-02-19:
> Is there a downside to treating no-returned-Event as a no-op?

Other than the one that I identified? It would be good if one were able to avoid the no-op virtual function call, but I think that's a pretty mild cost.

PR #14663 (2021-02-24): Make sure DoCalcNextUpdateTime() timed trigger is accompanied by an Event object.
When an overridden `DoCalcNextUpdateTime()` reported a trigger time but didn't return any Event object to serve as a handler, the trigger time was being ignored with time advancing as though nothing happened. But that could cause time-triggered events that should have occurred later in that same step to be missed entirely. LcmInterfaceSystem wasn't returning an Event object from its DoCalcNextUpdateTime() override and caused problems.

This PR changes the behavior to treat a no-Event timed trigger as an error. A  [... cut]
closed 2021-02-24 by sherm1