RobotLocomotion/drake #14652: Use clang 10 as the supported clang compiler on bionic and focal

jamiesnape, 2021-02-16:
Clang is 10 is the most recent clang available in both distributions and is the most similar clang to the currently supported compiler on macOS.

mwoehlke-kitware, 2021-12-07:
The most recent clang on Focal is 12. At this point, should this issue still be targeting 10? Do we still care about Bionic?

jwnimmer-tri, 2021-12-07:
> Do we still care about Bionic?

See #13391.  We need to support Bionic until April 1, 2022.

jwnimmer-tri, 2021-12-09:
If this needs a lot of work on the macOS side (i.e., rolling a new bottle), then it probably makes sense to wait until April at which point we can go straight to 12.

jwnimmer-tri, 2022-02-03:
One way forward might be to switch Focal and macOS (only) to a newer Clang?  That would be OK by me.

jwnimmer-tri, 2022-03-03:
New plan: wait until Bionic is gone and then switch to Focal's newest (probably 12).

mwoehlke-kitware, 2022-04-05:
It looks like we will also need to bottle our own clang-format for macOS? (It seems we already do this for clang-9...)

svenevs, 2022-04-11:
Mac side is ready, we bottled `12.0.0` since that's what is in focal right now (rather than `12.0.1`).  The dependencies in the brew formula are build only, so the only reason we may need to update those in the future is if `focal-updates` provides `clang-format-12` as `12.0.1` at some point.

```diff
--- a/setup/mac/source_distribution/Brewfile
+++ b/setup/mac/source_distribution/Brewfile
@@ -4,4 +4,4 @@
 tap 'robotlocomotion/director'
 
 brew 'bazel'
-brew 'robotlocomotion/director/clang-format@9'
+brew 'robotlocomotion/director/clang-format@12'
```

is ready to go for the PR that does the update.

jwnimmer-tri, 2022-05-10:
#16937 has merged, which was most of the work here.  I think all that remains is to update the Provisioned images to reflect the setup changes in that PR (removing some packages).

svenevs, 2022-05-11:
Focal provisioned AMIs are updated, [linux-focal-clang-bazel-experimental-release #579](https://drake-jenkins.csail.mit.edu/view/Clang/job/linux-focal-clang-bazel-experimental-release/579/console) successfully used clang-12 on `Running on [EC2 (us-east-1) - drake-linux-focal-2022-05-10 (i-057d1ea0519d1477b)`.