RobotLocomotion/drake #14654: libclang_setup.py should allow for customizable locations of libclang

brawner, 2021-02-16:
From feedback over at https://github.com/RobotLocomotion/drake/pull/14648#issuecomment-779499081, @jamiesnape suggested that the currently hardcoded path for libclang might be able to be parameterized. Currently the executable that calls it `mkdoc` doesn't accept any command-line arguments, but it seems like some toolchain configurations from bazel might be able to be passed on to mkdoc. 
 
https://github.com/RobotLocomotion/drake/blob/6f08f1b83b88a7932ab162f0412e820bddd4a580/tools/workspace/pybind11/libclang_setup.py#L40

I ran into this issue when trying to build with clang-11 and c++20 (https://github.com/RobotLocomotion/drake/pull/14648). The `mkdoc` targets won't make use of the CC variable here.
```
bazel build --cxxopt=-std=c++20 --repo_env=CC=clang-11  //...
```

jwnimmer-tri, 2021-02-16:
@jamiesnape It is not clear to me why the header file comment-scraping tool used to populate pydrake's help strings should have anything to do with the user's `$CC` that is used to generated Drake's object code.  We know that `libclang-9.so` will correctly parse the headers.  What is the downside of leaving the comment processor rigidly fixed to a given libclang path?

jamiesnape, 2021-02-16:
Apparently it made a difference in #4648. I do not know if it is a flags issue or a compilation issue there, but certainly libclang is doing much more than parse comments since it needs the signature associated with the comment.

jamiesnape, 2021-02-16:
Of course it does not need to match exactly (since we pair it with GCC often), but the capabilities need to match. I assume the mismatch between 9 and 11 was too great. It is a backlog issue anyway and wouldn't take long if it reached the top of the stack.

jwnimmer-tri, 2021-02-16:
@brawner could you post more details about the error that occurs when you don't locally edit `libclang-9.so` to be newer?  Possibly just paste the error message?  Perhaps it's only that `-std=c++20` was propagated along to the header parsing?

brawner, 2021-02-16:
@jwnimmer-tri Pretty much. It complains that c++20 is not a valid c++ value.

```
error: invalid value 'c++20' in '-std=c++20'
note: use 'c++98' or 'c++03' for 'ISO C++ 1998 with amendments' standard
note: use 'gnu++98' or 'gnu++03' for 'ISO C++ 1998 with amendments and GNU extensions' standard
note: use 'c++11' for 'ISO C++ 2011 with amendments' standard
note: use 'gnu++11' for 'ISO C++ 2011 with amendments and GNU extensions' standard
note: use 'c++14' for 'ISO C++ 2014 with amendments' standard
note: use 'gnu++14' for 'ISO C++ 2014 with amendments and GNU extensions' standard
note: use 'c++17' for 'ISO C++ 2017 with amendments' standard
note: use 'gnu++17' for 'ISO C++ 2017 with amendments and GNU extensions' standard
note: use 'c++2a' for 'Working draft for ISO C++ 2020' standard
note: use 'gnu++2a' for 'Working draft for ISO C++ 2020 with GNU extensions' standard
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/hash.h:138:39: error: no template named 'optional' in namespace 'std'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_monomial_basis_element.h:168:24: error: no member named 'is_same_v' in namespace 'std'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_monomial_basis_element.h:168:34: error: 'BasisElement' does not refer to a value
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_monomial_basis_element.h:166:22: note: declared here
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_monomial_basis_element.h:172:24: error: no member named 'is_same_v' in namespace 'std'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_monomial_basis_element.h:172:34: error: 'BasisElement' does not refer to a value
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_monomial_basis_element.h:166:22: note: declared here
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:157:14: error: no member named 'is_same_v' in namespace 'std'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:157:48: error: expected '(' for function-style cast or type construction
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:288:10: error: no member named 'is_base_of_v' in namespace 'std'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:288:23: error: 'PolynomialBasisElement' does not refer to a value
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_polynomial_basis_element.h:46:7: note: declared here
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:288:60: error: expected ';' after alias declaration
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:289:23: error: use of undeclared identifier 'BasisElement'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:292:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:298:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:305:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:311:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:317:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:323:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:329:1: error: no template named 'GenericPolynomialEnable'
bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_headers/drake/common/symbolic_generic_polynomial.h:335:1: error: no template named 'GenericPolynomialEnable'
fatal error: too many errors emitted, stopping now [-ferror-limit=]
Traceback (most recent call last):
  File "/home/brawner/.cache/bazel/_bazel_brawner/9e7b33f8778394ea8277807e1cb12d27/sandbox/linux-sandbox/1380/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 841, in <module>
    main()
  File "/home/brawner/.cache/bazel/_bazel_brawner/9e7b33f8778394ea8277807e1cb12d27/sandbox/linux-sandbox/1380/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 801, in main
    raise RuntimeError(
RuntimeError: Parsing headers using the clang library failed with 20 error(s) and 1 fatal error(s)
----------------
[... cut ...] ermine if that appears to be the problem. Since this target is built in the host configuration, the only way to change its version is to set --host_force_python=PY2, which affects the entire build.

If this error started occurring in Bazel 0.27 and later, it may be because the Python toolchain now enforces that targets analyzed as PY2 and PY3 run under a Python 2 and Python 3 interpreter, respectively. See https://github.com/bazelbuild/bazel/issues/7899 for more information.
----------------
```

jwnimmer-tri, 2021-02-16:
Ah, I see it now.  I did not realize we were passing the cxxopts from the object code build down into the header scraping:

https://github.com/RobotLocomotion/drake/blob/8f12522a976b1820e0e76fdfa1654a69e05131a2/tools/skylark/pybind.bzl#L371

jamiesnape, 2021-02-16:
I think the C++ version was even the reason we did that. So two paths: maybe we go further and hardcode everything or we allow the options through and allow library choice. It works at the moment because we are completely controlling compiler choice and assume no one is going to override certain flags when they use Drake in their project. Obviously we don't support CC=clang-11 and --cxxopt=-std=c++20, but are we being unnecessarily strict or difficult, when it is probably something that works (we usually warn about newer things and error on older things).

jamiesnape, 2021-02-16:
Honestly, I would be surprised if the C++ standard of 17 vs 20 made too much difference here, but there is definitely a minimum version required to catch language features (the default is probably 98 or 11, I forget).

PR #14669 (2021-02-18): pybind: Hard code -std=c++17 option spelling
We know that our hard-coded libclang handles Drake's minimum C++ standard revision, so we can just hard-code that spelling for mkdoc.  This allows users to build object code using -std=c++20 mode without angering libclang.

I confirmed that on both Ubuntu and macOS, in a default configuration, the _only_ cxxopts coming in through the fragment was `-std=c++17`.

Closes #14654.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14669)
closed 2021-02-18 by jwnimmer-tri