RobotLocomotion/drake #14670: Add C++20 support

jwnimmer-tri, 2021-02-18:
We should add official support for building drake under `-std=c++20`, for compilers which support it.  (Relates: #14652.)

We would still require that `-std=c++17` continue function as well, at least as long as we support Ubuntu 18.04 with GCC 7.5 still in play.

This might mean that we need an extra CI variant, or maybe we just flip all Clang CI to C++20 and have GCC CI with C++17.  To be determined.

(Adding this a discussion placeholder for now.)

jamiesnape, 2021-03-14:
I figure, given #1183, we would not want to push the envelope too far. C++20 compilation should certainly work, but perhaps C++17 may want to stick around for a while (which pushes me toward an extra variant, I think).

jwnimmer-tri, 2021-11-10:
Let's say the specific call to action here will be to add two new builds to Jenkins:
- `linux-focal-clang-bazel-experimental-release-cxx20`
- `linux-focal-clang-bazel-weekly-release-cxx20`

(Using some variation of those names is fine also, in case that makes the scripting easier.)

They should be identical to the current `linux-focal-clang-bazel-continuous-release` except that they will add `--copt=-std=c++20` to the bazel command line (and not be in "continuous", obviously).

jwnimmer-tri, 2022-03-02:
At this point, perhaps the best option is to actually just wait for Ubuntu 22 ("Jammy") CI builds to appear, and then use C++17 on Focal and C++20 on Jammy, by default.  That way, CI configs and user configs could be just the same.

PR #17943 (2022-09-29): Use C++20 on macOS and Ubuntu 22
Closes #14670.

We'd like to ensure that Drake's users who build in C++20 mode don't receive any errors. In order to get that covered by CI, the plan is to build Ubuntu 20.04 in C++17 mode (unchanged), but to upgrade Ubuntu 22.04 and macOS to C++20 mode. This PR handles the Ubuntu side of that change.

Prerequisites:
- [x] #17944
- [x] #17949
- [x] #17958
- [x] #17970
- [x] #17973
- [x] #17994
- [x] #17995

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17943)
closed 2022-09-29 by jwnimmer-tri