RobotLocomotion/drake #14680: Control force (vector) visualization for hydroelastic contact in drake_visualizer

SeanCurtis-TRI, 2021-02-18:
# The Problem

Drawing force vectors in dynamic visualization is problematic. We need to communicate magnitude and direction. For *point* contact we have a dialog giving three options:

  - __Fixed Length__: All force vectors are drawn as a fixed length arrows. Direction is clear, but there is no way to assess relative magnitude.
  - __Scaled__: All force arrows' lengths are drawn proportional to the force magnitude.
  - __Auto Scaled__: The largest *instantaneous* force is rendered with a fixed length arrow. All other force arrows are scaled smaller, based on their relative proportion to the maximum magnitude.

For all three options, the arrows can be globally scaled up or down, and forces below a certain threshold can be omitted. The dialog looks like this:

[image]
__Figure 1__: Force Vector Visualization dialog for point-pair contact

(In the past we've also played with arrow *thickness* and arrow *color* to communicate magnitude but they were insufficiently compelling to land in master.)

For hydroelastic contact, the *hard-coded* option is "fixed length". What we want is for the vectors in hydroelastic contact to be as configurable as those for point contact.

There are multiple TODOs in the code which would be resolved by addressing this issue:

  - `show_hydroelastic_contact.py`
    - [Request to make scaling force dialog box available](https://github.com/RobotLocomotion/drake/blob/cfb842aef7663bc73d1ff804bafa258f902da98a/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py#L19-L20)
    - [Additional request to make force visualization configurable](https://github.com/RobotLocomotion/drake/blob/cfb842aef7663bc73d1ff804bafa258f902da98a/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py#L408-L409)
    - [Marking some zombie code that currently can't be executed, because force visualization isn't configurable](https://github.com/RobotLocomotion/drake/blob/cfb842aef7663bc73d1ff804bafa258f902da98a/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py#L609-L610)
  - `show_point_pair_contact.py`
    - [Note to refactor the Qt GUI for force visualizatoin](https://github.com/RobotLocomotion/drake/blob/cfb842aef7663bc73d1ff804bafa258f902da98a/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_point_pair_contact.py#L17-L18)

# Design Challenges
Resolving this issue has some design challenges:

## Disparate quantities

`show_point_pair_contact.py` has only one vector quantity to display per contact: contact force. For each contact in hydroelastic, we have contact force, *moment*, *tractions at quadrature points*, *slip velocities at quadrature points*. The dialog above shows a *single* scaling factor and a *single* cut-off value. It is *not* reasonable to assume that a single "vector visualization" configuration is appropriate for all four quantities. They would need to be configured independently.

## Visualization state

Currently, depending on how the point-contact force visualization is configured, a piece of text is displayed at the bottom, center of the 3D viewport (see image below).

[image]
__Figure 2__: Display of force vector visualization shown in 3D viewport.

This should not be considered our guiding star as we move forward.
1. If we allow the point-pair force vectors to be configured independently of the hydroelastic force vectors, a single piece of text will no longer be sufficient. It won't be clear if it is relevant.
    a. We definitely have cases where we can have both point-pair and hydroelastic contact results visualized simultaneously.
2. If we allow the other hydroelastic vector quantities to be likewise independently configured, that would require even more text in the HUD.

In short, visualization feedback in the HUD is not a long term solution as it doesn't scale well.

## Code Reuse

It would be regrettable to copy and paste (and possibly mutate) the [GUI code present in `show_point_pair_contact.py`](https://github.com/RobotLocomotion/drake/blob/cfb842aef7663bc73d1ff804bafa258f902da98a/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_point_pair_contact.py#L54-L114) into `show_hydroelastic_contact.py`. It would be much better if the code were re-expressed such that it could be effectively reused.

# Proposal

1. We should have the concept of "vector" visualization (agnostic of what quantity the vector represents). A mechanism for mapping a vector quantity to a visual arrow should be introduced.
  a. A Qt `QFrame` should be introduced which captures the functionality in the dialog shown in Figure 1. It should be reusable in that words like "force" can be swapped at instantiation (i.e., once for "force", once for "moment", etc.). It should also allow for display of the appropriate units.
  b. Configuring the settings in a frame, control a corresponding vector->arrow mapping mechanism.
  c. This functionality should *not* belong to either `show_*_contact.py` files, but live in its own module, importable by either module.
  d. This is *largely* a refactoring of the existing code in `show_point_pair_contact.py` with some minor extensions to allow for variability.
2. `show_point_pair_contact.py` dialog becomes simpler.
    a. It loses its own definition of the frame, simply creating a dialog around the frame defined in its own module.
    b. Its bespoke implementation of the various vector->arrow mapping modes, gets replaced with an instance of a mechanism that handles the logic.
3. `show_hydroelastic_contact.py` will construct its own dialog in which the frame defined above would appear multiple times (once for each *type* of vector quantity).
  a. Make sure that all of the options have valid code paths.
4. We eliminate the display in the HUD of the "Contact vector: Thing".
  a. The best option for giving feedback is to make the dialogs modeless. That is outside the scope of this PR.
  b. Cluttering up the gui arbitrarily is simply a bad choice. It'd be better to simply remove the ambiguity and require the user to look it up than to provide incomprehensible feedback.
  c. Alternatively, if we devise an alternative solution that we're interested in implementing, that'd be fine too.

jwnimmer-tri, 2023-07-30:
The above solution text is very focused on `drake_visualizer`, which is legacy code that we are not going to improve anymore.

Would it be fair to close this issue as "unplanned", and leave it to #17680 to track the meshcat flavor of this request?  (Maybe copying a few ideas from this ticket into a comment in that one?)

DamrongGuoy, 2023-08-01:
I agree that we can close #14680 for drake_visualizer and leave it to #17680 for Meshcat.

closed 2023-08-01 by DamrongGuoy