RobotLocomotion/drake #14688: Unnecessary warnings/workarounds when creating plant for non-simulation uses

calderpg-tri, 2021-02-22:
In TRI code, when creating a `MultibodyPlant` for non-simulation uses (ex planning or control), we find ourselves either (a) setting `time_step = 0.0` and getting unnecessary console warnings about joint limits not being supported in continuous operation, or (b) setting `time_step = 1.0` to suppress these warnings.

Neither approach seems ideal - it appears that an additional constructor or option for creating a `MultibodyPlant` specifically for non-simulation uses would better capture the behavior we want.

cc @EricCousineau-TRI @jwnimmer-tri @rpoyner-tri

sherm1, 2021-02-22:
Better would be to support joint limits properly!

jwnimmer-tri, 2021-05-04:
As briefly discussed in `drakedevelopers#dynamics` slack yesterday, another possible solution is to relocate the warning from `Finalize` to immediately before the (skipped) call to `AddJointLimitsPenaltyForces`, so if and only if we try to do something that depends on the joint limits.

jwnimmer-tri, 2021-05-06:
In #15017, I've changed when the warning occurs -- not during Finalize, but rather only when computing contact forces.  Possibly this is sufficient to resolve the issue?  Up to you @calderpg-tri.

calderpg-tri, 2021-05-26:
Now that the relevant planning code in anzu has stopped needing to provide a non-zero timestep to suppress warnings, I consider this fixed.

closed 2021-05-26 by calderpg-tri