RobotLocomotion/drake #14691: Remove scipy dependency

jamiesnape, 2021-02-23:
It is barely used and the code that uses it really should be able to avoid doing so without too much more complexity. We are very disk space constrained on GitHub Actions so every megabyte counts with large dependencies, and Google Colab prefers certain versions of certain dependencies that may or may not align with an Ubuntu provided version.

FYI @RussTedrake @jwnimmer-tri.

jwnimmer-tri, 2021-11-10:
FYI the uses as of today:

```
bindings/pydrake/systems/planar_scenegraph_visualizer.py: from scipy import spatial
bindings/pydrake/systems/pyplot_visualizer.py: import scipy.interpolate
examples/acrobot/optimizer_demo.py: from scipy.optimize import fmin
```

Fixing the first two would at least move us from "runtime dependency" to "test / demo only dependency", which would be a good start.

SeanCurtis-TRI, 2021-11-10:
- `pyplot_visualizer.py` has a trivial resolution. It's only doing a linear interpolation which could simply be written out.
- `planar_scenegraph_visualizer.py` is much more problematic as it calls `spatial.ConvexHull()` for which there is no trivial replacement.

RussTedrake, 2021-11-11:
Re: the functionality for `spatial.ConvexHull()`, I also needed this for some of the convex sets work, too.  @mpetersen94 , @TobiaMarcucci and I have a working branch the brings in qhull as a (from source) dependency, and implements the equivalent functionality in C++.  At this point, we just need to write the tests and get it in.

jwnimmer-tri, 2022-01-14:
We have qhull linked as a dependency now, but AFAICT there are no Drake functions available that would let the planar_scenegraph_visualizer compute the convex hull in two dimensions.   I think we'll need to trench through some more kinds of call(s) to qhull in C++, before we can adjust the Python.

RussTedrake, 2022-01-15:
Actually, the ergonomics could be a little better, but I think that 
```
pydrake.geometry.optimization.VPolytope(query_obect, geometry_id).vertices()
```
should do what you want (almost).  Unfortunately, it looks like the call site currently has the `inspector`, not the `query_object`, and it looks like `VPolytope` needs `ImplementGeometry(Mesh)` to delegate to `ImplementGeometry(Convex)` (actually, probably the other way around).  Those are minor changes, though, and shouldn't involve grok'ing anything about qhull.

jwnimmer-tri, 2022-01-15:
We need both 3D hulls (during initialization) and 2D hulls (while drawing).  I think that API is 3D only?  Or maybe you're saying that we inflate it to 3D by filling in zeros for one of the dimensions?

RussTedrake, 2022-01-15:
I agree it would be better to not do the padding. Then perhaps we could offer `VPolytope::MakeConvexHull` which works for arbitrary dimensions. Or just take the qhull calls inside the `VPolytope` and put them somewhere more central in the repo.  The qhull calls in `VPolytope::ImplementGeometry(const Convex& convex)` should do what we want, and be trivially adaptable to arbitrary dimensions (or 2d specifically).

ggould-tri, 2022-01-20:
`examples/acrobot/optimizer_demo.py` is IMO important to *have* (it shows users how to rig an optimizer to a scenario) but is not essential to *run* -- the risk of the code invalidating the demo is low.  We could choose a stub fmin in the test or simply remove `optimizer_demo_smoke_test`.

jwnimmer-tri, 2022-01-20:
First, I should remind us (including myself!) about the "runtime dependency" to "test / demo only dependency" above.  I'd forgotten about that detail.  Once we fix the `scipy.spatial` and `scipy.interpolate` problems, we can move the scipy dependency to be test-only, which would unburden our users.  We should focus on that milestone as the first priority.

Second, then for removing scipy entirely by nixing `fmin` -- maybe something like this added to `optimizer_demo.py`?

```python
def fmin(*, func, x0, maxfun):
    """This is a stub implementation of scipy fmin, so Drake doesn't need to depend on scipy.
    Replace this with the real scipy to experience the full demo."""
    # ... etc ...
```

Where the "etc" is some simple stub implementation.  Either just a single call to `func(x0)` to prove out the plumbing, or else try `maxfun` calls with random perturbations to `x0` and return the best lucky guess.

jwnimmer-tri, 2022-01-27:
For the record, we do **not** need to remove this usage (from a comment):
```
./math/test/bspline_basis_test.cc\0177:    from scipy.interpolate import BSpline
```

svenevs, 2022-02-09:
Is somebody able to direct me to an example that actually calls `PyPlotVisualizer.animate` (for testing changes around `scipy.interpolate.interp1d`)?

jwnimmer-tri, 2022-02-10:
@rpoyner-tri could you please answer the question immediately above?

rpoyner-tri, 2022-02-11:
Strictly speaking, VectorLog{,Sink} can't guarantee anything. When used with a Simulator in the typical way, I think we have the guarantee that time is non-decreasing. There might be duplicate-time records (I don't recall the details), but I don't think you'll see backwards time flow.

Of course, it's drake, so with even moderate Context/time abuse, you could construct an unsorted  log.

\cc @sherm1 for further Simulator insight.

sherm1, 2022-02-11:
Correct, the Simulator will produce only non-decreasing times. Duplicates can occur (at least in principle) due to a state-modifying Event coinciding with a logging event. In that case you may get two entries at time t -- t⁻ before the Event is handled and t⁺ afterwards.

sherm1, 2022-02-11:
But as @rpoyner-tri says, any other way of producing log entries won't automatically come with the same guarantee.

svenevs, 2022-03-21:
I was looking into the PR that would remove things, we have another snag: `drake-visualizer`.  [RobotLocomotion/director search results for scipy](https://github.com/RobotLocomotion/director/search?q=scipy) and my own personal lack of understanding of what code is or is not actually executed makes me hesitant to want to remove the dependency from drake.

After #16765 is finished I believe we can safely

1. Remove `python3-scipy-dbg` from `setup/ubuntu/source_distribution/packages-bionic-test-only.txt` and `setup/ubuntu/source_distribution/packages-focal-test-only.txt` since none of the drake tests will be using `scipy` anymore.
2. We can remove it from `setup/mac/binary_distribution/Brewfile` after 04-01-2022 per #16386.

But the linux `drake-visualizer` deprecation is currently not planned for many moons (e.g., after focal is dropped).

------------------------

An alternative to consider: we could add a `try: import scipy` to the drake-visualizer python script that issues a `WARNING` of some kind stating `scipy is not installed, some functionality of Drake Visualizer may not work`?  I am able to run the examples I am familiar with using `drake_visualizer` **without** having `scipy` installed, but from the search results it seems like the "advanced functionality" of `drake_visualizer` is what is using `scipy`.

jwnimmer-tri, 2022-03-21:
Almost all the Python source code in director's repository (and Docker-built image) is dead code when used within Drake.  If we can't tickle it by running the visualizer with all of our plug-ins turned on, then I think it's fine to leave it alone.  I can help verify that assumption, when it comes time.

svenevs, 2022-03-22:
We've hit a roadblock with sparse matrices getting bound from Eigen.  [pybind11 maps `Eigen::SparseMatrix` to scipy](https://pybind11.readthedocs.io/en/stable/advanced/cast/eigen.html), so a decision needs to be made about how to bind things on the python side.  Some examples:

https://github.com/RobotLocomotion/drake/blob/11388ee6352d22acf62665d522a3963fb4e4123d/solvers/constraint.h#L327

https://github.com/RobotLocomotion/drake/blob/11388ee6352d22acf62665d522a3963fb4e4123d/solvers/constraint.h#L410

Bindings take place here

https://github.com/RobotLocomotion/drake/blob/11388ee6352d22acf62665d522a3963fb4e4123d/bindings/pydrake/solvers/mathematicalprogram_py.cc#L1851

`bazel test --cache_test_results=no //bindings/pydrake/solvers:py/mathematicalprogram_test` (making sure you `sudo apt-get remove python3-scipy`) is the success criteria.

Slack discussion: https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1647977988925529

jwnimmer-tri, 2022-03-22:
Here's my proposal.  We'll commit to never meaningfully using `scipy.sparse` in any of our tests or examples.  We still bind things that mention it, but we always unit test them using stubs that just convert to dense to help with testing.

The plan:
- [ ] Change our bindings unit tests to use a scipy.sparse stub (#16840).
- [ ] Announce in the next release notes that Drake's binary releases' dependency on scipy is deprecated; then after three months, we'll stop automatically pulling it in.  Users who want to use sparse matrices after that will need to pull in scipy on their own.
- [ ] The bindings of Drake functions that deal in Eigen::Sparse will fail at runtime ("Cannot import scipy") if the user doesn't have scipy.  We could imagine patching pybind11 to produce a better error message in that case, but _shrug_.

svenevs, 2022-03-22:
Second checklist item should also update https://drake.mit.edu/python_bindings.html to explain what happens with sparse matrices (namely, they still get bound and are usable just need to install `scipy`).

------------

To satisfy my curiosity (in spare time), the way to produce a better error message seems to be

```diff
diff --git a/bindings/pydrake/solvers/mathematicalprogram_py.cc b/bindings/pydrake/solvers/mathematicalprogram_py.cc
index 92b4065ddd..d14fa92e58 100644
--- a/bindings/pydrake/solvers/mathematicalprogram_py.cc
+++ b/bindings/pydrake/solvers/mathematicalprogram_py.cc
@@ -1729,7 +1729,19 @@ void BindEvaluatorsAndBindings(py::module m) {
           py::arg("A"), py::arg("b"),
           py::arg("eval_type") = LorentzConeConstraint::EvalType::kConvexSmooth,
           doc.LorentzConeConstraint.ctor.doc)
-      .def("A", &LorentzConeConstraint::A, doc.LorentzConeConstraint.A.doc)
+      .def(
+          "A",
+          [](const LorentzConeConstraint& self) {
+            try {
+              py::module_::import("scipy.sparse");
+            } catch(const std::exception&) {
+              throw pybind11::import_error(
+                  "LorentzConeConstraint.A() returns a scipy.sparse matrix, "
+                  "but `scipy` is not installed.");
+            }
+            return self.A();
+          },
+          doc.LorentzConeConstraint.A.doc)
       .def("b", &LorentzConeConstraint::b, doc.LorentzConeConstraint.b.doc)
       .def("eval_type", &LorentzConeConstraint::eval_type,
           doc.LorentzConeConstraint.eval_type.doc)
```

It doesn't seem possible to actually `try { return self.A() } catch (... anything ...)` but you can try to do the import first and give a nicer error message if desired.  Works with the stubs pr for testing.

```py
>>> constraint.A()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: LorentzConeConstraint.A() returns a scipy.sparse matrix, but `scipy` is not installed.
```

That said, the error without this is

```py
>>> constraint.A()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'scipy'
```

which is ultimately still going to point the user in the right direction.  Currently there are three places to do a fancy exception: `CsdpSolverDetails::Z_val`, and both of the `Lorentz` constraints.  If this does happen, the doxygen docs on python bindings should explain this.  Although nice error messages are nice... it's an opportunity for things to become inconsistent if/when the number of sparse bindings increases.

jwnimmer-tri, 2022-05-06:
The next action due here is on 2022-07-01, where we can drop scipy from our setup files.

closed 2022-07-28 by jwnimmer-tri

jwnimmer-tri, 2023-01-05:
For the record, on Ubuntu 22.04 "Jammy", Drake has regained an (implicit) dependency on scipy, via the chain `python3-matplotlib => python3-fonttools => python3-scipy`.

Between Ubuntu 20.04 and Ubuntu 22.04, the packaging of `matplotlib` gained a dependency on `python3-fonttools`.

PR #17559 (2022-07-28): [workspace] Remove scipy dependency
Closes #14691.

I cannot re-open #16836 after having force pushed first apparently, hence new PR.

Depends:

- [x] #17570

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17559)