RobotLocomotion/drake #14692: Separate from Drake and/or make optional drake-visualizer

jamiesnape, 2021-02-23:
It significantly increases the number of transitive dependencies of Drake (and so increases disk space used). Its own development is basically stopped so we are unlikely to get out of sync. Requires drake-visualizer to stop using dependencies of Drake, but that will be easy. Making sure the Drake specific libraries and plugins are tested and available also needs to be addressed (it could still be available to Bazel, but not installed, and that would be some way toward resolving this issue).

FYI @RussTedrake @jwnimmer-tri.

jwnimmer-tri, 2021-12-14:
See also #16215 for a discussion of platform support timelines, which dovetails with this issue.

jwnimmer-tri, 2022-05-06:
Resolving as a duplicate of #16780.

closed 2022-05-06 by jwnimmer-tri