RobotLocomotion/drake #14695: Drake website beta testing

jwnimmer-tri, 2021-02-23:
As the final step of #13832, we're going to ask a few people to beta test the website redesign.

Preview instructions are given at https://drake.mit.edu/documentation_instructions.html (though see [1], [2] errata at the bottom of this posting).   Remember to pull the latest master branch first.

Testers are welcome to open a PR with changes they'd like to suggest, and we can merge those at any point.

Alternatively, problems that people identify which they can't or won't themselves fix should be posted as independent comments in this issue.  We'll come by later and turn any required fixes based on those comments into a checklist in the top post here.

---

[1] Note that you need to run `install_prereqs` with extra flags first per the top of that page, and then jump down to the Jekyll section..

[2] The instructions have stale advice _"Currently, this process will generate or serve an empty page (index.html)."_ .  In fact, the preview has all of the content at this point.

ToffeeAlbina-TRI, 2021-02-24:
Are all the links supposed to be functional?
The [Getting Help](http://127.0.0.1:8000/getting-help.html) link on the homepage is Not Found
[2021-02-24 14:56:55] ERROR `/getting-help.html' not found.

jwnimmer-tri, 2021-02-24:
Yup, everything should be finished, so anything that looks or operates wrong is unexpected.

The link should point to `getting_help.html` not `getting-help.html`.

joemasterjohn, 2021-02-25:
There are quite a few explicit uses of `http://` rather than `https://`. For most websites that implement strict transport security (github, etc.) this is mostly fine, but a few sites (groups.csail.mit.edu/locomotion) don't and won't redirect to HTTPS. Any particular reason for the explicit http, or should they be changed? 

Hopefully everyone is using something like HTTPS Everywhere.

jwnimmer-tri, 2021-02-26:
Comments will be due by March 3rd.  I guess we'll pull the trigger and go live soon after that.

avalenzu, 2021-02-26:
In `from_binary.html`, the "There are **binary packages** of Drake available at:" link points to a comment on a github issue: https://github.com/RobotLocomotion/drake/issues/1766#issuecomment-318955338

Is that intentional? Seems odd given that the links to the actual tarballs are below.

jwnimmer-tri, 2021-02-26:
> In from_binary.html, ...

That matches the current website: https://drake.mit.edu/from_binary.html

I'll add it to the #13539 checklist as something to fix later on.

avalenzu, 2021-02-26:
So in our beta testing, should we only flag things that don't match the current website?

jwnimmer-tri, 2021-02-26:
I think its good to point out anything you don't like, but not all of the bugs will be blockers for switching over to the redesign.

avalenzu, 2021-02-26:
On `developers.html#supported-configurations`, the table didn't quite translate correctly. The original has merged cells for values that are common between rows (e.g. Bazel and Java versions)
[image]
but the new one does not
[image]

avalenzu, 2021-02-26:
In `developers.html#configuration-management-non-determinism` the "following the instructions" link points to `build_from_source.html` which does not exist. It should be `from_source.html`.

avalenzu, 2021-02-26:
In `developers.html#issue-tracking` , the "Platform Reviewer Checklists" link points to `platform-reviewer-checklist.html` which does not exist. It should be `platform_reviewer_checklist.html`.

avalenzu, 2021-02-26:
In `developers.html#review-process-tooling` the "Tips for participating ..." link has been incorrectly promoted to a heading.
Original:
[image]
New:
[image]

avalenzu, 2021-02-26:
In `jenkins.html` ("Scheduling Builds via the Jenkins User Interface" section), the links to continuous, nightly, and weekly builds all have spurious trailing `>`s.

sherm1, 2021-02-26:
On the front page, bottom left are the words "CORE LIBRARY". I assumed that was a link but it appears just to be text. It's unclear to me why it is there at all.

jwnimmer-tri, 2021-02-26:
> It's unclear to me why [CORE LIBRARY] is there at all.

I think it's supposed to be a header (title) for the three boxes below it within that same gray region.

sherm1, 2021-02-26:
Oh, NVM! It formatted so tidily I didn't realize I wasn't looking at the whole page (missed the skinny scrollbar).

sherm1, 2021-02-26:
The link at Resources/For Developers/Issue Tracking/Platform Reviewer Checklists (`http://127.0.0.1:8000/platform-reviewer-checklist.html`) says "Not Found" but other links on that page work.

sherm1, 2021-02-26:
FYI, just in case I'm not the only person making this mistake -- I didn't realize there was any more to the front page than I initially saw because (due to random window sizing I suppose) this is what I saw:
[image]

I didn't notice the skinny scroll bar to the right and there was nothing at the bottom indicating more below. Many websites that require scrolling have a down arrow at the bottom to notify the unobservant among us who might miss it.

hongkai-dai, 2021-02-26:
On Resources/Getting Help/Asking Your Questions/, we say "please email Russ Tedrake for access", shall we add Russ's email here?

jwnimmer-tri, 2021-02-26:
@sherm1 

> ... (due to random window sizing I suppose) ...

The style is actually written so that we always see this.  If you resize your window you'll see that the "core library" is always there on the bottom, even in significantly different viewport sizes.

ggould-tri, 2021-02-26:
* Lighthouse report: https://htmlpreview.github.io/?https://gist.githubusercontent.com/ggould-tri/65ccee694ff4420134224bdfcbae65e4/raw/f72bf321ea2e1129625e43befdba7804bff3f1e1/lighthouse.html identifies only minor usability issues; I don't see anything here that merits special consideration except for its complaint about contrast.

* [image] On a large window, the fonts are tiny and the majority of screen real estate blank.  This comes from the use of a fixed 'px' font size, which is always going to be tiny on large monitors.  It is possible that this is intentional but it is funny-looking  and not awesome responsive design.

hongkai-dai, 2021-02-26:
On Resources/Getting Help /Asking Your Question/, when clicking the link "submit a pull request", it jumps to the page `developers.html#pull-request`. But it seems to me that there is not a `pull-request` tag on `developers.html`.

ggould-tri, 2021-02-26:
`main.css`:285 sets a top margin on the dropdown menus.  This causes the menu to vanish if you move your mouse too slowly down from the menu heading (because a mouse event occurs inside of the margin).  Deactivating this margin in developer tools removes the issue.

https://user-images.githubusercontent.com/17596413/109342331-b5976900-7839-11eb-80e2-3a4b5fbab224.mp4

ggould-tri, 2021-02-26:
vim.html broken markdown formatting for link
[image]

ggould-tri, 2021-02-26:
Release playbook, section "cutting the release", bullet 4 has a broken link: https://drake-jenkins.csail.mit.edu/view/Documentation/job/linux-bionic-gcc-bazel-nightly-documentation/ (url should include "unprovisioned")

hongkai-dai, 2021-02-26:
In  Code Review Checklist, the facepalm emoji is missing

[image]

sherm1, 2021-02-26:
- On the bottom part of the front page (just below "CORE LIBRARY") are three boxes. My first attempt was to click on the boxes but nothing happened. Then I realized the tiny red text below was actually the clickable part and in fact "API | TUTORIAL" was actually two different links. To me that seems more obscure than it needs to be. I think anything that looks like a clickable box should be clickable, and it should be more clear that the red text is clickable but the "CORE LIBRARY" text is not.
- Under these (non-clickable) boxes, the text that on the current Drake site says "Doc" has been changed to "API". The second "API" (under "Solving Mathematical Programs") leads to non-API documentation, although there is eventually an API link there. The third one (under "Multibody Kinematics and Dynamics") links just to some obscure documentation and doesn't lead to any meaninful API as far as I could tell. Consider changing at least that one back to "Documentation" or (better) just make the box clickable?

SeanCurtis-TRI, 2021-02-26:
To @sherm1's point about the unobvious-ness of scrolling. While the "core library" footer is always present as you scale the window, it isn't a clickable link and scrolling isn't obvious. I'd contrast that with TRI's robotics page (https://www.tri.global/our-work/robotics/) which includes "Scroll & Explore" with a down arrow. Very ergonomic.

ggould-tri, 2021-02-26:
On release playbook, the link "link" is broken with a spurious `>`: `https://github.com/RobotLocomotion/drake/pull/14208/commits/674b84877bc08448b59a2243f3b910a7b6dbab43%3E`

sherm1, 2021-02-26:
At the bottom of the front page (after scrolling all the way) there is a list of Acknowledgements with links. All of those work except the Office Of Naval Research link (currently `http://www.onr.navy.mil/`). It needs to be `https`.

ggould-tri, 2021-02-26:
`python_bindings.html` : extra period after "package"
[image]

SeanCurtis-TRI, 2021-02-26:
Legacy issue in `buildcop.html`

> Check the Continuous Production build dashboard in Jenkins at least once an hour during on-call hours. These builds run after every merge to Drake. Also check the Nightly Production build dashboard every morning and Weekly Production build dashboard on Monday morning. These builds are unusually resource-intensive, and therefore run at most once per day.

It seems when the weekly CI info was inserted (6ecb9bb7bf1), the weekly info was squeezed before "nightly" and "run at most once per day".

ggould-tri, 2021-02-26:
python_bindings page, "using the python bindings", the link under "Python API" has weird trailing giblets `https://drake.mit.edu/pydrake/index.html#://`.  These don't break anything but are not needed in order to load.

ggould-tri, 2021-02-26:
website_licenses.html Two of the linked documents (for the github mark and for python) are markdown but are rendered as text.

ggould-tri, 2021-02-26:
[image]
developers.html still has a .rst-style `:ref:`

ggould-tri, 2021-02-26:
[image]
developers.html A bulleted list has failed to format one of its items.

SeanCurtis-TRI, 2021-02-26:
In `clion.html`

 - Under "Building and Running Targets"
   - The new site corrects a bug in the old site. "the Bazel manual" is an actual link in jekyll, but mangled in sphinx. So, hooray, Jekyll!
 - Under "Lint selected file for google style guide"
   -   The string `$FILE_PATH$:$LINE$` is no longer simply an indented, "code"-formatted string, but it gets the full treatment of black text box. Furthermore, it leaves backticks in the printed value.

ggould-tri, 2021-02-26:
[image]
older_releases  Missing punctuation between sentences.

SeanCurtis-TRI, 2021-02-26:
In `code_style_guide.html`

 - Under Python - Additional Rules. 
   - The footnote on "Executable Python files should be limited..." is a live link in sphinx, but not in Jekyll.
   - The style for a "shell" block is horrible to read. There's basically a white screen, with a black block, with a bit of dark blue text in that black. Humans are bad at blue as it is, and when your pupil has shut down because of the white background, reading becomes taxing at best.

ggould-tri, 2021-02-26:
older_releases page is the only one with a leading "Back to ..." link.  This is ideosyncratic, especially as it is linked from other places than its backto parent.

SeanCurtis-TRI, 2021-02-26:
in `code_review_checklist.html

 - Under "Is the code the minimal set of what you want"
   - Sphinx contains guidance for finding deprecations. They are missing in Jekyll.

SeanCurtis-TRI, 2021-02-26:
in `documentation_instructions.html`

- I'm a bit surprised that there's not an extra warning that the sphinx portion will be going away.  :)

SeanCurtis-TRI, 2021-02-26:
In `code_style_tools.html`

- Under "Automated style checks"
  - Under "Bazel: Uses both pycodestylce like Python, and also `buildifier`.
  - `buildifier` is a link into `bazel.html`. Specifically, to the `#buildifier` tag. In Sphinx, it takes us to the middle of the page ("Updating build files"). In Jekyll, it takes us to the top of `bazel.html`.

SeanCurtis-TRI, 2021-02-26:
On `mac.html`

The link to `Source Instanllation (macOS, Ubuntu) is broken. Given as `from_-_source.html`.

hongkai-dai, 2021-02-26:
On developers.html/#review-process
[image]
Not sure if this :doc:buildcop should show up as in the screenshop, especially ":doc:" seems suspicious.

SeanCurtis-TRI, 2021-02-26:
In `getting_help.html`

- Under "Helpful Information"
  - Language (C++, Python), Python has a link to the python bindings. It lists `python-bindings.html`. It should be `python_bindings.html`.

SeanCurtis-TRI, 2021-02-26:
A passing note inspired by `python_bindings.html`.

Given a fixed-width browser, on the whole the pages have been readable. In this case, because of:

  - presence of a table of contents
  - Huge horizontal margins on said table
  - large font in code blocks

Much of the text is no longer available and requires me to scroll individual windows.  Sad. :(

jwnimmer-tri, 2021-03-01:
We should probably also test the site with javascript disabled?

ggould-tri, 2021-03-03:
Javascript disabled:

The gallery videos fail as expected, and with a correct error message with a working link
[image]

Everything else appears unchanged.

jamiesnape, 2021-03-03:
Is anyone testing with a mobile device profile (or even better an actual mobile device)?

ggould-tri, 2021-03-03:
I've viewed a few pages with chrome dev tools' mobile renderer; they are neither especially good nor especially bad.  If we really care about mobile users we should follow the lighthouse recommendations linked above (mobile readability being especially contrast sensitive, for example, because of glare) rather than mostly ignore them as I had suggested.

However debugging mobile presentation will be a lot easier once we're serving the pages; Drake isn't exactly useful from mobile, so I would propose deferring any serious mobile optimization until after the new site is up on `drake.mit.edu` in a viewable form.

jamiesnape, 2021-03-03:
However debugging mobile presentation will be a lot easier once we're serving the pages; Drake isn't exactly useful from mobile, so I would propose deferring any serious mobile optimization until after the new site is up on drake.mit.edu in a viewable form.

Certainly, though the Google indexer is mobile first now.

jwnimmer-tri, 2021-03-10:
The remaining open items have been moved into #14757.

closed 2021-03-10 by jwnimmer-tri