RobotLocomotion/drake #14725: python, doc: Should provide conventions / utilities to ensure consistent logging in Python, parity w/ spdlog

EricCousineau-TRI, 2021-03-02:
@calderpg-tri asked about how logging should happen in pure Python code, and how it should maintain parity w/ our `spdlog` usage via `drake::log()`.

Solution options:

1. Bind `drake::log()`, and ensure that our usage of the builtin Python `logging` module has analogous log levels
2. Ensure that cmdline parsing, etc., has some entry point for Python programs that allows parity btw `spdlog`.

My vote is for (2), since we already have redirects in place for `spdlog`, and it would be awkward repeated redirects (#13568).

jwnimmer-tri, 2021-03-02:
For context, when using pydrake, all of the `drake::log()` calls from C++ code end up routed to the python `logging` module, not stdout directly:
https://github.com/RobotLocomotion/drake/blob/f47387b29e6a51e15c6374f35fcb0dd594217d7b/bindings/pydrake/common/text_logging_pybind.cc#L57-L67

jwnimmer-tri, 2021-03-02:
And yes, I agree with not binding `spdlog` API into pydrake.  Pure python code should call into the `logging` module.

We can invent some `main()` sugar if we want to make it easy to configure log levels on the command line, but I think it's pretty easy already even without that.

jwnimmer-tri, 2023-06-08:
We have https://drake.mit.edu/pydrake/pydrake.common.html#pydrake.common.configure_logging now, which makes the timestamps / log formatting not be as sucky as Python's default formatting.

I'm not sure what else we should do here.

Are we looking for a Drake style guide amendment that says "Yes, please use `import logging` when logging from Python."?

EricCousineau-TRI, 2023-06-08:
I think what we have now suffices - thanks!

closed 2023-06-08 by EricCousineau-TRI