RobotLocomotion/drake #14727: geometry added to "world" body no longer appears (gets parsed?)

RussTedrake, 2021-03-02:
Previously, I had written [cartpole.urdf](https://github.com/RussTedrake/underactuated/blob/master/underactuated/models/cartpole.urdf) to include a ground via
```
  <link name="world">
    <visual>
      <origin xyz="0 0 -5" rpy="0 0 0" />
      <geometry>
        <box size="1000 1000 10" />
      </geometry>
      <material>
        <color rgba="0.93 .74 .4 1" />
      </material>
    </visual>
  </link>
```
colab repro using meshcat is [here](https://colab.research.google.com/github/RussTedrake/underactuated/blob/master/acrobot.ipynb)

Expected: the ground should appear in the visualizer.  

Actual: no ground is visible.

I initially suspected `MeshCatVisualizer`, but then a related example (see ballbot with floating base) which is runnable in that same [notebook](https://colab.research.google.com/github/RussTedrake/underactuated/blob/master/acrobot.ipynb).  If you use this same approach to add collision geometry for the ground, the robot will fall right through.  So now I'm suspicious that the change happened at the parsing level.

Changing the ballbot example to 
```
  <link name="ground">
    <visual>
      <origin xyz="0 0 -5" rpy="0 0 0" />
      <geometry>
        <box size="1000 1000 10" />
      </geometry>
      <material>
        <color rgba="0.93 .74 .4 1" />
      </material>
    </visual>
  </link>

  <joint name="ground_weld" type="fixed">
    <parent link="world" />
    <child link="ground" />
  </joint>
```
resolves the issue.  I'll do the same with my cartpole.urdf if necessary.  Just want to make sure this wasn't a bug that was introduced, and we agree on the correct behavior.

I haven't looked into it more yet, but @SeanCurtis-TRI recommended I open this issue.

SeanCurtis-TRI, 2021-03-02:
To the best of my knowledge there has been no active effort to disable that behavior. I'll investigate.

sammy-tri, 2021-03-02:
This has been the behavior since I converted the RBT-based URDF parser for MBP in #9698.   AFAICT this was also the behavior for the RBT parser: https://github.com/RobotLocomotion/drake/blob/90c1c1d7ee462e6298386b5065e259a8bd79baba/attic/multibody/parsers/urdf_parser.cc#L504

amcastro-tri, 2021-03-04:
AFAICT we do not parse geometry for the world, [see here](https://github.com/RobotLocomotion/drake/blob/452a3433a5990a76746287fe41a083901af07356/multibody/parsing/detail_urdf_parser.cc#L116). Where `kWorldName=world` [here](https://github.com/RobotLocomotion/drake/blob/452a3433a5990a76746287fe41a083901af07356/multibody/parsing/detail_urdf_parser.cc#L42).

SeanCurtis-TRI, 2021-03-04:
So, the fundamental question is, should we change the behavior? Should we be able to declare anchored geometries by declaring a "world" link?

Pros:
 - It would definitely be convenient.
Cons:
 - It introduces a link that may appear to share meaningful syntax and semantics the others, but doesn't actually.

If we were to add this functionality, what are the implications?

Thoughts?

sherm1, 2021-03-04:
I think the convenience and principal-of-least-astonishment favor recognition of a "world" link even though the interpretation is "add stuff to this link" rather than "define a new link". If we allow it we should reject attempts to specify its mass properties though!

RussTedrake, 2021-04-30:
the "principle of least astonishment" is ringing true.  i've had this confuse a bunch of students working on their projects.

SeanCurtis-TRI, 2021-09-09:
@sherm1 What form of "rejection" are you thinking (if someone tries to specify mass properties on "world"? Ignore, warn, or throw?

sherm1, 2021-09-09:
@SeanCurtis-TRI Quietly ignoring specified world mass properties doesn't seem good because the user is likely making some kind of conceptual error. But a warning that they were ignored would be adequate assuming that's likely to be seen, likely better than a hard throw. A possible use case would be working on just part of an sdf by modifying some intermediate body temporarily to be "world" with the intent of making it immobile. Leaving the mass props there could make sense so that it would be easy to change back to the real body later. In that case a throw would be annoying.

SeanCurtis-TRI, 2021-09-13:
@EricCousineau-TRI As my go-to SDF expert, what are your thoughts on doing something similar for sdf? My understanding of SDF includes the following:

  - SDF has a `<world>` tag specifically (which most definitely does not take geometries)
  - Declaring a joint with the world as parent can be done by specifying "world" as parent (or nothing at all).

So, what is the advisibility of allowing the declaration of a "world" *link* that admits the possibility of assigning geometries?

closed 2021-09-13 by SeanCurtis-TRI

EricCousineau-TRI, 2021-09-14:
At present, SDFormat will outright reject adding `//link/@name="world"`:
https://github.com/ignitionrobotics/sdformat/blob/82cb977b/test/sdf/model_invalid_reserved_names.sdf

So in the scope of this issue, so this should at least obey the "principle of least astonishment".
For now, I'd advise against adding a "world" link declaration in SDFormat. That sound OK?

FWIW: To add to your above options, SDFormat also has `//model/@static`, supported in Drake under limited circumstances:
https://github.com/RobotLocomotion/drake/blob/v0.33.0/multibody/parsing/test/detail_sdf_parser_test.cc#L475-L646
So you still have to add in effectively "fake" links, but you can avoid adding multiple welds.

SeanCurtis-TRI, 2021-09-14:
@EricCousineau-TRI 

So, `//link@name="world"` is outright bad.

Let me rephrase the question differently: can you conceive of an SDF-compatible method for easily declaring geometries affixed to the world body?  One in which we don't create new bodies, but can concsiely indicate that we want some visual and collision geometries affixed to the world?

EricCousineau-TRI, 2021-09-16:
At the moment, not one whose benefit outweighs the cost. IMO, creating new bodies seems OK enough, at least at present.

commit 3e70ef3c (2021-09-13): [parsing] Allow assigning geometries to "world" in URDF (#15763)

In a URDF file, we now allow the declaration of a link named "world". We
also allow the assignment of geometry to that link (which registers
geometry to MultibodyPlant's world body (and as anchored geometry in
SceneGraph).

(Opportunistically cleans up a related typo in parser.h.)

Closes #14727