RobotLocomotion/drake #14739: Bazel Build error : pybind11:generate_pybind_coverage

KarimHassanieh, 2021-03-04:
Ive been trying to compile using Bazel build however I keep getting the below error is there anyway around it?
Note: The failure of target //tools/workspace/pybind11:generate_pybind_coverage (with exit code 1) may have been caused by the fact that it is running under Python 3 instead of Python 2. Examine the error to determine if that appears to be the problem. Since this target is built in the host configuration, the only way to change its version is to set --host_force_python=PY2, which affects the entire build.

jwnimmer-tri, 2021-03-04:
Please post the entire error message, and also the answers to the [helpful information](https://drake.mit.edu/getting_help.html#helpful-information) questions (OS, compiler, etc).

KarimHassanieh, 2021-03-04:
OS : Ubuntu 20.04 
Build system : Bazel Build label: 4.0.0
Compiler : GCC 9.3 
Below is the error message : 
INFO: Build option --host_force_python has changed, discarding analysis cache.
INFO: Analyzed 8155 targets (86 packages loaded, 31187 targets configured).
INFO: Found 8155 targets...
ERROR: /home/karim/Desktop/Projects/ROBOTICS/Robot_Control/drake/bindings/pydrake/BUILD.bazel:578:25: Action bindings/pydrake/file_coverage.csv failed: (Exit 1): generate_pybind_coverage failed: error executing command bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage '--file_coverage=bazel-out/k8-opt/bin/bindings/pydrake/file_coverage.csv' ... (remaining 57 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox generate_pybind_coverage failed: error executing command bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage '--file_coverage=bazel-out/k8-opt/bin/bindings/pydrake/file_coverage.csv' ... (remaining 57 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
libclang: crash detected during parsing: {
  'source_filename' : 'bindings/pydrake/autodiffutils_py.cc'
  'command_line_args' : ['clang', '-std=c++17'],
  'unsaved_files' : [],
  'options' : 0,
}
Traceback (most recent call last):
  File "/home/karim/.cache/bazel/_bazel_karim/7a8d1755166c8b009fd8b0d194906e3a/sandbox/linux-sandbox/753/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage.runfiles/drake/tools/workspace/pybind11/generate_pybind_coverage.py", line 44, in <module>
    main()
  File "/home/karim/.cache/bazel/_bazel_karim/7a8d1755166c8b009fd8b0d194906e3a/sandbox/linux-sandbox/753/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage.runfiles/drake/tools/workspace/pybind11/generate_pybind_coverage.py", line 31, in main
    pybind_strings = get_docstrings_from_bindings(
  File "/home/karim/.cache/bazel/_bazel_karim/7a8d1755166c8b009fd8b0d194906e3a/sandbox/linux-sandbox/753/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage.runfiles/drake/tools/workspace/pybind11/pybind_coverage_libclang_parser.py", line 211, in get_docstrings_from_bindings
    _, final_array = replace_tokens_in_file(f)
  File "/home/karim/.cache/bazel/_bazel_karim/7a8d1755166c8b009fd8b0d194906e3a/sandbox/linux-sandbox/753/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage.runfiles/drake/tools/workspace/pybind11/pybind_coverage_libclang_parser.py", line 182, in replace_tokens_in_file
    token_spellings_original = get_tokens(filename)
  File "/home/karim/.cache/bazel/_bazel_karim/7a8d1755166c8b009fd8b0d194906e3a/sandbox/linux-sandbox/753/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage.runfiles/drake/tools/workspace/pybind11/pybind_coverage_libclang_parser.py", line 164, in get_tokens
    tu = cindex.TranslationUnit.from_source(filename, ["-std=c++17"])
  File "/home/karim/.cache/bazel/_bazel_karim/7a8d1755166c8b009fd8b0d194906e3a/sandbox/linux-sandbox/753/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/generate_pybind_coverage.runfiles/clang_cindex_python3/clang/cindex.py", line 2537, in from_source
    raise TranslationUnitLoadError("Error parsing translation unit.")
clang.cindex.TranslationUnitLoadError: Error parsing translation unit.
----------------
Note: The failure of target //tools/workspace/pybind11:generate_pybind_coverage (with exit code 1) may have been caused by the fact that it is running under Python 3 instead of Python 2. Examine the error to determine if that appears to be the problem. Since this target is built in the host configuration, the only way to change its version is to set --host_force_python=PY2, which affects the entire build.

If this error started occurring in Bazel 0.27 and later, it may be because the Python toolchain now enforces that targets analyzed as PY2 and PY3 run under a Python 2 and Python 3 interpreter, respectively. See https://github.com/bazelbuild/bazel/issues/7899 for more information.
----------------
INFO: Elapsed time: 628.375s, Critical Path: 119.05s
INFO: 1059 processes: 303 internal, 756 linux-sandbox.
FAILED: Build did NOT complete successfully

jwnimmer-tri, 2021-03-04:
Hmm, interesting.  That's a new one to me.

(Aside: the "python2" vs "python3" warning text is a side-effect of error reporting; this is not a python version problem.)

The `libclang: crash detected during parsing` is interesting, though -- that's where things start to go wrong.

Is there any chance that your machine is running out of RAM?  If so, try creating a file named `/home/karim/Desktop/Projects/ROBOTICS/Robot_Control/drake/user.bazelrc` with a single line `build --jobs=1` and see if that helps.  (It will force the build to run only one command at a time.)

KarimHassanieh, 2021-03-04:
Not really I tried with single linke `build --jobs=1` still same error 
Not sure whats causing this, maybe its irrelevant but I am using a  Nvidia display driver 460 not the traditional nouvea display driver

jwnimmer-tri, 2021-03-04:
To check, you've already run `sudo ./setup/ubuntu/install_prereqs.sh`, correct?

Is either `$CC` or `$CXX` set in your shell environment (and if so, to what)?

KarimHassanieh, 2021-03-05:
I did run `sudo ./setup/ubuntu/install_prereqs.sh` so thats taken care of 
`$CC` is part of the enviroment 
```
Using built-in specs.
COLLECT_GCC=cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.3.0-17ubuntu1~20.04' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-HskZEa/gcc-9-9.3.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04) 
```

jwnimmer-tri, 2021-03-05:
Given that Drake has continuous integration builds that test this exact configuration (Ubuntu 20.04, GCC 9.3), there must be something else present in your environment that differs from our default builds.  I'm going to need more information to try to reproduce your problem.

What revision (git commit hash) of Drake are you using)?

What is the exact command line you are running to perform the build?

What is the output of `env` (a dump of all of your environment variables)?

What does `which gcc` report?

What does `file /usr/bin/gcc` report?

What does `file /usr/bin/gcc-9` report?

KarimHassanieh, 2021-03-05:
So below is the feedback that I have : 

What revision (git commit hash) of Drake are you using)?
So when I type `git log` i get the following 
```
commit 85deed36898871582409e1afcbea788afec03644 (HEAD -> master, origin/nightly-release, origin/master, origin/HEAD)
Author: Hongkai Dai <daihongkai@gmail.com>
Date:   Thu Mar 4 18:43:09 2021 -0800

    Add CLP to drake build. (#14729)
    
    * Add CLP to drake build.

commit 3d3dad3e7ffc7d02b31fb00fb58f69c9d8626267
Author: xuchenhan-tri <68254008+xuchenhan-tri@users.noreply.github.com>
Date:   Thu Mar 4 13:08:19 2021 -0800

    [FEM] Add mesh utilities to support FEM demo (#14734)
    
    * Add mesh utilities to support FEM demo
    
    In particular, add the ability to create a box volume mesh that is
    discretized with the diamond cubic pattern.

commit 8ad60d1da9509ed50bc99919e9c1ca7f01ea7272
Author: Rick Cory <rick.cory@tri.global>
Date:   Thu Mar 4 11:41:47 2021 -0500

    Updates MBP::AddJointActuator() documentation to alert users of unused effort limit argument. (#14736)

commit 2b6ceaf8b28035dfe1f368bcd5019387c213795c
Author: Jeremy Nimmer <jeremy.nimmer@tri.global>
Date:   Thu Mar 4 07:11:15 2021 -0800

    doc: Fix 29 nits on new Jekyll website (#14737)
    
    * doc: Fix 29 nits on new Jekyll website

commit 452a3433a5990a76746287fe41a083901af07356
Author: Rick Cory <rick.cory@tri.global>
Date:   Wed Mar 3 18:34:33 2021 -0500

    Adds LCM period argument to AllegroCommandReceiver's constructor. (#14732)
    
    * Adds lcm period argument to AllegroCommandReceiver's constructor.

commit 938c400950e00a5ef5633a26febc3ef575aace41
Author: xuchenhan-tri <68254008+xuchenhan-tri@users.noreply.github.com>
Date:   Wed Mar 3 13:26:53 2021 -0800

    [FEM] Improve API for FemStateBase (#14723)
    
    * Improve API for FemStateBase
    
    - Move all state related API from FemState to FemStateBase for easier
    access. FemState now only handles state-dependent quantities whereas
    FemStateBase handles all state quantities.
:
```

What is the exact command line you are running to perform the build?
I am using this command line in the drake directory `bazel build //...`

What is the output of env (a dump of all of your environment variables)?
yes I receive the following : 
```
SHELL=/bin/bash
ROS_VERSION=1
SESSION_MANAGER=local/karimh89:@/tmp/.ICE-unix/1431,unix/karimh89:/tmp/.ICE-unix/1431
WM_COMPILER=Gcc
[... 110 lines cut ...]
FOAM_SOLVERS=/opt/openfoam8/applications/solvers
WM_PROJECT_DIR=/opt/openfoam8
WM_COMPILER_TYPE=system
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
FOAM_USER_APPBIN=/home/karim/OpenFOAM/karim-8/platforms/linux64GccDPInt32Opt/bin
PATH=/home/karim/Nek5000/bin:/opt/ros/noetic/bin:/opt/ThirdParty-8/platforms/linux64Gcc/gperftools-svn/bin:/opt/paraviewopenfoam56/bin:/home/karim/OpenFOAM/karim-8/platforms/linux64GccDPInt32Opt/bin:/opt/site/8/platforms/linux64GccDPInt32Opt/bin:/opt/openfoam8/platforms/linux64GccDPInt32Opt/bin:/opt/openfoam8/bin:/opt/openfoam8/wmake:/home/karim/FDS/FDS6/smvbin:/home/karim/FDS/FDS6/bin/INTEL/bin:/home/karim/FDS/FDS6/bin:/home/karim/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/karim/.dotnet/tools:/home/karim/DAKOTA/dakota_installation/bin:/usr/local/bin:/home/karim/Desktop/Projects/CFD/nektar++/build/dist/bin:/usr/local/visit/bin
PV_PLUGIN_PATH=/opt/openfoam8/platforms/linux64GccDPInt32Opt/lib/paraview-5.6
GDMSESSION=ubuntu
WM_COMPILER_LIB_ARCH=64
SU2_HOME=/home/karim/Desktop/Projects/CFD/SU2/SU2
FOAM_SIGFPE=
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic
LC_NUMERIC=en_US.UTF-8
OLDPWD=/home/karim
_=/usr/bin/env
```
What does which gcc report?
I get the following output `/usr/bin/gcc`

What does file /usr/bin/gcc report?
I get the following output  `/usr/bin/gcc: symbolic link to gcc-9`
What does file /usr/bin/gcc-9 report?
I get the following output  `/usr/bin/gcc-9: symbolic link to x86_64-linux-gnu-gcc-9`

jwnimmer-tri, 2021-03-05:
Hmm, that `LD_LIBRARY_PATH` is huge.  Can you try disabling that and see if it works?

```
env LD_LIBRARY_PATH= bazel build //...
```

KarimHassanieh, 2021-03-06:
I did `export LD_LIBRARY_PATH=""` and then ran  `bazel build // ...` however got the same error  :(

jwnimmer-tri, 2021-03-06:
Here's a patch to print which library we're trying to use:

```patch
--- a/tools/workspace/pybind11/pybind_coverage_libclang_parser.py
+++ b/tools/workspace/pybind11/pybind_coverage_libclang_parser.py
@@ -161,6 +161,7 @@ def write_to_file(arr, file_name):
 
 
 def get_tokens(filename):
+    print(f"Using parsing library {cindex.conf.lib._name}")
     tu = cindex.TranslationUnit.from_source(filename, ["-std=c++17"])
     FILE = tu.get_file(bytes(filename, 'utf8'))
 
```

Can you add that and see what it prints?  When operating correctly, I see this output:
```
Using parsing library /usr/lib/x86_64-linux-gnu/libclang-9.so`
```
---

After that, here's how to check for any stray library dependencies:
```
ldd /usr/lib/x86_64-linux-gnu/libclang-9.so | sed -e 's/ (.*//'
```

When operating correctly, I see this output:
```
	linux-vdso.so.1
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2
	libLLVM-9.so.1 => /lib/x86_64-linux-gnu/libLLVM-9.so.1
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6
	/lib64/ld-linux-x86-64.so.2
	libffi.so.7 => /lib/x86_64-linux-gnu/libffi.so.7
	libedit.so.2 => /lib/x86_64-linux-gnu/libedit.so.2
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0
```

---

In both cases, I'm wondering if you might have software installed into `/usr/local` that is getting in the way.

Another option would be to try the Drake build in a clean terminal (one that doesn't have any ROS or FOAM variables set).  Those settings are all _supposed_ to be nixed and ignored automatically by Bazel during the build, but it's possible something is leaking through.

---

On another note, while I _am_ curious to figure out why the build is broken (and fix it in Drake proper), I'll also note that `bazel build //...` is building the _entire_ project, and you probably don't particularly need the target that is failing, so skipping it might be a work-around.

For example, if you only need pydrake you can `bazel build //bindings/pydrake`, or for only the kuka demo, you can `bazel build //examples/kuka_iiwa_arm/...`.

KarimHassanieh, 2021-03-06:
I recevied the same output for parsing library 
`Using parsing library /usr/lib/x86_64-linux-gnu/libclang-9.so`
and when trying `ldd /usr/lib/x86_64-linux-gnu/libclang-9.so | sed -e 's/ (.*//'`
its the same output as well 
```
	linux-vdso.so.1
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2
	libLLVM-9.so.1 => /lib/x86_64-linux-gnu/libLLVM-9.so.1
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6
	/lib64/ld-linux-x86-64.so.2
	libffi.so.7 => /lib/x86_64-linux-gnu/libffi.so.7
	libedit.so.2 => /lib/x86_64-linux-gnu/libedit.so.2
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0
```

this is becoming difficult to resolve

jwnimmer-tri, 2021-03-06:
At this point, I am not yet sure what to try next.  Skipping that problematic build target (by not building `//...`, per the above) should get you unstuck for now.

jwnimmer-tri, 2021-03-08:
As a workaround, here's a patch you could apply to disable the problematic target:

```patch
--- a/bindings/pydrake/BUILD.bazel
+++ b/bindings/pydrake/BUILD.bazel
@@ -580,6 +580,7 @@
     class_coverage = ":class_coverage.csv",
     file_coverage = ":file_coverage.csv",
     pybind_coverage_data = [":pybind_coverage_data"],
+    tags = ["manual"],
     xml_docstrings = ":documentation_pybind.xml",
 )
 
```

Given that, and my inability to reproduce the problem, I'm going to close this out for now.

If you find out anything new, please post back here.  I do want to fix this if its a problem in Drake, but can't really do anything until I can reproduce the problem.

closed 2021-03-08 by jwnimmer-tri

jamiesnape, 2021-03-08:
> (Aside: the "python2" vs "python3" warning text is a side-effect of error reporting; this is not a python version problem.)

We should suppress that at this stage. I will open an issue. I nuked it in Russ's repo's somehow.