RobotLocomotion/drake #14745: Nuke almost certainly spurious "failure of target" caused by "running under Python 3" warning from Bazel.

jamiesnape, 2021-03-08:
Specifically `"The failure of target ..." may have been caused by the fact that it is running under Python 3 instead of Python 2.` I fixed the issue in `underactuated` and `manipulation`, so I think it is one-liner somewhere.

jamiesnape, 2021-03-08:
Russ's targets always explicitly set `srcs_version = "PY3"` and explicitly set `python_version = "PY3"` where appropriate.

jamiesnape, 2021-03-08:
* https://github.com/RussTedrake/htmlbook/blob/26cb5cd9184f6cba1ed42b750ad81ed6497a6769/tools/python/defs.bzl#L18
* https://github.com/RussTedrake/htmlbook/blob/26cb5cd9184f6cba1ed42b750ad81ed6497a6769/tools/python/defs.bzl#L29

jwnimmer-tri, 2021-12-20:
I'm calling this good enough for now.  I only fixed the `drake_py_...` targets.  Any `py_...` targets from externals might still experience this warning, but _shrug_.

closed 2021-12-20 by jwnimmer-tri

jwnimmer-tri, 2022-01-07:
It turns out that mkdoc frequently spews about this as well, so I've done another PR (above) to fix that as well.

PR #16259 (2021-12-20): [tools] Set Python version to 3 explicitly
This removes spurious warnings from Bazel when other things go wrong.

Closes #14745.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16259)