RobotLocomotion/drake #14747: Support continuous joints from SDFormat

calderpg-tri, 2021-03-08:
SDFormat defines "continuous" as a supported joint type since 1.6; however, our SDF parser does not support this joint type and throws a "Joint type not supported" exception when parsing a SDF file containing such a joint.

calderpg-tri, 2021-07-12:
I believe I was writing a new model - specifically, using the initial version of the mecanum-chassis model generator - with wheels attached via continuous joints.

jwnimmer-tri, 2021-12-01:
Does MbP already offer a compatible joint type?  In other words, is this only a parsing issue, or is it also a missing feature issue with the plant itself?

calderpg-tri, 2021-12-01:
This is just a parsing issue, presuming that continuous joint behavior between URDF and SDFormat should be the same, `continuous` joints are already supported from URDFs.

jwnimmer-tri, 2022-04-26:
@EricCousineau-TRI any chance to get this onto the OpenRobotics dashboard?  It's a simple Drake parsing fix (+ test case).

Otherwise maybe ping @xuchenhan-tri for MbP hitlist board.

EricCousineau-TRI, 2022-04-27:
I could put it on the board - is it OK if it gets done w/in next ~3 weeks?

jwnimmer-tri, 2022-04-27:
Yes, thanks!  I'd be satisfied with any non-zero velocity towards resolving the issue.

RussTedrake, 2022-09-20:
It's been a lot more than three weeks.  This is a trivial thing to fix, and pretty surprising that we don't support this.  (Our FIRST robotics team just hit it!)

xuchenhan-tri, 2022-09-20:
I can pick up this ticket if @EricCousineau-TRI has other priorities.

EricCousineau-TRI, 2022-09-21:
My apologies and most definitely my bad. I had queued it on the SDFormat board, but did not discuss it w/ OSRC team in that following meeting and it feel to the wayside.

Xuchen - if you're handling it, then yes, that'd be great! Apologies you had to pick up my slack.

Otherwise, will bring up in OSRC meeting on Thursday.
Will still discuss assigning priorities for the SDFormat project as well.

xuchenhan-tri, 2022-09-21:
No worries! Happy to learn more about SDFormat.

PR #17956 (2022-09-22): [parsing] Support parsing continuous joint in SDFormat
Closes #14747.

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17956)
closed 2022-09-22 by amcastro-tri