RobotLocomotion/drake #14770: Website main page layout is bad on narrow/mobile browsers

ggould-tri, 2021-03-11:
To reproduce this, visit drake.mit.edu and narrow your browser window until the top menu vanishes and is replaced by the hamburger icon.  Now try to click on the install button.  Either nothing will happen or you'll go to a random page.  This is because the logic to show and hide the menu when you click the hamburger is merely hiding the menu but leaving it taking up room in the layout and leaving it clickable.

The issue appears to be that the css logic in `main.css` to show/hide the menu is using `visibility: hidden` (which makes the menu invisible) rather than `display:none` (which makes the menu not exist at all for rendering purposes.).  If you use devtools to set `display:none` on the `site-menu` element, it all seems to work correctly.

cc: @jamiesnape

ggould-tri, 2021-03-11:
Although see https://css-tricks.com/places-its-tempting-to-use-display-none-but-dont/ it may be an accessibility error to do exactly this; it might be more correct to zero out the size of the menu.  An actual web designer would need to make a judgment here.

jamiesnape, 2021-03-11:
You are certainly supposed to keep an accessible alternative on page, but we have a pretty bad accessibility issue at present.

jamiesnape, 2021-03-11:
I guess try
```
.visuallyhidden { 
  position: absolute; 
  overflow: hidden; 
  clip: rect(0 0 0 0); 
  height: 1px;
  width: 1px; 
  margin: -1px
  padding: 0;
  border: 0; 
}
```

`clip: rect(1px 1px 1px 1px);` is another option, I think, if a zero area clip misbehaves.

jamiesnape, 2021-03-11:
There is some other CSS weirdness when you look at the code, but that is probably another issue.

jwnimmer-tri, 2021-04-09:
I actually just hit this today in my desktop browser (when using a tiled window manager) -- the https://drake.mit.edu/release_playbook.html could not be copy-pasted from because they were hidden by some overlay, when the window is too narrow.  (At least, I _think_ it was the wame root cause as this issue.)

jamiesnape, 2021-04-09:
Yes, just tried with a narrow desktop Safari window. Same problem. The active div is the navigation one if you try to select anything.

EricCousineau-TRI, 2021-04-09:
Confirmed in Firefox, 87.0. (For the future, can y'all please incl. video and browser version, 'cause it's suuuuuuper easy to do that?)
See attached video.

https://user-images.githubusercontent.com/26719449/114249161-cfaf8580-9967-11eb-95aa-816c2a414569.mp4

jwnimmer-tri, 2021-04-09:
> can y'all please incl. video 

No.

> and browser version,

Good point.  I was using `87.0+build3-0ubuntu0.18.04.2`.

jamiesnape, 2021-04-12:
Safari 14.0.3, but this is unlikely to be one of those browser quirks, unfortunately. It is quite old CSS functionality. If someone chimes in with the same problem on Chrome/Chromium, that pretty much has all the engines covered anyway (except for the random IE users).

FWIW The page works great on Lynx...

jwnimmer-tri, 2021-11-10:
This bit me a couple days ago on my phone.

@ToffeeAlbina-TRI is there any way we can hire a JS/CSS contractor to fix this for us?  Maybe even the same company, if we already have them in our vendor system.

jwnimmer-tri, 2021-11-10:
Or maybe +@EvelynDixon-TRI can help facilitate that.

ToffeeAlbina-TRI, 2021-11-17:
We should reach out to the same company we used before as they are already in our system.

Perhaps Evelyn would be willing to push this forward?  Also, it would be helpful to point Evelyn to other issues concerning the website and maybe we can get them fixed/implemented too.

jwnimmer-tri, 2021-11-17:
Sure, that's  #14757, now also assigned.

PR #16665 (2022-02-25): [doc] Website enhancements
Specifically:
- Add search box to landing page.
- Add "Learn More" to landing page.
- Fix invisible menu bar shadowing page elements.
- Clarify landing page display of library / API boxes.
- Fix italics to actually work.

Towards #14757.
Closes #14770.

FYI https://drake.mit.edu/documentation_instructions.html#previewing-changes to try it locally.

The merge plan is to squash everything, but for now I'm leaving separate commits to help make the history clear during review.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16665)
closed 2022-02-25 by jwnimmer-tri