RobotLocomotion/drake #14779: diff ik: Computation may not stable as a fixed-point?

EricCousineau-TRI, 2021-03-15:
Confirmed what we (@BenBurchfiel-TRI, @siyuanfeng-tri, and I) have speculated on.

I've got some code that maps from spatial velocties to joint velocities, and had a vague notion of a fixed point (either marginally stable, or within machine tolerance).

This shows that this isn't true, at least for certain configurations:
https://github.com/EricCousineau-TRI/repro/blob/8a69830a42252f9d5324a18d4c60499c4e98be4d/drake_stuff/diff_ik_fixed_point_issue/diff_ik_fixed_point_repro.ipynb

Relevant output from notebook:
```
q[0]: -2.897     <=  0.041     <=  2.897       |   dist:  2.856    
q[1]: -1.763     <= -0.715     <=  1.763       |   dist:  1.048    
q[2]: -2.897     <= -0.092     <=  2.897       |   dist:  2.805    
q[3]: -3.072     <= -2.790     <= -0.070       |   dist:  0.282    
q[4]: -2.897     <=  0.011     <=  2.897       |   dist:  2.886    
q[5]: -0.018     <=  2.900     <=  3.752       |   dist:  0.853    
q[6]: -2.897     <= -0.800     <=  2.897       |   dist:  2.097    
v[0]: [0 0 0 0 0 0 0]

[ Use diff IK ]
V_WGdes[0]: [0 0 0 0 0 0]
v[1]:       [-2.59e-23 -1.7e-22 6.29e-23 -2.57e-23 1.07e-23 -2e-22 4.96e-23]
Δv[1]:      [-2.59e-23 -1.7e-22 6.29e-23 -2.57e-23 1.07e-23 -2e-22 4.96e-23]
---
V_WGdes[1]: [1.9e-23 5.45e-23 -1.77e-24 -3.35e-23 1.6e-23 2.33e-23]
v[2]:       [-0.00188 -0.0141 0.00486 -0.00213 0.000218 -0.0166 0.00459]
Δv[2]:      [-0.00188 -0.0141 0.00486 -0.00213 0.000218 -0.0166 0.00459]
---
V_WGdes[2]: [0.00157 0.00452 -0.000147 -0.00278 0.00133 0.00193]
v[3]:       [-0.00376 -0.0282 0.00972 -0.00426 0.000436 -0.0331 0.00918]
Δv[3]:      [-0.00188 -0.0141 0.00486 -0.00213 0.000218 -0.0165 0.00459]
---
V_WGdes[3]: [0.00315 0.00904 -0.000294 -0.00555 0.00266 0.00386]
v[4]:       [-0.00564 -0.0423 0.0146 -0.00639 0.000654 -0.0496 0.0138]
Δv[4]:      [-0.00188 -0.0141 0.00485 -0.00213 0.000218 -0.0165 0.00458]
---
V_WGdes[4]: [0.00472 0.0135 -0.00044 -0.00832 0.00399 0.00578]
v[5]:       [-0.00751 -0.0563 0.0194 -0.00851 0.000872 -0.0661 0.0183]
Δv[5]:      [-0.00187 -0.014 0.00484 -0.00212 0.000217 -0.0165 0.00457]
---

[ Use lstsq ]
V_WGdes[1]: [0 0 0 0 0 0]
v[2]:       [0 0 0 0 0 0 0]
Δv[2]:      [0 0 0 0 0 0 0]
---
V_WGdes[2]: [0 0 0 0 0 0]
v[3]:       [0 0 0 0 0 0 0]
Δv[3]:      [0 0 0 0 0 0 0]
---
V_WGdes[3]: [0 0 0 0 0 0]
v[4]:       [0 0 0 0 0 0 0]
Δv[4]:      [0 0 0 0 0 0 0]
---
V_WGdes[4]: [0 0 0 0 0 0]
v[5]:       [0 0 0 0 0 0 0]
Δv[5]:      [0 0 0 0 0 0 0]
---
```

EricCousineau-TRI, 2021-03-15:
The weirdest thing about this (IMO) is that `Δv` is constant (meaning "accelerating" across re-mapping)... dunno what that means yet.

EricCousineau-TRI, 2021-03-15:
(Not important for now 'cause I can workaround it)

hongkai-dai, 2021-03-15:
If you switch to a different solver, like gurobi/Mosek, does the solution change?

EricCousineau-TRI, 2021-03-15:
Later this week (mebbe Fri?), I'll def. try out different solvers - thanks!
Ideally, I can just re-implement the more direct version (`DoDifferentialInverseKinematics(q, v, V, J, param)`.

hongkai-dai, 2021-03-15:
Sounds good. It is possible by this Friday I will merge another solver (CLP) which can solve QP with a high precision. So if the culprit is that OSQP (which I think is the default solver) has low accuracy, we could potentially switch to CLP.

siyuanfeng-tri, 2021-03-15:
what happens if you try a nonzero desired V?

RussTedrake, 2021-09-24:
Possibly related to #9773.  At least, I'd prefer to scrutinize the behavior of a formulation that we think is correct.