RobotLocomotion/drake #14782: InverseKinematics program is not thread-safe

brawner, 2021-03-15:
I was trying to run several IK solvers in parallel to find a valid solution faster than iterating in a single loop. However, I believe that the program created with a position and orientation constraint cannot be shared across threads even though MathematicalProgram is taken as a const-ref when running `Solve()`. 

Is there a recommended manner of running this concurrently? I have a separate example which requires creating a new IK program in each thread, but performance suffers quite a bit. 

Unfortunately, calling `ik.prog().Clone()`  doesn't seem to work either.

I'm using nightly binary release from 3/11/2021, Ubuntu Focal (20.04). 

Example of a google benchmark, which illustrates the issue.

```
static void BM_inverse_kinematics_not_threadsafe(benchmark::State& state) {
    const auto [plant, model_instance, lower_limits, upper_limits] = load_plant();

    constexpr size_t num_threads = 32;
    std::atomic<int> num_successes = 0;

    drake::multibody::InverseKinematics ik(*plant, true);
    add_random_constraints(*plant, model_instance, lower_limits, upper_limits, ik);
    for (auto _ : state) {
        std::array<std::thread, num_threads> threads;
        for (int i = 0; i < num_threads; ++i) {
            threads[i] = std::thread{[&]() {
                const auto random_configuration =
                    sample_random_configuration(lower_limits, upper_limits);
                const auto result = Solve(ik.prog(), random_configuration);
                if (result.is_success()) {
                    num_successes++;
                }
            }};
        }

        for (auto& thread : threads) {
            thread.join();
        }
    }
}
```

It terminates with:

```
terminate called after throwing an instance of 'std::logic_error'
  what():  CacheEntryValue(::_:H_PB_W(q))::GetMutableAbstractValueOrThrow(): the current value is already up to date.
Aborted (core dumped)
```

Possible related issue: https://github.com/RobotLocomotion/drake/issues/10320

Full code available at:
https://gist.github.com/brawner/1df81eac5f4bf2137e9cff7552602ad6

sherm1, 2021-03-16:
Assigning to @ToffeeAlbina-TRI for disposition.

calderpg-tri, 2021-04-01:
We do similar parallel IK operations at TRI. In our versions, we used to maintain a pool of per-thread contexts (for parallelism via OpenMP) and did something like
```
vector<VectorXd> guesses;
...
#pragma omp parallel for
for (size_t num_guess = 0; num_guess < guesses.size(); num_guess++) {
  // Get per-thread context
  // Make constraint
  // Try to solve IK
}
```
but to simplify this and avoid accidental sharing of these contexts we switched to creating a new context for each constraint. 

Either approach will work, just with a per-thread pool you need to be careful to never share one of the contexts (for example, if combining parallelism with `std::thread`/`std::async` and OpenMP).