RobotLocomotion/drake #14796: cmake build fails on commit b562211

ryanelandt, 2021-03-18:
When building on Ubuntu 18 with CMake, I get the following error:

(base) ryan@ryan-desktop:~/DR3/drake$ cmake -DCMAKE_BUILD_TYPE="Release" && make -j4
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Bazel: /usr/bin/bazel (found suitable version "4.0.0", minimum required is "3.0") 
CMake Error at CMakeLists.txt:229 (message):
  Could NOT find the python3 executable

-- Configuring incomplete, errors occurred!
See also "/home/ryan/DR3/drake/CMakeFiles/CMakeOutput.log".

jwnimmer-tri, 2021-03-18:
Looks like  6903f7bfa5e716ce4ed47d91067c1abb273db3df broke this by removing `set(FIND_PROGRAM_PYTHON3_PATHS /usr/bin)`, and it wasn't caught in CI because our CI uses a Kitware build of CMake, not the Ubuntu build of CMake, so we test a version newer than the one we declare support for.

I'll work up a patch.

closed 2021-03-18 by jwnimmer-tri

ryanelandt, 2021-03-18:
You're a build-saver @jwnimmer-tri

PR #14798 (2021-03-18): cmake: Fix python3 search for CMake 3.10 (Ubuntu 18.04)
This was broken in #14605.

Closes #14796.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14798)