RobotLocomotion/drake #14802: Undesired heap allocation in DiscreteValues::SetFrom(.)

edrumwri, 2021-03-19:
LimitMalloc points to unaryExpr(.) on the line below as the culprit.

https://github.com/RobotLocomotion/drake/blob/dd18fde30250c0a7e89a436cd3f73e4c6ce948d5/systems/framework/discrete_values.h#L155

SeanCurtis-TRI, 2021-03-22:
@sherm1 I'm passing it to you for any further disposition, as you see fit.

sherm1, 2021-03-22:
@edrumwri is this another case of heap allocation for creating a functor? In that case just doing the U->T conversion elementwise should fix it?

sherm1, 2021-03-22:
(Assigning to Evan for now)

edrumwri, 2021-03-25:
> @edrumwri is this another case of heap allocation for creating a functor? In that case just doing the U->T conversion elementwise should fix it?

I'm confused as to why you suggested this might be related to creating a functor. And it does appear to be doing the conversion element-wise. The solution would seem to me to be to introduce a second SetFrom(.) function:

```
void SetFrom(const DiscreteValues& other) {
    DRAKE_THROW_UNLESS(num_groups() == other.num_groups());
    for (int i = 0; i < num_groups(); i++)
      data_[i]->set_value(other.get_vector(i).get_value());
  }
```

edrumwri, 2021-06-02:
@rpoyner-tri it certainly wouldn't hurt though I'll explain my reasoning for why Systems with continuous state have not been on my radar. 

Reiterating from elsewhere, we're using Drake to build our soft real-time perception and control systems. All of our Systems have discrete and/or abstract state; none have continuous state. This is mostly due to the functionality of these systems (e.g., an object recognition loop runs at 8 Hz). So removing heap allocations related to the continuous state parts of Drake won't hold much value for us, though others might appreciate it. I do expect plenty of heap allocations in the integrator codes, FYI.

PR #15118 (2021-06-03): discrete values: Avoid heap in non-converting SetFrom()
Closes: #14802.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15118)
commit dced982f (2021-06-03): discrete values: Avoid heap in non-converting SetFrom() (#15118)

* discrete values: Avoid heap in non-converting SetFrom()

Closes: #14802.
closed 2021-06-03 by rpoyner-tri