RobotLocomotion/drake #14808: DrakeVisualizer displays robot_0, robot_1, etc. instead of meaningful names

edrumwri, 2021-03-22:
It's possible to get meaningful names by expanding the data, but there's no "expand all" feature in Drake Visualizer, so it's painful to find the body of interest when there are many bodies being visualized.

SeanCurtis-TRI, 2021-04-06:
@edrumwri I'm going to take the liberty of hijacking this issue. The very valid initial concern is a subset of a larger concern. Exploring and discovering entities in `drake_visualizer` is a real pain in the can. If you agree, I'll change the name and move this up into the summary.

## Enumeration of issues
- Every object loaded into `drake_visualizer` from Drake ends up on a `Robot #` folder, even if it's not a robot. The names are meaningless.
- The frames (typically representing bodies) under the `Robot #` folder can often be painfully verbose. For example, running the four-bar linkage example produces frames with names like `plant::four_bar::C`. In short, there's a great deal of name mangling which may, nevertheless, not be sufficient for the purpose.
- Visualized geometries lose all unique name information and become simply: `[body name] geometry data`. (And if you consider the previous point, this leads mildly obnoxious frame names to become mildly obnoxious geometry names).
- Any geometry affixed to the world frame ends up in `Robot 0` folder under a `world` frame name.

## What we'd like

Imagine a simple scenario where we load a URDF twice. The name of the robot (as defined by the URDF) is "Bob". It has three bodies named "A", "B", and "C". Each of those have a geometry called "collision". We load it twice, producing two model instances (which we name "Bob Left" and "Bob Right" with model instance indices 2 & 3, respectively) We also add a half space to represent the ground which we rigidly affix to the world frame. Currently, this would produce the following hierarchy:

```
  - Robot 0
    - world
      - world geometry data
  - Robot 2
    -  Source_3::Bob Left::A
      - Source_3::Bob Left::A geometry data
    -  Source_3::Bob Left::B
      - Source_3::Bob Left::B geometry data
    -  Source_3::Bob Left::C
      - Source_3::Bob Left::C geometry data 
  - Robot 3
    -  Source_3::Bob Right::A
      - Source_3::Bob Right::A geometry data
    -  Source_3::Bob Right::B
      - Source_3::Bob Right::B geometry data
    -  Source_3::Bob Right::C
      - Source_3::Bob Right::C geometry data 
```

What we'd *like* is:

```
  - world
    - ground
  - Bob Left
    -  A
      - collision
    -  B
      - collision
    -  C
      - collision
  - Bob Right
    -  A
      - collision
    -  B
      - collision
    -  C
      - collision
```

Calling out some key properties
  - Anchored geometries are children of a folder simply called world (at the same level of the hierarchy as model instances).
  - Geometry names in the visualizer are exactly what they are registered -- i.e., the name they have in a URDF/SDF file or provided via the `SceneGraph` API.
  - We don't encode the paths into entity names (e.g., the body name is just a body name and we rely on the actual hierarchy to disambiguate similarly named frames).

## What is it going to take to get there?

This is what is hardest. There are multiple contributors to this behavior. Generally, this is going to require effort applied to multiple entities. But the *crux* of the solution is creating *new* LCM message(s) that contain sufficient information for `drake_visualizer` to do the good thing (and modifications to `drake_visualizer` to make use of that info).

### `SceneGraph` (and ecosystem)
 - Frames are associated with "frame groups" (a number which `MBP` interprets as model instance index). We need to associate __*names*__ with those frame groups.
   - `SG` gets a new API so that sources can declare a name (provide a mechanism to prevent accidental collisions).
   - `MBP` exercises that API to announce names for its model instances (aka frame groups).
 - `DrakeVisualizer` should not automatically [prefix the owning *source* name to frame names](https://github.com/RobotLocomotion/drake/blob/master/geometry/drake_visualizer.cc#L429-L431).
   - However, `drake_visualizer` has the requirement that (link_name, group_name) be a unique identifier. This is not currently enforced in `SG`. It should be enforced to guarantee compatibility. Right now, it'll work as long as the data is well formed, but it will inexplicably misbehave when the invariant is violated.

### `MultibodyPlant`
  - `MBP` currently creates frame names that are a [concatenation of model instance name and frame name](https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/multibody_plant.cc#L244-L256) (assuming its model instance is neither world nor default). Although that code references an issue (#9128), it's not clear how it is relevant. However, given the current state of visualization, the logical inference is that this is the sole avenue by which model instance *names* reach `drake_visualizer`. Once the LCM messages are augmented to include "frame group" names in place of generic "Robot #" names, this need evaporates.

### `drake_visualizer` and LCM messages
  - Pass "robot names" (i.e., strings associated with `SG` frame groups/model instance names) in the lcm message so the "robot folders" can use that name, [instead of "Robot #"](https://github.com/RobotLocomotion/director/blob/master/src/python/director/drakevisualizer.py#L365-L366).
  - We need to include geometry names in lcm messages
    - `drake_visualizer` should use those names rather than [simply appending 'geometry name' to the link name](https://github.com/RobotLocomotion/director/blob/master/src/python/director/drakevisualizer.py#L298-L299).

#### Incidental note

The `drake_visualizer` side of things can be easily resolved by introducing *novel* lcm messages and using a plug-in. In fact, it might be ideal to do this immediately following #14532 and use that as the basis for introducing the new lcm message that subsequently gets evolved to support this effort. By making it a Drake plugin, we get full control of message, authoring, and reading all in the Drake repo.

jwnimmer-tri, 2021-12-14:
@joemasterjohn @edrumwri I propose closing this as a"WONTFIX" per #16215. What do you think?

@edrumwri For as long as we have the `DrakeVisualizer` system in Drake, we'd still be open to accepting PRs that improve it's user experience.  My thesis here is only that tracking the request as a Drake project issue is no longer valuable, since the core team would not plan to work on it.

jwnimmer-tri, 2022-01-27:
We can revisit / re-open if someone speaks up, but so far it sounds like WONTFIX is sufficient.

closed 2022-01-27 by jwnimmer-tri