RobotLocomotion/drake #14814: pydrake: Debug why `assertIs(joint, joint_out)` really failed in plant_test?

EricCousineau-TRI, 2021-03-23:
Relates #14813 (hotfix) #14805 (original PR)

From Slack:
https://drakedevelopers.slack.com/archives/C270MN28G/p1616500538010200

> K, I know that scope change fixes it, so maybe there's a bug somewhere.
> https://github.com/EricCousineau-TRI/drake/tree/feature-pr14813-rca-attempt
> 
> ```$ git log -n 1 --oneline --no-decorate
> 03d4b6a031 try simple
> $ bazel run --run_under='env PYTHONUNBUFFERED=1' //bindings/pydrake/multibody:py/plant_test -- -f TestPlant.test_multibody_add_joint 2>&1 
> ...
> <function TestPlant.test_multibody_add_joint.<locals>.make_noparams_weld_joint at 0x1dc8e1790>
> F
> ======================================================================
> FAIL [0.029s]: test_multibody_add_joint (plant_test.TestPlant)
> Tests joint constructors, `AddJoint`, `AddJointActuator` and
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "{runfiles}/drake/bindings/pydrake/common/test_utilities/numpy_compare.py", line 254, in wrapper
>     check_func(*args, T=float, **kwargs)
>   File "{runfiles}/drake/bindings/pydrake/multibody/test/plant_test.py", line 1586, in test_multibody_add_joint
>     self.assertIs(joint, joint_out)
> AssertionError: <WeldJoint_[float] name='weld' index=0 model_instance=1> is not <WeldJoint_[float] name='weld' index=0 model_instance=1>```

Possible issues:
- `[](...) -> const Joint<T>&` is still tripping up pybind11
- Bad bookkeeping in our fork of pybind1
- Something else

\cc @rpoyner-tri @liangfok

rpoyner-tri, 2021-03-23:
So, without the full detail dive, the upshot is that pybind on mac and ubuntu can lead to different results?

EricCousineau-TRI, 2021-03-23:
Yes. But that's been true for other one-offs, e.g. https://github.com/RobotLocomotion/drake/issues/14686

jwnimmer-tri, 2023-05-22:
Possibly related #19335.

closed 2024-03-21 by jwnimmer-tri