RobotLocomotion/drake #14827: autopybind: Should test current workflows for automated bindings

EricCousineau-TRI, 2021-03-25:
As part of #7889, using #14265

I've chosen the following symbols to genuinely try this workflow out on:

* New bindings:
  * Non-templated:
    * `drake::geometry::render::RenderEngineGlParams`
    * `drake::multibody::StaticEquilibriumConstraint`
  * Templated:
    * `drake::systems::sensors::Accelerometer`
    * `drake::systems::PortSwitch`
* Update bindings
  * Non-templated:
    * `drake::solvers::MixedIntegerBranchAndBound`
    * Templated:
      * `drake::systems::SurfaceMesh`
      * `drake::multibody::MultibodyPlant`

EricCousineau-TRI, 2021-03-25:
FYI @josephsnyder

EricCousineau-TRI, 2021-03-25:
My plan is to make the PR, but will strive to record what was necessary to go from generated bindings to my own liking as well as reviewer acceptance.

EricCousineau-TRI, 2021-03-26:
For `RenderEngineGlParams`, see commits 56700ca7a4 and bde2a425f1 for final products.

@jamiesnape and @josephsnyder I've added opinionated steps that I took here:
https://github.com/RobotLocomotion/drake/blob/56700ca7a4/tools/autopybind/README.md#steps-taken

Can both of you please review my steps here and let me know if there is anything better that I could have done?
I will file an issue about adding an option to suppress warnings, and possibly provide more user-friendly errors if a symbol isn't found in a file.

Given current workflow for a simple case, I would still like to wait until certain parts of the denoising happen before asking anyone else to run this.

EricCousineau-TRI, 2021-03-26:
Tried `drake::multibody::StaticEquilibriumConstraint`. Punted because of weird API (our fault), but there were still pain points w.r.t. autopybind11:
https://github.com/EricCousineau-TRI/drake/blob/f25751c7ca64755e55ade1ae10073937b25aeb5b/tools/autopybind/README.md#for-staticequilibriumconstraint

EricCousineau-TRI, 2021-03-26:
I'm pausing any more of my effort on this until denoising lands.

jwnimmer-tri, 2021-11-10:
Closing out as "unlikely to reach an actionable state within a reasonable timeframe".

closed 2021-11-10 by jwnimmer-tri