RobotLocomotion/drake #14841: MultibodyPlant mutable default parameters don't end up in default contexts

joemasterjohn, 2021-03-30:
Reproducible  branch: https://github.com/joemasterjohn/drake/tree/default_parameter_bug

Running:
`bazel test //multibody/plant:multibody_plant_reflected_inertia_test`

There is a bit of missing machinery for a multibody plant's mutable model parameters to get set in a default context. What's specifically shown in the bug example is a default model parameter (a joint actuator's gear ratio) being mutated after Finalize().
The new default value does not end up in a newly created default context. This is because the model values in `LeafSystem` are allocated and populated during finalize, and have no way to be modified after the fact.

I believe to solve this issue, some things need to be added to various system components:
-  Adding a new method `LeafSystem::SetNumericParameter(const NumericParameterIndex index, const BasicVector<T>& model_vector)`
- Adding new methods `ModelValues::SetVectorModel(int index, std::unique_ptr<BasicVector<T>> model_vector)` and `ModelValues::SetModel(int index, std::unique_ptr<AbstractValue> model_value)`

This would allow the multibody element to modify the corresponding model value for its parameters after finalize.

cc: @sherm1

sherm1, 2021-03-30:
Per f2f, consider `VectorBlock = LeafSystem::GetMutableDefaultNumericParameter()` to
- prevent resizing of parameter from its Declared shape
- avoid heap allocations

joemasterjohn, 2021-03-31:
Overriding SetDefaultParameters was the other solution Sherm and I discussed. I think just having conflicting values between the base LeafSystem's members and MultibodyPlant's corresponding model values was what irked me. 

I guess "declaration-time" vis-a-vis MutltibodyPlant feels murky to me. Finalize() is the definite moment where model state/parameters are declared. So any number of calls to `MultibodyElement::set_some_default_parameter()` pre-finalize are all modifying the "declaration-time" model values, but post-finalize they aren't?

sherm1, 2021-03-31:
Overriding SetDefaultParameters() is a plausible solution but felt ugly to me when the intent is just to substitute new default values for the original ones (where the originals are now meaningless). LeafSystem already has an elegant way to store the appropriate default values for parameters -- the model value. Updating that so that the model value continues to represent the appropriate defaults for a new Context seems cleaner and preserves a useful invariant for model values -- they are the values parameters will have in a newly-created Context.

jwnimmer-tri, 2021-03-31:
> ... pre-finalize are all modifying the "declaration-time" model values, but post-finalize they aren't?

I'm not disagreeing that there is a bug related to `MultibodyElement::set_some_default_parameter()` -- it should clearly behave the same pre- & post-Finalize.

If the correct values are _already_ stored in memory somewhere (in MbT element member fields, for example) then overriding `LeafSystem::SetDefaultParameters` could solve the problem without framework changes.  (You'd then change MbP call `DeclareNumericParameter` with zero or NaNs, to highlight that the model values' data were unused.)  In this hypothetical, it's actually more robust because there is only one source of truth -- the MbT -- rather than carefully propagating the defaults into the model values as well as the MbT.

If the correct values are _not_ already stored in the MbT, then I could see how adding the ability to alter the LeafSystem's model values would be of service.

I'm just unclear yet whether the values live in MbT already.

closed 2023-10-05 by sherm1