RobotLocomotion/drake #14901: systems: LimitMalloc + Asan may not play nicely together?

EricCousineau-TRI, 2021-04-14:
See Slack: https://drakedevelopers.slack.com/archives/C270MN28G/p1618400443116900

Failure: https://drake-cdash.csail.mit.edu/test/171454211
```
+ git log -n 1 --oneline --no-decorate
a005c4081e simulator: Add test to track heap hygiene (#14900)
+ export CC=clang-9 CXX=clang++-9
+ bazel run --config=asan //systems/analysis:simulator_test
...
[ RUN      ] SimulatorTest.MonitorFunctionAndStatusReturn
munmap_chunk(): invalid pointer
{runfiles}/drake/tools/dynamic_analysis/asan: line 10: 47562 Aborted                 (core dumped) "$@"
```

Shallow investigation, post #14900, testing on a005c4081e
(1) Commenting out only `LimitMalloc` line allows it to pass.
(2) Commenting out everything in new test *except* for `LimitMalloc` causes the same issue.

Theory: `LimitMalloc` and `asan` checking do not want to play nicely together.

soonho-tri, 2021-04-14:
To be specific, the failure in MonitorFunctionAndStatusReturn comes from this line:
https://github.com/RobotLocomotion/drake/blob/a005c4081e1830078525177e51be9d7812db57dd/systems/analysis/test/simulator_test.cc#L2320-L2321

jamiesnape, 2021-04-14:
It won't. It should be a no-op. If it is a no-op with LSan, it needs to be a no-op with ASan.

https://github.com/RobotLocomotion/drake/blob/b09e40db4b1c01232b22f7705fb98aa99ef91f87/common/test_utilities/limit_malloc.h#L23-L26

jamiesnape, 2021-04-14:
(There is slight variation between different platforms and compilers with regard the differences between ASan and LSan, but typically ASan will catch a superset of LSan issues.)

jamiesnape, 2021-04-14:
According to our configuration `LSAN_OPTIONS` is defined during ASan builds, so something unexpected is happening.

https://github.com/RobotLocomotion/drake/blob/b09e40db4b1c01232b22f7705fb98aa99ef91f87/common/test_utilities/limit_malloc.cc#L28-L34

https://github.com/RobotLocomotion/drake/blob/a005c4081e1830078525177e51be9d7812db57dd/tools/dynamic_analysis/bazel.rc#L51

rpoyner-tri, 2021-04-14:
There is some interaction among multiple factors. If I leave the LimitMalloc stuff as-was, and comment out all the EXPECT_THAT clauses that use MatchesRegex, the crashes also go away. IsSupportedConfiguration is returning false in my debugging runs, but the link-time hooks are still in place. I'll see what else I can find out.

rpoyner-tri, 2021-04-14:
From further investigation, it looks like ASAN provides its own implementation of `strdup` -- whenever we call that and hand the result to (our hooked version of) `free`, we've crossed the heaps and hence come crashing down. The call to strdup() is buried inside the gtest code.

jamiesnape, 2021-04-14:
https://github.com/llvm/llvm-project/blob/ad294e572bc5c16f9dc420cc994322de6ca3fbfb/compiler-rt/lib/asan/asan_interceptors.cpp#L445-L460

rpoyner-tri, 2021-04-14:
Indeed, adding an overriding implementation of `strdup` "fixes" the immediate situation. It's worth noting that the problems arise not from any run-time LimitMalloc instrumentation; the problems arise because our link-time hook set is not complete enough to provide consistent heap access.

The follow-up question is this: is there a stable set of hooks we can provide that will allow "interop" with asan, or should we more aggressively bow out or refuse to execute in ASAN situations?

Possible paths:
 * implement more hooks to allow weakened interop (some ASAN errors won't be reported)
   * unsound (may require maintenance revisions), maybe questionable value
 * blow up at run-time if ASAN/LSAN configuration is detected
   * sound and easy to implement; probably annoying for developers (needs bazel tags to avoid explosions)
 * use compile-time detection (ifdefs) to avoid hooking at all under ASAN builds
   * sound; might be fiddly to get right across various compilers

jamiesnape, 2021-04-14:
I think `__SANITIZE_ADDRESS__` (GCC) and `__has_feature(address_sanitizer)` (Clang/Apple) cover our support configurations.

rpoyner-tri, 2021-04-14:
Cool. I will attempt to spell a proper patch to not hook in ASAN builds.

jamiesnape, 2021-04-14:
Since we control the configurations, you could also `--copt=-DDRAKE_SANITIZE_ADDRESS` if that really did not work for some reason (though no reason to think it wouldn't).

EricCousineau-TRI, 2021-04-14:
Along what Jeremy mentioned in PR, maybe it's worth doing the 2nd, blow up if ASAN/LSAN detected, rather than silently not running?

(Also: Maybe not, if it's self-evident given that you hafta consciously run asan.)

rpoyner-tri, 2021-04-14:
No-op situations are currently documented; do you think more is required @EricCousineau-TRI ? This is a test-only utility module after all.

https://github.com/RobotLocomotion/drake/blob/master/common/test_utilities/limit_malloc.h#L23-L26

EricCousineau-TRI, 2021-04-14:
Ah, gotcha; then nah.

PR #14905 (2021-04-15): limit_malloc: Avoid ASAN heap hazard
Closes #14901.

ASAN instrumentation hooks more heap-returning API than just the basic
set. strdup() is one; there may be others. The returned heap blocks come
from a private heap and can't be returned by libc's free(). Avoid the
potential crashes by removing hooks completely when ASAN is detected at
compile time.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14905)
commit 55e27755 (2021-04-15): limit_malloc: Avoid ASAN heap hazard (#14905)

* limit_malloc: Avoid ASan heap hazard

Closes #14901.

ASan instrumentation hooks more heap-returning API than just the basic
set. strdup() is one; there may be others. The returned heap blocks come
from a private heap and can't be returned by libc's free(). Avoid the
potential crashes by removing hooks completely when ASan is detected at
compile time.
closed 2021-04-15 by rpoyner-tri