RobotLocomotion/drake #14916: CalcJacobianCenterOfMassTranslationalVelocity() should take in model_instances as an input

hongkai-dai, 2021-04-17:
Currently when we compute the center-of-mass position through `CalcCenterOfMassPositionInWorld` https://github.com/RobotLocomotion/drake/blob/a939b72dfd547f7f2cfb145f400408e4e1423e46/multibody/plant/multibody_plant.h#L2476-L2478, the function takes `model_instances` as an input, so that we can evaluate the CoM position for a subset of plant.

On the other hand, when we compute the CoM Jacobian, we don't take the `model_instances` as the input
https://github.com/RobotLocomotion/drake/blob/a939b72dfd547f7f2cfb145f400408e4e1423e46/multibody/plant/multibody_plant.h#L3122-L3125
I think we should also add `model_instances` as an input to `CalcJacobianCenterOfMassTranslationalVelocity`

sherm1, 2021-04-18:
--> @mitiguy

mitiguy, 2021-04-19:
Got it -- yes, I'll work on it today/tomorrow.

mitiguy, 2021-07-13:
PR #15387 opened to address this issue.

mitiguy, 2021-09-07:
Closed with PR #15691 and PR #15387.

closed 2021-09-07 by mitiguy