RobotLocomotion/drake #14920: install_prereqs installs to user pip on mac

RussTedrake, 2021-04-18:
https://github.com/RobotLocomotion/drake/blob/625669ce0e0f4de831465707311996c0e365c51d/setup/mac/source_distribution/install_prereqs.sh#L63
https://github.com/RobotLocomotion/drake/blob/dd2e6378910d8f1da388c864eb97e61f74a72368/setup/mac/binary_distribution/install_prereqs.sh#L79

Isn't this anthetical to our philosophy on pip? In practice, this is causing another downstream issue for me in my manipulation repo, which installs a different version of matplotlib based on it's dependency management.   Related to #14250.

jamiesnape, 2021-04-19:
We could/should remove the `--upgrade`. That would at least stop drake clobbering any particularly choose version downstream.

jamiesnape, 2021-04-19:
But if you are saying we should use `venv`, then ideally yes.

RussTedrake, 2021-04-19:
I would think that installing pip should be an optional secondary installation command for users, who could use venv, etc to accomplish their goals.  In fact, I had assumed as much in the documentation here: http://underactuated.csail.mit.edu/drake.html (section A.3.2).

jwnimmer-tri, 2021-04-19:
We call `pip3.9` without a path qualifier, so I think if the user was inside a venv, we already respect that?

https://github.com/RobotLocomotion/drake/blob/v0.29.0/setup/mac/binary_distribution/install_prereqs.sh#L79

RussTedrake, 2021-04-24:
I see.  that's a different workflow than I had in my head.  I definitely think we should allow a user to opt-out of calling `pip` like this, and would ideally like us to provide requirements in the form of >= that I could merge with my requirements from other packages, but I understand that this would be hard to cover in CI.  As it stands, my downstream repo has to deal with drake as a black box that installs some versions of some packages without exposing any details.

jamiesnape, 2021-04-28:
I do not have a huge objection with providing a `constraints.txt` file, but since it would need to be opt in, we would need extra testing that we do not have resources for. Until we get `pip install drake`, I think the best I can offer is removing the `--upgrade`. As an option, its behavior changed a little while ago, so it does not work the same way as Homebrew anyway.

jamiesnape, 2021-04-28:
Now if CI on all platforms were always using a `venv` and able to use `pip` where its package manager does not provide a package, that would be great, and my answer might be different, but that is a different issue that would need to be created causing a (perhaps unwritten) policy shift and infrastructure change.

FWIW I do not really like the Mac situation either, but there is not a perfect solution that is reasonably analogous to Ubuntu that works both inside and outside the Bazel world.

jamiesnape, 2021-04-28:
BTW I think the title is wrong here, and is implying we pass `--user` to `pip` IMO, which we do not.

jwnimmer-tri, 2021-04-28:
If it helps, the Drake binary tarballs for macOS already contain `drake/share/drake/setup/requirements.txt` and `drake/share/drake/setup/Brewfile`.  Users on macOS who want to customize the prereqs process could have their own scripts that process those files, and not shell out to Drake's `install_prereqs`.

In Anzu, for example, we do _not_ shell out do Drake's Ubuntu `install_prereqs`, because we want to have a careful allowlist and control over exactly what is happening, and because we need to spin it into Docker images of various kinds.  Instead, we consume the `drake/share/drake/setup/packages-bionic.txt` file as part of Anzu's `install_prereqs` machinery, on our own terms.

Broadly speaking, `install_prereqs` is intended to provide safe and nominal default, adhering to all of the best practices on any given platform.  On Ubuntu, that means using `sudo` only for calling `apt-get` or `dpkg` (or, sigh, `locale-gen`), using only stock Ubuntu packages (no PPAs allowed), not removing anything unless we prompt the user first, etc.  On macOS, that means using homebrew, at minimum.  I _thought_ that using `pip` for Python was standard on macOS also, but if that ends up being not a best practice in the macOS ecosystem, we can look into evolving away from it.

It's possible that our installation-related documentation on macOS should explain more of these details so users can make more informed choices.

jwnimmer-tri, 2021-04-28:
If I'm understanding this thread correctly, removing the `--upgrade` would be consistent with best practices, and would resolve the most immediate pain point.  @jamiesnape if you agree, would you PR that to Drake?  (There is also an 2021-05-01 deprecation in that script you could nix at the same time, if you like.)

jamiesnape, 2021-04-28:
> (or, sigh, locale-gen)

Aside: I think we could get around that with `C.UTF-8` if someone has the the time to go through everywhere we make `en_US.UTF-8` assumptions.

>  I thought that using pip for Python was standard on macOS also, but if that ends up being not a best practice in the macOS ecosystem, we can look into evolving away from it.

It is standard and is consistent with what Homebrew recommend. Whether it is best practice is a different question.

> If I'm understanding this thread correctly, removing the --upgrade would be consistent with best practices, and would resolve the most immediate pain point. @jamiesnape if you agree, would you PR that to Drake? 

Yes. I can do that today.

jwnimmer-tri, 2021-10-05:
> We call pip3.9 without a path qualifier, so I think if the user was inside a venv, we already respect that?

I'm unable to test it, but the instructions https://drake.mit.edu/python_bindings.html#inside-virtualenv appear to call `install_prereqs` in a way that installs the prereqs system-wide, instead of the venv that's just been set up.  Maybe we should adjust those instructions so that the `pip` call during setup only affects the venv.