RobotLocomotion/drake #14926: Compile/run-time check of whether type T can resolve itself to a real number (double).

mitiguy, 2021-04-20:
While working on PR #14816 and PR #14910, it would have been helpful to be able to provide better written output from the ostream << operator, better exception messages, and more streamlined code if it was possible whether a quantity that was templatized on type T was able to be evaluated to a real number (double).

Ideally, this would be a no-op (compiler time optimization) for type T of double and AutoDiffXd, whereas it may be a runtime for type T of symbolic::expression.  

In a vague way, this relates to ExtractDoubleOrThrow() -- except not wanting to throw.

SeanCurtis-TRI, 2021-04-20:
I believe you've missed capturing one of the *driving* motivations for this whole thing. Here's my understanding, correct me if I'm wrong.

We've got numeric types that we would like to validate. The validation consists of analyzing their numerical values. This is a straightforward endeavor for `T = double` or `T = AutoDiffXd`. However, it's not generally possible for `T = symbolic::Expression`. Therefore, much of the validation code is *currently* a no op for `T = symbolic::Expression`.

However, there are limited circumstances in which the practice is feasible for `T = symbolic::Expression` -- specifically, where the expression does not contain any free variables (if I'm using the term right) -- an expression that can be evaluated without an `Environment`. In *those* cases, you'd like to perform the validation, but it needs to be preconditioned on a way to find out if you have such an `Expression` without throwing.

This is my understanding from our conversations -- would you agree with it?

If you *do* agree, I'd ask the following question: how would you assess the *value* of doing such a thing? I would suspect that the *vast* majority of quantities (`RigidTransform`, `RotationMatrix`, etc.) would arise from analyzing the system and that, during the process, computed quantities would generally have free variables. So, we'd be introducing code that wouldn't get executed for the *vast* majority of quantities. That said, I can more easily imagine that the *initial* conditions will have the majority of quantities based on constants and validating those would have value.

I think to help motivate this issue, you should continue this thought experiment to assess what the expected value of this feature would be.

jwnimmer-tri, 2021-04-20:
Symbolic code paths that implement functional requirements (i.e., when we are doing our everyday kinds of computations) should not change their control flow based on the presence or absence of free variables.  We must not validate the domain of our inputs when T=Expression.  Doing so will lead to many surprises, and defeats the theory of Expression.

There is a grey area when the branching is for non-functional purposes.  For example, we could imagine that printing (stringizing) data types with T=Expression could branch on free variables, in order to customize the diagnostic output.

jwnimmer-tri, 2021-04-20:
With respect to printing out RigidTransform, RollPitchYaw, etc., I think we can resolve this pretty easily with any novel, project-wide helper function.  If it helps, I can submit a PR for that after the in-flight PR lands.

PR #16097 (2021-11-16): [math] Improve printing of symbolic RollPitchYaw and RigidTransform
Closes #14926.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16097)
closed 2021-11-16 by jwnimmer-tri