RobotLocomotion/drake #14927: Test/update RollPitchYaw class with template type T symbolic::expression

mitiguy, 2021-04-20:
While creating PR #14910, there was an exception thrown due to a template type T symbolic::expression in the method  
RollPitchYaw::CalcRollPitchYawFromQuaternionAndRotationMatrix().
This indicates that the class RollPitchYaw may need further testing for symbolic::expression.

There is a TODO(Mitiguy) comment about this issue in roll_pitch_yaw.cc in
RollPitchYaw::CalcRollPitchYawFromQuaternionAndRotationMatrix().

There is a TODO(Mitiguy) comment about this issue in rotation_matrix_test.cc in
GTEST_TEST(RigidTransform, StreamInsertionOperatorForRigidTransform)

SeanCurtis-TRI, 2021-04-20:
One example where the code is unfriendly to `symbolic::Expression` can be found here:

https://github.com/RobotLocomotion/drake/blob/master/math/roll_pitch_yaw.cc#L141-L145

Rather than using `if_then_else` actual comparisons to `M_PI` require successful evaluation of the `symbolic::Expression`. This is not *sole* issue, merely the first one.

RussTedrake, 2021-04-26:
I hit this myself earlier today.  This was my simple reproduction (before finding this existing issue)
```
from pydrake.all import Variable, Expression, Quaternion_, RollPitchYaw_

w = Variable("w")
x = Variable("x")
y = Variable("y")
z = Variable("z")
quat = Quaternion_[Expression](w,x,y,z)
rpy = RollPitchYaw_[Expression](quat)
```
which results in 
```
RuntimeError: The following environment does not have an entry for the variable w
```

jwnimmer-tri, 2024-11-01:
> For template type T symbolic expression, improve code, documentation, throwing better exception messages, and testing for the RotationMatrix class, e.g., methods involving conversion to RollPitchYaw or quaternion.

This is not an actionable statement.  Closing as invalid.

> I hit this myself earlier today. This was my simple reproduction (before finding this existing issue)

This example no longer crashes.  Closing as fixed.

closed 2024-11-01 by jwnimmer-tri