RobotLocomotion/drake #14928: Support Santizer and Valgrind builds on Ubuntu 20.04 Focal

jamiesnape, 2021-04-20:
We are approaching a point where our primary platform may switch, or at least where developers may want to switch (Ubuntu 18.04 Bionic is a little long in the tooth), so it is time to think about supporting these builds. At the moment they are mostly functional, so it is mostly when we switch. There is only a small amount of work for everything to pass (just making sure the symbolizer works, I think).

FYI @jwnimmer-tri.

jamiesnape, 2021-04-20:
Possibly relates #14652, if It makes life easier.

jwnimmer-tri, 2021-04-20:
My primary requirement is that TRI's Drake Developers can run these configurations locally, natively (without a docker image or chroot).  My estimate at this point is that those folks will be have 18.04 as their baseline through early 2022.

jamiesnape, 2021-04-20:
I think, given we use the same compiler versions, they can co-exist on both platforms. We should just get them turning over on CI, even if it is not in the production tab.

jamiesnape, 2021-04-20:
I think the rest of the world is already moving on, so we should be a little holistic of the community at large (though obviously I know who is paying the bills). Historically we ran sanitizers on multiple platforms, it is only the last 12 months we have not.

jamiesnape, 2021-04-20:
(We are already seeing the effects of components (e.g., Python 3.6) of Bionic going EOL, for instance.)

jwnimmer-tri, 2021-11-11:
TRI is ready to swap these CI builds to 20.04 now.

BetsyMcPhail, 2021-12-15:
Check-in with the current state on Focal. After running all of the sanitizer and valgrind jobs, the following have failures:

**asan**
- [linux-focal-clang-bazel-experimental-address-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-address-sanitizer/2/) `//geometry/optimization:iris_test` Timeout
- [linux-focal-clang-bazel-experimental-gurobi-address-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-gurobi-address-sanitizer/2/)

**valgrind**
Note: All valgrind jobs have the same failure and  `//geometry/optimization:iris_test` Timeout
- [linux-focal-clang-bazel-experimental-gurobi-valgrind-memcheck](https://drake-jenkins.csail.mit.edu/view/Valgrind%20Memcheck/job/linux-focal-clang-bazel-experimental-gurobi-valgrind-memcheck/2/)
- [linux-focal-clang-bazel-experimental-mosek-valgrind-memcheck](https://drake-jenkins.csail.mit.edu/view/Valgrind%20Memcheck/job/linux-focal-clang-bazel-experimental-mosek-valgrind-memcheck/2/)
- [linux-focal-clang-bazel-experimental-snopt-valgrind-memcheck](https://drake-jenkins.csail.mit.edu/view/Valgrind%20Memcheck/job/linux-focal-clang-bazel-experimental-snopt-valgrind-memcheck/2/)
- [linux-focal-clang-bazel-experimental-valgrind-memcheck](https://drake-jenkins.csail.mit.edu/view/Valgrind%20Memcheck/job/linux-focal-clang-bazel-experimental-valgrind-memcheck/3/)

**tsan**
Note: All thread sanitizer jobs have the same failures
- [linux-focal-clang-bazel-experimental-everything-thread-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-everything-thread-sanitizer/2/)
- [linux-focal-clang-bazel-experimental-gurobi-thread-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-gurobi-thread-sanitizer/2/)
- [linux-focal-clang-bazel-experimental-mosek-thread-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-mosek-thread-sanitizer/2/)
- [linux-focal-clang-bazel-experimental-snopt-thread-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-snopt-thread-sanitizer/2/)
- [linux-focal-clang-bazel-experimental-thread-sanitizer](https://drake-jenkins.csail.mit.edu/view/Sanitizer/job/linux-focal-clang-bazel-experimental-thread-sanitizer/2/)

jwnimmer-tri, 2021-12-15:
The TSan problems look like the `tsan.supp` file is ineffective because the symbolizer is broken on focal.

For the `vector_gen` test report that's yelling about `sed`, it's fine to disable the santizier configs for that `sh_test` rule.  It's not really adding any value.

For the `iris_test` timeouts, we'll need to attempt to isolate why it's so much slower on Focal.  My best guess is that some third-party solver that it's calling has a some bad interaction with the memory checker(s).  We can have TRI people look into that, if that's easier.

RussTedrake, 2021-12-21:
When I just tried, I wasn't even able to successfully run an asan build on focal.
```
bazel test --config=asan :vpolytope_test
```
generated a spew of errors like
```
Use --sandbox_debug to see verbose messages from the sandbox
gcc: error: unrecognized command line option '-fstandalone-debug'
ERROR: /home/russt/drake/geometry/proximity/BUILD.bazel:306:17: Compiling geometry/proximity/make_box_mesh.cc failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 117 argument(s) skipped)
```

Not having focal asan on CI is one thing, but it seems more urgent if we can't run asan locally?

BetsyMcPhail, 2021-12-21:
@RussTedrake can you try to run asan again using clang?

jwnimmer-tri, 2021-12-21:
Our https://drake.mit.edu/bazel.html cheat sheet has bitrotted somewhat.  To see how to run ASan, for now the best bet is to go to one of the recent https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/ build outputs, and look at the bottom of the console log for how to re-run it, e.g.

https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/ =>
https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/job/linux-bionic-clang-bazel-nightly-address-sanitizer/ =>
https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/job/linux-bionic-clang-bazel-nightly-address-sanitizer/1133/ =>
https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/job/linux-bionic-clang-bazel-nightly-address-sanitizer/1133/console =>

```
[2:17:30 AM]    ------------------------------------------------------------------------------
[2:17:30 AM]    *** Bazel Command: CC=clang-9 CXX=clang++-9 bazel  test --compilation_mode=dbg --config=asan  ...
[2:17:30 AM]    ------------------------------------------------------------------------------
```

BetsyMcPhail, 2021-12-21:
> Our https://drake.mit.edu/bazel.html cheat sheet has bitrotted somewhat. 

I have an update to this document to PR momentarily.

BetsyMcPhail, 2022-01-27:
View in Jenkins with all of the Focal Valgrind and Sanitizer jobs --> https://drake-jenkins.csail.mit.edu/view/Focal%20Sanitizer%20and%20Valgrind/

BetsyMcPhail, 2022-02-07:
In drake-ci, remote caching is switched on for Bionic [leak-sanitizer](https://github.com/RobotLocomotion/drake-ci/blob/dada89b12ff3d3defe01b06854cb22e37aefec92/driver/environment.cmake#L255) and [debug](https://github.com/RobotLocomotion/drake-ci/blob/dada89b12ff3d3defe01b06854cb22e37aefec92/driver/environment.cmake#L260) jobs. This needs to be switched to Focal. 

I couldn't find an issue related to this change, it appears to be in response to [this ](https://drakedevelopers.slack.com/archives/C270MN28G/p1549493207082300) slack conversation.

BetsyMcPhail, 2022-02-08:
To complete this issue we need to:

1. Disable any failing tsan tests
2. Finish #16223, including any documentation updates
3. In drake-ci, change remote caching to Focal [leak-sanitizer](https://github.com/RobotLocomotion/drake-ci/blob/dada89b12ff3d3defe01b06854cb22e37aefec92/driver/environment.cmake#L255)
4. In coordination with merging 2. and 3., update Jenkins. Specifically:
   - remove Bionic valgrind jobs (no longer supported)
   - upgrade Focal valgrind and sanitizer jobs to production
   - move Bionic sanitizer jobs out of production

@jwnimmer-tri or @mwoehlke-kitware did I miss anything here?

jwnimmer-tri, 2022-02-08:
> move Bionic sanitizer jobs out of production

I think we might as well just delete them (along with deleting the valgrind jobs)?

BetsyMcPhail, 2022-02-11:
Changes to Jenkins are ready [here](https://github.com/RobotLocomotion/drake-jenkins-jobs/commit/f25e56ceb77f4d68ecd82543bed5012d3f6d53d7). They have NOT been deployed yet, waiting for #16557.

BetsyMcPhail, 2022-02-11:
All pull requests have been merged and Jenkins is updated.

closed 2022-02-11 by BetsyMcPhail