RobotLocomotion/drake #14936: pydrake: Should handle deprecation of implicit conversions to `_int__`

EricCousineau-TRI, 2021-04-22:
From @RussTedrake on Slack: https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1619088373041800

> Today I encountered
> ```
> DeprecationWarning: an integer is required (got type pydrake.systems.framework.InputPortIndex).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
> return LinearQuadraticRegulator(plant, context,`
> ```
> ... the LQR method indeed takes an int for (possible) use as an input port index.  Of course I can do the cast here, but is this a widespread pattern that we need to fix?

I'm not sure where this arises (CPython / Python, or `pybind11`), but once we identify it, two possible solutions may just be:
- Use a different spelling, such as `__long__`
- Workaround it in bindings (either just binding `InputPortIndex1` flavor, or add some utilities so that our pybind11 bindings can still do it)

Russ - I assume this was encountered in CPython 3.7 via Google Colab?

RussTedrake, 2021-04-24:
actually it was python3.9 on mac.

rpoyner-tri, 2026-11-10:
The deprecation period is long gone. Index types now correctly implement `__index__` instead. Closing.

closed 2026-11-10 by rpoyner-tri