RobotLocomotion/drake #14948: multibody: feature request: "Jacobian" of `MapVelocityToQDot`

RussTedrake, 2021-04-26:
I've got a mathematical program that can almost get away with using only `MultibodyPlant<double>`, but needs a separate `MultibodyPlant<AutoDiffXd>` for getting the gradients of [MapVelocityToQDot](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a7bb24fed19f16a76450eece1857b92dc)
I suspect I'm not alone.

Specifically, the existing method computes `qdot = N(q) * v`. I would like to compute `d qdot / dq = ∑ᵢ dN(q)/dq_i * v` and `d qdot / dv = N(q)` without needing `AutoDiffXd`. 

cc @hongkai-dai

sherm1, 2021-04-26:
A few notes from f2f discussion:
- Having an RPYXYZ free joint makes this less urgent
- This requires extending the System API since MapVelocityToQDot() exists at the System level (because q and v are understood at that level).

We may want to consider related method MapAccelerationToQDotDot() since that can be computed faster than forming the partials.

RussTedrake, 2021-04-27:
`MapAccelerationToQDotDot` doesn't solve the original need.  What I actually have is variables `q`, `v`, and `h` (for timestep) in a `MathematicalProgram`, and I want to relate them with Euler integration constraints.  I want to implement a constraint and its derivative that is the generalization of `q = q + h*v`.

I was implementing this as `q + h*MapVelocityToQDot(q,v)`, but @hongkai-dai pointed out that this might not be sufficient, since I have a separate nonlinear "unit quaternion" constraints on `q` already.  What is the best way to implement this?

hongkai-dai, 2021-04-27:
I think I need to add a separate constraint for integrating unit quaternions. BTW, I think `MapQDotToVelocity(q[n], (q[n+1] - q[n])/h)` isn't completely correct either, since if `q` contains quaternions, then q[n+1] - q[n] won't be on the tangent surface of the unit sphere, which should be true for qdot. Hence (q[n+1] - q[n])/h isn't a valid qdot.

sherm1, 2021-04-27:
> Hence (q[n+1] - q[n])/h isn't a valid qdot.

I don't think that's exactly true, Hongkai. By definition of derivative this approaches qdot as h --> 0. So it is a valid low-order approximation of qdot. Definitely not perfect with finite h though! Real quaternion qdots are perpendicular to the quaternion direction so if integrated perfectly will preserve the length constraint automatically. The need for projection back to the manifold is due to numerical error.

hongkai-dai, 2021-04-27:
@sherm1 I think we are talking about different use cases. You are probably thinking about time integration with small `h` (like in the order of 1ms). The use case that I am thinking about is the motion planning, where `h` is typically in the order of 100ms. In these cases (q[n+1] - q[n])/h is obviously not a valid qdot.

sherm1, 2021-04-27:
Oh! Yes that wouldn't be much of a qdot.

xuchenhan-tri, 2022-05-23:
Circling the discussion back to the OP's original request. First of all, is this request still relevant?

It seems like the non-trivial change required to support this would be to add an API to get the derivative of the kinematics jacobian (`dN(q)/dq`) in an operator form that directly gets applied to a input velocity `v`. This would require a virtual function for the mobilizer class -- something like `MatrixX<T> GetDerivativeOfQdotWrtQ(const VectorX<T>& v)`. 

Skimming through the joint mobilizers that we support, most mobilizers would just return the zero matrix with the exception of free joint and gimbal joint mobilizers that require some math.

Seems like this falls in the area of expertise of @mitiguy if the issue is still relevant. WDYT?

RussTedrake, 2022-05-25:
@xuchenhan-tri -- i agree with your assessment.  I'm not working on code that is blocked by this at the moment, but I suspect I will hit it again soon enough.  Backlog is reasonable, but I do consider it still relevant.