RobotLocomotion/drake #14968: Prepare for Eigen 3.4

jwnimmer-tri, 2021-04-29:
Eigen has entered the release candidate process for a 3.4.0 release.  It's likely that Drake will incur some kinds of compilation errors due to the changes.

It's not clear yet how long it will be until a final release, but we should probably test against the candidates, and either adapt Drake to meet the new requirements, or else file Eigen upstream bugs if appropriate.

Since homebrew will immediately adopt Eigen 3.4.0 once it's tagged, we should try to get ahead of that failure mode.

jwnimmer-tri, 2021-04-29:
Looking at https://eigen.tuxfamily.org/index.php?title=Main_Page and clicking on the rc1 announcement, we see
> Eigen 3.4-rc1 has been released on April 19, 2021.
> Depending on the amount of reported issues, 3.4 will be released late April or early May.

Upgrading priority.

jamiesnape, 2021-04-29:
> Depending on the amount of reported issues, 3.4 will be released late April or early May

Oh dear.

jamiesnape, 2021-04-29:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-big-sur-unprovisioned-clang-bazel-nightly-release/70/consoleFull was a failed build on macOS with the RC (link should be available through mid-July).

BetsyMcPhail, 2021-06-08:
PR #15142 was created to try out the release candidate (6/7/2021) on both Linux and Mac. A summary of results with relevant CDash links to full output  are listed below:

**Build Errors**

1. [ ERROR: .../multibody/tree/BUILD.bazel:99:17: Compiling multibody/tree/multibody_tree.cc failed](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=1423080)
```
multibody/tree/multibody_tree.cc:2939:36: error: no viable overloaded '='
    Su(actuator_index, user_index) = 1.0;
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~
```
2. [ERROR: .../common/trajectories/BUILD.bazel:105:17: Compiling common/trajectories/bspline_trajectory.cc failed](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=1423080)
```
  In file included from common/trajectories/bspline_trajectory.cc:1:
  In file included from bazel-out/k8-opt/bin/common/trajectories/_virtual_includes/bspline_trajectory/drake/common/trajectories/bspline_trajectory.h:6:
  In file included from bazel-out/k8-opt/bin/common/_virtual_includes/drake_bool/drake/common/drake_bool.h:5:
  In file included from external/eigen/Eigen/Core:19:
  external/eigen/Eigen/src/Core/util/Macros.h:1325:34: error: value of type 'Eigen::CwiseBinaryOp<Eigen::internal::scalar_cmp_op<double, double, Eigen::internal::cmp_EQ>, const Eigen::ArrayWrapper<const Eigen::Matrix<double, -1, -1, 0, -1, -1> >, const Eigen::ArrayWrapper<const Eigen::Matrix<double, -1, -1, 0, -1, -1> > >' is not contextually convertible to 'bool'
  bool all(T t, Ts ... ts){ return t && all(ts...); }
```
3. [ERROR: .../solvers/BUILD.bazel:962:17: Compiling solvers/csdp_solver.cc failed](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=1423080)
```
solvers/csdp_solver.cc:102:24: error: assigning to 'Eigen::DenseCoeffsBase<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 1>::Scalar' (aka 'double') from incompatible type 'typename internal::enable_if<(!IsRowMajor) && (!(internal::get_compile_time_incr<typename IvcType<TypeSafeIndex<FreeVariableTag> >::type>::value == 1 || internal::is_valid_index_type<TypeSafeIndex<FreeVariableTag> >::value)), IndexedView<const Matrix<double, -1, 1, 0, -1, 1>, typename IvcType<TypeSafeIndex<FreeVariableTag> >::type, IvcIndex> >::type' (aka 'Eigen::IndexedView<const Eigen::Matrix<double, -1, 1, 0, -1, 1>, drake::TypeSafeIndex<drake::solvers::internal::FreeVariableTag>, Eigen::internal::SingleRange>')
      (*prog_sol)(i) = s_val(std::get<SdpaFreeFormat::FreeVariableIndex>(
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
4. [ERROR: .../multibody/plant/BUILD.bazel:490:20: Compiling multibody/plant/test/multibody_plant_reflected_inertia_test.cc failed](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=1423080)
```
multibody/plant/test/multibody_plant_reflected_inertia_test.cc:88:48: error: no viable conversion from 'typename internal::enable_if<(!IsRowMajor) && (!(internal::get_compile_time_incr<typename IvcType<TypeSafeIndex<JointActuatorElementTag> >::type>::value == 1 || internal::is_valid_index_type<TypeSafeIndex<JointActuatorElementTag> >::value)), IndexedView<const Matrix<double, -1, 1, 0, -1, 1>, typename IvcType<TypeSafeIndex<JointActuatorElementTag> >::type, IvcIndex> >::type' (aka 'Eigen::IndexedView<const Eigen::Matrix<double, -1, 1, 0, -1, 1>, drake::TypeSafeIndex<drake::multibody::JointActuatorElementTag>, Eigen::internal::SingleRange>') to 'double'
      joint_actuator.set_default_rotor_inertia(rotor_inertias(index));
                                               ^~~~~~~~~~~~~~~~~~~~~
bazel-out/k8-opt/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/joint_actuator.h:183:41: note: passing argument to parameter 'rotor_inertia' here
  void set_default_rotor_inertia(double rotor_inertia) {
                                        ^
multibody/plant/test/multibody_plant_reflected_inertia_test.cc:89:45: error: no viable conversion from 'typename internal::enable_if<(!IsRowMajor) && (!(internal::get_compile_time_incr<typename IvcType<TypeSafeIndex<JointActuatorElementTag> >::type>::value == 1 || internal::is_valid_index_type<TypeSafeIndex<JointActuatorElementTag> >::value)), IndexedView<const Matrix<double, -1, 1, 0, -1, 1>, typename IvcType<TypeSafeIndex<JointActuatorElementTag> >::type, IvcIndex> >::type' (aka 'Eigen::IndexedView<const Eigen::Matrix<double, -1, 1, 0, -1, 1>, drake::TypeSafeIndex<drake::multibody::JointActuatorElementTag>, Eigen::internal::SingleRange>') to 'double'
      joint_actuator.set_default_gear_ratio(gear_ratios(index));
                                            ^~~~~~~~~~~~~~~~~~
bazel-out/k8-opt/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/joint_actuator.h:189:38: note: passing argument to parameter 'gear_ratio' here
  void set_default_gear_ratio(double gear_ratio) {
```
5. [ERROR: .../multibody/inverse_kinematics/BUILD.bazel:257:20: Compiling multibody/inverse_kinematics/test/global_inverse_kinematics_reachable_test.cc failed](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=1423080)
```
multibody/inverse_kinematics/test/global_inverse_kinematics_reachable_test.cc:80:17: error: no viable conversion from 'typename internal::enable_if<(!IsRowMajor) && (!(internal::get_compile_time_incr<typename IvcType<TypeSafeIndex<JointElementTag> >::type>::value == 1 || internal::is_valid_index_type<TypeSafeIndex<JointElementTag> >::value)), IndexedView<Matrix<double, 7, 1, 0, 7, 1>, typename IvcType<TypeSafeIndex<JointElementTag> >::type, [... cut ...]
bazel-out/k8-dbg/bin/common/_virtual_includes/_symbolic_private_headers_impl/drake/common/symbolic_monomial_basis_element.h:28:7:   required from here
external/eigen/Eigen/src/Core/functors/AssignmentFunctors.h:24:102: error: cannot convert 'const Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1> >' to 'double' in assignment
   EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE void assignCoeff(DstScalar& a, const SrcScalar& b) const { a = b; }
                                                                                                    ~~^~~
```

BetsyMcPhail, 2021-07-01:
Fixes and workarounds are in PR #15142. There is currently one build error and 20 failing tests. Results are the same across all platforms and configurations.

To test locally, checkout branch from PR and revert referenced commit. To sanity check against Eigen 3.3, update `eigen/repository.bzl` to `branch = "3.3.9"` instead of specific commit.

- [x] Remove deprecated function ([1a3083f](https://github.com/RobotLocomotion/drake/pull/15142/commits/1a3083ff273118d920dbfd935f869c660bc3ac3c))
- [x] Ensure we're using `drake::all` not `Eigen::all` ([505f2e2](https://github.com/RobotLocomotion/drake/pull/15142/commits/505f2e2908680ee7be2b5f7c31079bc1376e890c))
- [ ] `EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR` error (temp. workaround to get tests running [8a3532d](https://github.com/RobotLocomotion/drake/pull/15142/commits/8a3532d6281f1614e8218b5d6e2ce8f09072ab76), need an actual solution)

To test (for example, there will be lots of errors if `8a3532d` is reverted):

```
bazel test multibody/fixed_fem/dev:stretch_test
...
ultibody/fixed_fem/dev/fem_state_base.cc:39:19: error: no viable overloaded '='
    q_(dof_index) = boundary_state(0);
    ~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~
multibody/fixed_fem/dev/fem_state_base.cc:52:42: note: in instantiation of member function 'drake::multibody::fixed_fem::FemStateBase<double>::ApplyBoundaryCondition' requested here
    class ::drake::multibody::fixed_fem::FemStateBase);
                                         ^
external/eigen/Eigen/src/Core/MatrixBase.h:139:14: note: candidate function not viable: no known conversion from 'const double' to 'const Eigen::MatrixBase<Eigen::IndexedView<Eigen::Matrix<double, -1, 1, 0, -1, 1>, drake::TypeSafeIndex<drake::multibody::fixed_fem::DofTag>, Eigen::internal::SingleRange> >' for 1st argument
    Derived& operator=(const MatrixBase& other);
             ^
external/eigen/Eigen/src/Core/IndexedView.h:114:3: note: candidate function not viable: no known conversion from 'const double' to 'const Eigen::IndexedView<Eigen::Matrix<double, -1, 1, 0, -1, 1>, drake::TypeSafeIndex<drake::multibody::fixed_fem::DofTag>, Eigen::internal::SingleRange>' for 1st argument
  EIGEN_INHERIT_ASSIGNMENT_OPERATORS(IndexedView)
  ^
external/eigen/Eigen/src/Core/util/Macros.h:1228:5: note: expanded from macro 'EIGEN_INHERIT_ASSIGNMENT_OPERATORS'
    EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR(Derived) \
    ^
external/eigen/Eigen/src/Core/util/Macros.h:1195:52: note: expanded from macro 'EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR'
    EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE Derived& operator=(const Derived& other) { Base::operator=(other); return *this; } \
                                                   ^
external/eigen/Eigen/src/Core/MatrixBase.h:150:14: note: candidate template ignored: could not match 'EigenBase<type-parameter-0-0>' against 'const double'
    Derived& operator=(const EigenBase<OtherDerived>& other);
             ^
external/eigen/Eigen/src/Core/MatrixBase.h:154:14: note: candidate template ignored: could not match 'ReturnByValue<type-parameter-0-0>' against 'const double'
    Derived& operator=(const ReturnByValue<OtherDerived>& other);
             ^
external/eigen/Eigen/src/Core/IndexedView.h:114:3: note: candidate template ignored: could not match 'DenseBase<type-parameter-0-0>' against 'const double'
  EIGEN_INHERIT_ASSIGNMENT_OPERATORS(IndexedView)
  ^
external/eigen/Eigen/src/Core/util/Macros.h:1228:5: note: expanded from macro 'EIGEN_INHERIT_ASSIGNMENT_OPERATORS'
    EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR(Derived) \
    ^
external/eigen/Eigen/src/Core/util/Macros.h:1197:52: note: expanded from macro 'EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR'
    EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE Derived& operator=(const DenseBase<OtherDerived>& other) { Base::operator=(other.derived()); return *this; }
                                                   ^
...
```
- [x] robot_plan_interpolator_test: Use `EXPECT_NEAR` instead of `EXPECT_FLOAT_EQ` ([7b03907](https://github.com/RobotLocomotion/drake/pull/15142/commits/7b039071d2e7aae681f0d47668c5d6e85443d98c)) 
- [ ] Fix errors introduced in [Eigen commit a235ddef](https://gitlab.com/libeigen/eigen/-/commit/a235ddef394602229a938765fc2e5fe12e5b73e1) (workaround [7447a23](https://github.com/RobotLocomotion/drake/pull/15142/commits/7447a23ce79add5f5a99ef947bdb16bf571d3d2e))

To test (for example, there will be lots of errors if `7447a23` is reverted):

```
bazel test //examples/pendulum:print_symbolic_dynamics_test
...
ERROR: /Users/kitware/drake-distro/drake/geometry/proximity/BUILD.bazel:448:17: Compiling geometry/proximity/mesh_plane_intersection.cc failed: (Aborted): wrapped_clang failed: error executing command external/local_config_cc/wrapped_clang '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 -DNDEBUG '-DNS_BLOCK_ASSERTIONS=1' ... (remaining 140 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from geometry/proximity/mesh_plane_intersection.cc:1:
In file included from bazel-out/darwin-opt/bin/geometry/proximity/_virtual_includes/mesh_plane_intersection/drake/geometry/proximity/mesh_plane_intersection.h:9:
In file included from bazel-out/darwin-opt/bin/geometry/proximity/_virtual_includes/bvh/drake/geometry/proximity/bvh.h:12:
In file included from bazel-out/darwin-opt/bin/common/_virtual_includes/essential/drake/common/eigen_types.h:14:
[... cut ...]
The following tests are failing without much output in the normal test results. Leak sanitizer results may provide some clues?
- [ ] [//multibody/fixed_fem/dev](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=1435155&filtercount=1&showfilters=1&field1=testname&compare1=63&value1=multibody%2Ffixed_fem)
- [ ] [//math:quadratic_form_test](https://drake-cdash.csail.mit.edu/test/263999208)
- [ ]  [//math:rotation_matrix_test](https://drake-cdash.csail.mit.edu/test/263999095)
- [ ] [//systems/controllers:linear_quadratic_regulator_test](https://drake-cdash.csail.mit.edu/test/263998922)

BetsyMcPhail, 2021-07-01:
@jwnimmer-tri it would be helpful to have some Eigen experts dig into these failures.

jwnimmer-tri, 2021-07-01:
@BetsyMcPhail

These three completed checkbox commits seem like good fixes.  Could you PR each one to Drake individually, so that we can get them merged and out of the way to narrow down the branch?

You can assign these two to me for review.
- `Remove deprecated function (1a3083f)`
- `robot_plan_interpolator_test: Use EXPECT_NEAR instead of EXPECT_FLOAT_EQ (7b03907)`

This one _looks_ correct at first glance, but please assign to @soonho-tri to review just to be sure.
- `Ensure we're using drake::all not Eigen::all (505f2e2)`

jwnimmer-tri, 2021-07-01:
> `EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR error (temp. workaround to get tests running 8a3532d, need an actual solution)`

I spent a few minutes investigating, and I don't think there is a good systemic fix.  I think a more polished version of your current work-around is the best way to go, as follows.

For lines like this, which fail because an `Eigen::Matrix` lookup uses a `TypeSaveIndex` (in this case, the `actuator.index()` ...

```c++
B(actuator.joint().velocity_start(), actuator.index()) = 1;
```

... we just need to guide the implicit conversion to int, like so ...

```
B(actuator.joint().velocity_start(), int(actuator.index())) = 1;
```

Note that I use the int constructor `int(...)` instead of `static_cast<int>(...)`, and that I keep it inline instead of factoring it into a different named variable just to change the type.

I suspect that more brief approach will lead to a slightly more readable patch, that would be acceptable to merge to master now.  Could you try opening a PR with that?

jwnimmer-tri, 2021-07-01:
>  Build error: 'static_assert failed ...'
> ...
>  cassie_benchmark test failures

I suspect that our forked copy of `common/autodiffxd.h` might be incompatible with Eigen 3.4.  Maybe I can nominate @rpoyner-tri to root cause and fix these two failures.

jwnimmer-tri, 2021-07-01:
Any tests that are failing related to `symbolic::Expression` are probably due to #15298.  We could apply that work-around to the branch for now, to help clear the deck of those errors until upstream gets fixed.

BetsyMcPhail, 2021-07-01:
> Any tests that are failing related to `symbolic::Expression` are probably due to #15298. We could apply that work-around to the branch for now, to help clear the deck of those errors until upstream gets fixed.

Sounds good to me. I will apply the patch now.

BetsyMcPhail, 2021-07-07:
> > `EIGEN_INHERIT_ASSIGNMENT_EQUAL_OPERATOR error (temp. workaround to get tests running 8a3532d, need an actual solution)`
> 
> I spent a few minutes investigating, and I don't think there is a good systemic fix. I think a more polished version of your current work-around is the best way to go, as follows.
> 
> For lines like this, which fail because an `Eigen::Matrix` lookup uses a `TypeSafeIndex` (in this case, the `actuator.index()`) ...
> 
> ```c++
> B(actuator.joint().velocity_start(), actuator.index()) = 1;
> ```
> 
> ... we just need to guide the implicit conversion to int, like so ...
> 
> ```
> B(actuator.joint().velocity_start(), int(actuator.index())) = 1;
> ```
> 
> Note that I use the int constructor `int(...)` instead of `static_cast<int>(...)`, and that I keep it inline instead of factoring it into a different named variable just to change the type.
> 
> I suspect that more brief approach will lead to a slightly more readable patch, that would be acceptable to merge to master now. Could you try opening a PR with that?

Using just the int constructor makes cpplint unhappy, For example, in multibody_plant.cc, 

```
--- a/multibody/plant/multibody_plant.cc
+++ b/multibody/plant/multibody_plant.cc
@@ -768,7 +768,7 @@ MatrixX<T> MultibodyPlant<T>::MakeActuationMatrix() const {
     // This method assumes actuators on single dof joints. Assert this
     // condition.
     DRAKE_DEMAND(actuator.joint().num_velocities() == 1);
-    B(actuator.joint().velocity_start(), actuator.index()) = 1;
+    B(actuator.joint().velocity_start(), int(actuator.index())) = 1;
   }
   return B;
 }
```

Results in style error:

```
==================== Test output for //multibody/plant:py/multibody_plant_core_cpplint:
multibody/plant/multibody_plant.cc:771:  Using deprecated casting style.  Use static_cast<int>(...) instead  [readability/casting] [4]
```

jwnimmer-tri, 2021-07-07:
Oops.  What about `int{actuator.index()}` with curlies instead?

jwnimmer-tri, 2021-07-21:
At this point, I think this meta-issue isn't useful anymore, so I'll close it.

We'll leave #15142 open as our canary branch.  Once it's passing _all_ CI (even post-merge) with zero diffs vs master other than using Eigen 3.4 in the repository rule, then our job is done.

As we identify acute problems in the canary branch, we'll open specific issues for them.

closed 2021-07-21 by jwnimmer-tri