RobotLocomotion/drake #14978: Diagram.ToAutoDiffXd fails with a lousy message when the diagram has MeshcatVisualizer

RussTedrake, 2021-04-30:
Running the following code:
```
from pydrake.all import DiagramBuilder, MeshcatVisualizer

builder = DiagramBuilder()
visualizer = builder.AddSystem(MeshcatVisualizer(
    scene_graph=scene_graph,
    zmq_url="new"))
diagram = builder.Build()

ad = diagram.ToAutoDiffXd()
```
produces the message:
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-8-2f53f326b656> in <module>()
     10 diagram = builder.Build()
     11 
---> 12 ad = diagram.ToAutoDiffXd()

RuntimeError: The object named [] of type drake::systems::Diagram<double> does not support ToAutoDiffXd.
```

Can we make that more helpful?  I've confirmed that the actual `MeshcatVisualizer` system *is* setting it's name.  But I guess the Diagram is gobbling it up?

@jwnimmer-tri -- could you take a look?

jwnimmer-tri, 2021-04-30:
The Diagram _pre-computes_ its supported scalar types as the intersection of its child systems' scalar types [1].  Later on, when `System::ToAutoDiffXd` tries to do a conversion, it doesn't even try delegate to the Diagram's children, because it already knows it would fail.  So, the error message is that the Diagram _itself_ doesn't support conversion, just like if it were any other leaf system.  This is factually correct but not super helpful.

I will think about how to end up with a better explanation in the exception message, but nothing quickly comes to mind.

---

[1] https://github.com/RobotLocomotion/drake/blob/v0.29.0/systems/framework/diagram.cc#L1416-L1423

jwnimmer-tri, 2021-04-30:
My best idea so far is that we'd supplement `bool SystemScalarConverter::IsConvertible()` with a new method (also on `SystemScalarConverter`) that could provide a string explanation for why not, in the `false` case.  Then individual leaf systems could explain their lack of scalar support, and the diagram could concatenate any unsupported subsystem explanations as to why itself can't convert.

RussTedrake, 2021-04-30:
Thanks!  

It seems that even the addition to the logic around `SystemScalarConverter:: RemoveUnlessAlsoSupportedBy` that would remember the name of the system (if not a bespoke/overloaded explanation from that subsystem) that caused support to be removed, that would already be super helpful.  

I can think of only a few examples (like MBP/SceneGraph) where it would actually pay for the subsystem to explain *why* we refuse to convert; or where a leaf system author would take the time to write such an explanation.

jwnimmer-tri, 2021-05-01:
In this hypothetical design, we would compute a default reason in the typical case where the author did not supply one, which pointed out whether the class has specialized its traits and/or used a SystemTypeTag, with citations to the scalar conversion documentation.

PR #16131 (2021-11-29): [framework] Improve error message for diagram scalar conversion
Closes #14978.

Running the issue's example now produces: `RuntimeError: System ::_ of type drake::systems::Diagram<double> does not support scalar conversion to type drake::AutoDiffXd (because System ::_::meshcat_visualizer of type pydrake.systems.meshcat_visualizer.MeshcatVisualizer does not support scalar conversion to type drake::AutoDiffXd)`

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16131)
closed 2021-11-29 by SeanCurtis-TRI