RobotLocomotion/drake #14988: ImportError: undefined symbol

zoukj, 2021-05-01:
Today installed Drake via binary [https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-focal.tar.gz.](https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-focal.tar.gz.)

Run ```sudo /opt/drake/share/drake/setup/install_prereqs```

Add drake to python path

Finally, 
I run ```python3 -c 'import pydrake; print(pydrake.__file__)'``` 
and get 
```ImportError: /opt/drake/lib/python3.8/site-packages/pydrake/common/../../../../libdrake.so: undefined symbol: _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_```

jwnimmer-tri, 2021-05-01:
Hmm, that's unfortunate.  I just did a quick check on my focal system and it worked fine for me.

Can you please share
(1) the contents of the `drake/share/doc/drake/VERSION.TXT` file
(2) the output of `lsb_release -a`
(3) the output of `dpkg -l libstdc++* | grep ^ii`

zoukj, 2021-05-01:
(1) the contents of ```VERSION.TXT```: ```20210501070504 3e959415af53dfc1f67c5edd4d4c28194af93613```
(2) ```lsb_release -a```
```No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal```
(3) ```dpkg -l libstdc++* | grep ^ii```
```ii  libstdc++-7-dev:amd64     7.5.0-6ubuntu2        amd64        GNU Standard C++ Library v3 (development files)
ii  libstdc++-9-dev:amd64     9.3.0-17ubuntu1~20.04 amd64        GNU Standard C++ Library v3 (development files)
ii  libstdc++6:amd64          10.2.0-5ubuntu1~20.04 amd64        GNU Standard C++ Library v3
ii  libstdc++6:i386           10.2.0-5ubuntu1~20.04 i386         GNU Standard C++ Library v3```

jwnimmer-tri, 2021-05-01:
Here are some commands that illustrate how my Focal test system is finding its library paths.

Could you run the same commands and post your output?

```console
(focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ ldd lib/python3.8/site-packages/pydrake/symbolic.cpython-38-x86_64-linux-gnu.so | grep spdlog
	libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f0ce4a9a000)

(focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ dpkg -l libspdlog1
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libspdlog1     1:1.5.0-1    amd64        Very fast C++ logging library

(focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ readelf --wide -s /lib/x86_64-linux-gnu/libspdlog.so.1 | grep _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_
   519: 0000000000037360   298 FUNC    GLOBAL DEFAULT   14 _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_
```

If you don't have `readelf` on your system, you can `sudo apt install binutils` to get it.

jwnimmer-tri, 2021-05-01:
My best guess is that `ldd` will show that pydrake is picking up an incompatible build of spdlog from somewhere else on your library paths, so be sure to run these commands from within the same shell environment (a venv, it looks like?) where you were trying to run the `python3 -c ...` that was failing.

zoukj, 2021-05-02:
> Here are some commands that illustrate how my Focal test system is finding its library paths.
> 
> Could you run the same commands and post your output?
> 
> ```
> (focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ ldd lib/python3.8/site-packages/pydrake/symbolic.cpython-38-x86_64-linux-gnu.so | grep spdlog
> 	libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f0ce4a9a000)
> 
> (focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ dpkg -l libspdlog1
> Desired=Unknown/Install/Remove/Purge/Hold
> | Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
> |/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
> ||/ Name           Version      Architecture Description
> +++-==============-============-============-=================================
> ii  libspdlog1     1:1.5.0-1    amd64        Very fast C++ logging library
> 
> (focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ readelf --wide -s /lib/x86_64-linux-gnu/libspdlog.so.1 | grep _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_
>    519: 0000000000037360   298 FUNC    GLOBAL DEFAULT   14 _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_
> ```
> 
> If you don't have `readelf` on your system, you can `sudo apt install binutils` to get it.

(1) ```libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f5504f88000)```

(2) ```ii  libspdlog1     1:1.6.1-1ubuntu1-20200610061810-cc9f7d4e1 amd64        Very fast C++ logging library```

(3) ```885: 0000000000038610   314 FUNC    GLOBAL DEFAULT   14 _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_St13unordered_mapIcSt10unique_ptrINS_21custom_flag_formatterESt14default_deleteISA_EESt4hashIcESt8equal_toIcESaISt4pairIKcSD_EEE```

> Here are some commands that illustrate how my Focal test system is finding its library paths.
> 
> Could you run the same commands and post your output?
> 
> ```
> (focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ ldd lib/python3.8/site-packages/pydrake/symbolic.cpython-38-x86_64-linux-gnu.so | grep spdlog
> 	libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f0ce4a9a000)
> 
> (focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ dpkg -l libspdlog1
> Desired=Unknown/Install/Remove/Purge/Hold
> | Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
> |/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
> ||/ Name           Version      Architecture Description
> +++-==============-============-============-=================================
> ii  libspdlog1     1:1.5.0-1    amd64        Very fast C++ logging library
> 
> (focal)jwnimmer@call-cps:~/Downloads/drake-latest-focal/drake$ readelf --wide -s /lib/x86_64-linux-gnu/libspdlog.so.1 | grep _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_
>    519: 0000000000037360   298 FUNC    GLOBAL DEFAULT   14 _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_
> ```
> 
> If you don't have `readelf` on your system, you can `sudo apt install binutils` to get it.

(1) ```libspdlog.so.1 => /lib/x86_64-linux-gnu/libspdlog.so.1 (0x00007f5504f88000)```

(2) ```ii  libspdlog1     1:1.6.1-1ubuntu1-20200610061810-cc9f7d4e1 amd64        Very fast C++ logging library```

(3) ```885: 0000000000038610   314 FUNC    GLOBAL DEFAULT   14 _ZN6spdlog17pattern_formatterC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_17pattern_time_typeES6_St13unordered_mapIcSt10unique_ptrINS_21custom_flag_formatterESt14default_deleteISA_EESt4hashIcESt8equal_toIcESaISt4pairIKcSD_EEE```

jwnimmer-tri, 2021-05-02:
Interesting!

> libspdlog1 1:1.6.1

This is the proximate cause of your problem.  Your machine has version 1.6.1 somehow, but Focal's default is supposed to be 1.5.0 (https://packages.ubuntu.com/source/focal/spdlog).

We can ask `apt-cache` where your version came from.  Here's mine:

```console
(focal)jwnimmer@call-cps:~$ apt-cache showpkg libspdlog1
Package: libspdlog1
Versions: 
1:1.5.0-1 (/var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_focal_universe_binary-amd64_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_focal_universe_binary-amd64_Packages
                  MD5: b2d503aa303f67bc884a02e53e082428
 Description Language: en
                 File: /var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_focal_universe_i18n_Translation-en
                  MD5: b2d503aa303f67bc884a02e53e082428

Reverse Depends: 
  libspdlog-dev,libspdlog1 1:1.5.0-1
Dependencies: 
1:1.5.0-1 - libc6 (2 2.14) libgcc-s1 (2 3.0) libstdc++6 (2 9) 
Provides: 
1:1.5.0-1 - 
Reverse Provides: 
```

Perhaps your 1.6.1 came from a PPA?

If you can figure out where it came from and disable that, then you can `auto apt remove libspdlog1` to uninstall the unwanted version of the library, and then re-run Drake `install_prerereqs` to restore the default (Ubuntu Focal) version.

zoukj, 2021-05-02:
I remove ```libspdlog1=1:1.6.1-1ubuntu1-20200610061810-cc9f7d4e1```, and it does work. 
thanks for your patience!

closed 2021-05-02 by jwnimmer-tri