RobotLocomotion/drake #15002: multibody: MultibodyPlant should register *all* frames with SceneGraph (not just body frames)

RussTedrake, 2021-05-04:
As discussed in #14979.  Discussion summarized here:

@RussTedrake : "I think this hits at the heart of a deeper problem with the pipeline: not all Frames from a MultibodyPlant are registered with the SceneGraph (only the body frames are). This sure seems like a defect on the MBP <-> SG relationship to me. "

@hjsuh94: "as a user, I did briefly struggle to find the right syntax for this (which is what motivated me to include GetBodyFrameIdorThrow in the documentation for frames_to_draw) - it is certainly a mouthful!
To illustrate your second point, I couldn't add a Frame tag in my sdf file and visualize it using frames_to_draw because of the exact reasons you mentioned. (The workaround was to create a dummy link just for visualization, which works but is not ideal)."

@SeanCurtis-TRI "There is no reason that MBP couldn't register a geometry::FrameId with SceneGraph for every non-body frame. It just currently doesn't do it.  The historical reason is pretty straightforward. Frames are there to move geometries. If we don't have geometries, what's the point of having a frame? This particularly application is the answer to that. I'd actually like to visualize frames!"

SeanCurtis-TRI, 2024-01-08:
Note: the single biggest obstacle to this is the fact that `MultibodyPlant` allows adding a frame *after* finalization. At finalization, `MultibodyPlant` loses its pointer to its related `SceneGraph` instance and can no longer automatically register the frame to `SceneGraph`.

The most logical way to unblock this is to change `MultibodyPlant::AddFrame()`'s semantics to be pre-finalize only.

@amcastro-tri Is there an obvious reason why we can't require frames prior to finalization?

jwnimmer-tri, 2024-01-09:
> Is there an obvious reason why we can't require frames prior to finalization?

For the record -- in pondering this question, I searched both Drake and Anzu and did not find any post-Finalize calls to AddFrame.  So, it seems safe to change it to require pre-Finalize.

Later, I realized that it seems to _already_ be disallowed in the tree code:

https://github.com/RobotLocomotion/drake/blob/eb2958a177b7fa05f1fae0bb32179e28aed2a27a/multibody/tree/multibody_tree_topology.h#L580-L592

Maybe the plant docs are just a bit stale and failed to mention (or check?) the requirement.

SeanCurtis-TRI, 2024-01-09:
Ha. I'd dug into the multibody *tree*, but hadn't dug all the way down to the topology. So, yes, it seems like the requirement was retconned in.

marip8, 2025-01-30:
I'm currently running into some challenges regarding this issue. In my current workflow (with Drake v1.36), I define quite a few frames *without* geometry in my SDF files; these correspond to virtual locations (e.g., tool center points, camera frames, etc.) for various uses in planning and simulation. My hope was to use the scene graph to be able to query transforms between any type of frame, not just those containing geometry. Right now, I have to write quite a lot of extra code to pass around `FrameId`s and corresponding body frame offset poses in order to do the transform queries that I need. A lot of this complexity would go away if the plant registered all frames with the scene graph.

Has there been any further thought about resolving this issue? Or is there some other workaround that can simplify querying transforms between all types of frames (e.g., denoting virtual frames as links without visual or collision geometry in the SDF such that they show up as body frames, etc.)?