RobotLocomotion/drake #15004: Missing -DHAVE_SPDLOG when used as CMake external

jwnimmer-tri, 2021-05-04:
In the `drake_cmake_installed` or `drake_cmake_external` workflows (per https://github.com/RobotLocomotion/drake-external-examples) on Ubuntu 20.04, we are compiling Drake's shared library with `HAVE_SPDLOG` defined (so that is has logging support), but then in the client project they do a "find_package" for spdlog, which does not define that directive, and so `drake/common/text_logging.h` degrades into no-op mode for any new calls, and does not let the user configure the log destination anymore.

At build time once we choose a spdlog on/off, we need to sure that same setting propagates into our installed headers, binaries, or cmake config as well.

This is not a problem on 18.04, since we ship our own spdlog.

It's likely a problem on macOS also since we use homebrew spdlog, but I haven't tested that.

jamiesnape, 2021-05-04:
Probably adding something along the lines of `COMPILE_DEFINITIONS/INTERFACE_COMPILE_DEFINITIONS HAVE_SPDLOG=1`

jwnimmer-tri, 2021-05-04:
Yup, something like that would be right, presuming we keep the toggle.  I was also thinking of inverting the logic, so that spdlog is opt-out, since `bazel_drake_external` are the only people who can skip it in the first place.

jamiesnape, 2021-05-04:
> I was also thinking of inverting the logic, so that spdlog is opt-out, since `bazel_drake_external` are the only people who can skip it in the first place.

I think that is reasonable, though I am all for simplification if we can lose the toggle.

jwnimmer-tri, 2021-05-04:
A fair question.  It's definitely still in use internally at TRI, but I'll also do a sweep to see if we can change that around somehow.  Maybe now that it's part of Focal that's good enough.  Since it's in `//common:essential` its one of the dependencies we need to be most careful about.  I've also been pondering demoting it to `text_logging.cc` and re-implementing the ~7 formatter methods in the header ourselves, so that we can reduce it to a linker dependency instead of interface dependency.

PR #16044 (2021-11-05): include HAVE_SPDLOG compile definition for cmake consumers
Fixes #15004.

I could not follow the conversation about turning on or off the "toggle" for spdlog.  Currently, that CMake file assumes that `spdlog` is a hard dependency -- `find_dependency` more or less the same thing as `find_package(... REQUIRED)`.

To make spdlog optional

1. Please tell me how to actually do that with bazel :upside_down_face:
2. We'll need to `configure_file` on the `drake-config.cmake` to optionally `find_dependency` on `spdlog` and conditionally add that interface compile definition.

Sample consumer code to va [... cut]
closed 2021-11-05 by EricCousineau-TRI