RobotLocomotion/drake #15006: OpenCLPointCloudVoxelizer fails on Catalina

calderpg-tri, 2021-05-04:
From #15005, this test no longer passes on macOS [(link)](https://drake-cdash.csail.mit.edu/test/192474043) after a recent Apple upgrade [(link)](https://discussions.apple.com/thread/252710156) and has been disabled.

cc @jwnimmer-tri @jamiesnape

calderpg-tri, 2021-05-06:
To elaborate, OpenCL device *enumeration* still appears to work on Catalina, but the ability to *compile* OpenCL kernels was broken by Security Update 2021-002, so this test identifies devices on which to run but then fails to run on them.

Note that OpenCL-accelerated pointcloud voxelization continues to work on Big Sur and Ubuntu.

calderpg-tri, 2021-05-26:
My understanding of the latest [update](https://shapeof.com/archives/2021/4/apple_april_2021_security_update_and_acorn.html) ([Security Update 2021-003](https://support.apple.com/en-us/HT212530)) is that Apple has fixed the bug in Catalina and we can reenable this test.

@jamiesnape How quickly are the mac CI machines updated?

jamiesnape, 2021-05-26:
I am just (coincidentally) updating them, so later today.

PR #15095 (2021-05-26): Reenable VGT test now that Catalina OpenCL is fixed
This test was previously disabled due to a bug introduced by Apple security update 2021-002 on Catalina, which caused OpenCL kernel compilation to fail. Now that the bug has been fixed by security update 2021-003, it can be reenabled.

See #15006 and #15005 for more details.

_Edit: Closes #15006._

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15095)
closed 2021-05-26 by calderpg-tri