RobotLocomotion/drake #15018: sdformat: Should exercise newer EnforcementPolicy for deprecation in Drake, and make nominal policy strict

EricCousineau-TRI, 2021-05-06:
Given merge of upstream: https://github.com/osrf/sdformat/pull/543

And merge of local: https://github.com/RobotLocomotion/drake/pull/14401

Should then bump version of `libsdformat11` and leverage feature (in same PR?)

jwnimmer-tri, 2021-05-06:
> sdformat: Should exercise newer EnforcementPolicy for deprecation in Drake, and make nominal policy strict

To be clear, is this what you're proposing for Drake's parsers?
- EnforcementPolicy for deprecations => log a warning
- EnforcementPolicy for anything else => throw an exception

EricCousineau-TRI, 2021-05-06:
Relates discussion: https://github.com/RobotLocomotion/drake/pull/14852#pullrequestreview-636839171

Yeah, I think it was option (3) from your post.

EricCousineau-TRI, 2021-06-03:
Upstream:
- Addition of `EnforcementPolicy`: https://github.com/osrf/sdformat/pull/481
- Addition of deprecation policy: https://github.com/osrf/sdformat/pull/543

Notably, unrecognized elements are actually a separate code path from unrecognized attributes:
* Config flags: https://github.com/osrf/sdformat/blob/sdformat11_11.1.0/include/sdf/ParserConfig.hh#L131-L161
* Code handling unrecognized elements (not part of custom `xmlns`): https://github.com/osrf/sdformat/blob/sdformat11_11.1.0/src/parser.cc#L1388-L1408
* Code handling unrecognized attributes: https://github.com/osrf/sdformat/blob/sdformat11_11.1.0/src/parser.cc#L1049-L1066

Given this, I will say that this issue should stay open until we handle unrecognized attributes. That should be handled as a PR separate from #15075 (since it's more-or-less been proven out, and has shown value).
I have added some checkboxes to that effect.

EricCousineau-TRI, 2021-06-04:
Also, see: https://github.com/osrf/sdformat/issues/583

I'm not sure if I forgot the rationale for this, but it's a bit odd to have extra top-level elements be governed by `EnforcementPolicy` only when it's via `//include`, and not at a root level.
\cc @azeey

EricCousineau-TRI, 2021-07-02:
Just ran into a case where at least a warning here would've been useful.

@marcoag Any chance you'd have time to address this in next two weeks?

marcoag, 2021-07-02:
Sure, I'll get into it asap!

EricCousineau-TRI, 2021-08-11:
I'll file a PR to put a deprecation-compatible date in the TODO listed in PR.

azeey, 2021-09-09:
Per VC, leaving this open until deprecation happens.

jwnimmer-tri, 2022-01-06:
The date has passed, so I think we're ready to make the next PR in this train.  Is anyone already working on that?

marcoag, 2022-01-06:
I can have a look.

EricCousineau-TRI, 2022-01-09:
Excellent, thank you! I've added some additional bullet points to the top-level - I will need to help some on the Anzu side.

EricCousineau-TRI, 2022-01-12:
Per discussion in #16347, reassigning to @rpoyner-tri (\cc @jwnimmer-tri).

rpoyner-tri, 2022-03-18:
Given ongoing Adoption work, @jwnimmer-tri is likely to get to this first.

jwnimmer-tri, 2022-04-26:
@EricCousineau-TRI the subject line says this:

>  sdformat: Should exercise newer EnforcementPolicy for deprecation in Drake, and make nominal policy strict.

By "strict", I think you mean that when a user loads deprecated SDFormat syntax using Drake's `multibody::Parser`, then Drake should throw an exception, rather than only printing a console warning?

Assuming that's the case, then in the discussion threads here I could not find any discussion about _why_ the deprecation policy should be strict?

For both C++ API or Python API that we deprecate, we default to only displaying warnings, not halting the user's program.  If the user wants to _opt-in_ to more strictness in those cases (`-Werror=deprecated-declarations`, `warnings.simplefilter`), they can do so.

By analogy, I think that using deprecated SDFormat syntax from Drake should result in a warning, with an opt-in option to promote it to an error (#17017).

To be clear, I do think it's good to opt-in to warnings-promotion for CI canary builds, in order to call attention to the future removal.  That goes for both API function deprecations, and SDFormat syntax deprecations.  However, I do _not_ support turning on such strictness by default in downstream uses of Drake.

Possibly what you meant here is that Drake's own `*.sdf` model files should be warnings-free (in the same way that we don't allow internally-triggered deprecation warnings in C++ or pydrake code to ever leak out to users)?

In any case, can you clarify the objective here?

EricCousineau-TRI, 2022-04-27:
My original intent was to only permit valid SDFormat files to be used in Drake, but given the newer diagnostic work introduced, I think it makes sense to relax this.

So yeah, the objective here should now be to ensure both Drake + Anzu have warning-free SDFormat files.
Have y'all already made CI canary builds available for Drake -- or make parsing strict for Drake CI itself? If so, would love to mirror that mechanism to Anzu.

FWIW In Anzu, I've just added some dumb patches for the new diagnostic policy:
```patch
diff --git a/multibody/parsing/parser.cc b/multibody/parsing/parser.cc
index 3ece8e8518..c3606371df 100644
--- a/multibody/parsing/parser.cc
+++ b/multibody/parsing/parser.cc
@@ -24,6 +24,8 @@ Parser::Parser(
     : plant_(plant) {
   DRAKE_THROW_UNLESS(plant != nullptr);

+  is_strict_ = true;  // HACK
+
   if (scene_graph != nullptr && !plant->geometry_source_is_registered()) {
     plant->RegisterAsSourceForSceneGraph(scene_graph);
   }
```

Lemme know if you think there's a better way (e.g. a global / envvar for CI canary builds to use).

jwnimmer-tri, 2022-04-28:
Yes, for the purposes of Drake CI, my plan is to have a separate linter so that warnings do not get in the way of any other tasks.  Having warnings unconditionally promoted would be a problem in cases where we have unit test tests over deprecated syntax.

For downstream projects that want a canary build (as they would do with, e.g., `-Werror=deprecated-declarations`) we might want to offer an environment variable to that effect.  TBD.

jwnimmer-tri, 2022-04-28:
FYI In any case, I intend to open a separate issue to track the goal of keeping warnings (and errors!) out of Drake's model files.

I have one more PR on the way before that, which should close out this issue.

PR #17060 (2022-04-28): [parsing] Invalid SDFormat syntax is now an error (not warning)
Closes #15018.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17060)
closed 2022-04-28 by jwnimmer-tri