RobotLocomotion/drake #15019: solvers: OSQP requires a Quadratic Cost, but doesn't tell the user that.

RussTedrake, 2021-05-06:
A student was getting the message:
```
ValueError: The capabilities of drake::solvers::OsqpSolver do not meet the requirements of the MathematicalProgram ({ProgramAttributes: LinearConstraint, LinearEqualityConstraint})
```
for a very simple problem, which had only linear constraints and linear equality constraints.  Looking at the code, I found a hidden:
```
prog.required_capabilities().count(ProgramAttribute::kQuadraticCost) > 0;
```
I believe that they needed to add a Quadratic Cost to satisfy the requirements.  Can we make a better error message?

(I can try to come up with a repro after office hours).

hongkai-dai, 2021-05-06:
Will send a PR to throw a better message.

RussTedrake, 2021-05-06:
Just for completeness
```
from pydrake.all import MathematicalProgram, OsqpSolver

prog = MathematicalProgram()
x = prog.NewContinuousVariables(1)
prog.AddLinearConstraint(x[0] == 1)

solver = OsqpSolver()
solver.Solve(prog)
```
does, indeed, give the error message above.

PR #15028 (2021-05-18): solvers: Amend SolverInterface for more detailed error messages
This is breaking change; any direct subclasses of SolverInterface need to implement the new ExplainUnsatisfiedProgramAttributes method.

Suggested order for code review:
- osqp_solver_test
- osqp_solver
- linear_system_solver_test
- linear_system_solver
- solver_interface
- solver_base and its unit test
- program_attribute and its unit test
- pydrake changes

Resolves #15019.  Closes #15020.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15028)
closed 2021-05-18 by rpoyner-tri