RobotLocomotion/drake #15060: AD in Drake

devesh1508, 2021-05-19:
>Hi I have a question regarding automatic differentiation (AD) in Drake. If someone writes the DAE for a system in python, how does Drake compute the derivatives for optimization? Does it have a AD package at the backend or it uses finite differences. Also, how does it implement integration and differentiation for quaternion representation. Thanks a lot for your time in advance.

SeanCurtis-TRI, 2021-05-19:
This kind of question is far better suited to a stackoverflow question with the drake tag. Here's a recent question as an example: https://stackoverflow.com/questions/67604078/pydrake-about-penetration-in-discrete-time-simulation.

We strongly encourage these kind of questions to be asked in StackOverflow, because it's easier for the larger community to benefit from the question and answers. Drake issues are best reserved for bug reports, feature requests, and the like.

devesh1508, 2021-05-19:
Thanks, I posted it here https://stackoverflow.com/questions/67608144/automatic-differentiation-in-drake

SeanCurtis-TRI, 2021-05-19:
Phenomenal. I'll close this in favor of that stack overflow.

closed 2021-05-19 by SeanCurtis-TRI