RobotLocomotion/drake #15061: Templatize RandomSource LeafSystem, add a port for density

hongkai-dai, 2021-05-19:
Currently [RandomSource](https://github.com/RobotLocomotion/drake/blob/master/systems/primitives/random_source.h) system is only templated on double. It draws a sample from the Gaussian distribution and output that sample. I would like to add the two following features to this class

1. Templatize this class, such that it supports both double and AutoDiffXd. When it is templated with ADS, the sample output carries no gradient.
2. Add a `density` port, which reports the density/probability of that sample. When `RandomSource` is templated with ADS, that density port also carries the gradient of the density with the sample, namely `d p(x) / dx` where `x` is the sample. This is gradient information is crucial in the policy-gradient based reinforcement learning.

cc @RussTedrake @jwnimmer-tri

jwnimmer-tri, 2021-05-19:
FYI Here's the starting point for how to immediately gain access to this feature, while not making any breaking changes.

This approach buys us four things:
1. Users with a `RandomSource` in their double-valued diagrams will be able to scalar-convert the diagram to AutoDiffXd without any new hassles.
2. Users will be able to call `AddRandomInputs` on an AutoDiffXd diagram directly, inserting AutoDiffXd-valued random sources.
3. Users who only ever called `AddRandomInputs` (and not the constructor directly) will not need to make any changes.
4. Users who create `RandomSource` systems directly will receive a deprecation warning, to add a `d` suffix to the class name at their call sites.

The remaining work on this patch would be to fill in the `RandomSourceT` code so that it compiles, while leaving the `RandomSource` API unchanged, including making sure that its Doxygen page does not lose any information.  Also be sure to pass a `SystemTag<RandomSource>` down into the `LeafSystem` constructor, and add tests.

```patch
diff --git a/systems/primitives/random_source.h b/systems/primitives/random_source.h
index 38c00a957..c6e4ba0a5 100644
--- a/systems/primitives/random_source.h
+++ b/systems/primitives/random_source.h
@@ -4,6 +4,7 @@
 #include <vector>
 
 #include "drake/common/drake_copyable.h"
+#include "drake/common/drake_deprecated.h"
 #include "drake/common/random.h"
 #include "drake/systems/framework/diagram_builder.h"
 #include "drake/systems/framework/leaf_system.h"
@@ -11,6 +12,18 @@
 namespace drake {
 namespace systems {
 
+namespace internal {
+template <typename T>
+class RandomSourceT : public LeafSystem<T> {
+ public:
+  DRAKE_NO_COPY_NO_MOVE_NO_ASSIGN(RandomSourceT)
+
+ protected:
+  RandomSourceT(RandomDistribution, int, double);
+  ~RandomSourceT() override;
+};
+}  // namespace internal
+
 /// A source block which generates random numbers at a fixed sampling interval,
 /// with a zero-order hold between samples.  For continuous-time systems, this
 /// can be interpreted as a band-limited approximation of continuous white noise
@@ -69,7 +82,11 @@ namespace systems {
 /// @see @ref stochastic_systems
 ///
 /// @ingroup primitive_systems
-class RandomSource final : public LeafSystem<double> {
+class DRAKE_DEPRECATED("2021-09-01",
+  "Re-spell uses of this class as RandomSourced, not RandomSource.  In the"
+  " future, this class will gain a template argument <typename T> in order to"
+  " support multiple scalar types.")
+RandomSource final : public internal::RandomSourceT<double> {
  public:
   DRAKE_NO_COPY_NO_MOVE_NO_ASSIGN(RandomSource)
 
@@ -111,17 +128,25 @@ class RandomSource final : public LeafSystem<double> {
   std::optional<Seed> fixed_seed_;
 };
 
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wdeprecated-declarations"
+/// A convenient alias for a RandomSource that uses the `double` scalar type.
+using RandomSourced = RandomSource;
+#pragma GCC diagnostic pop
+
 /// For each subsystem input port in @p builder that is (a) not yet connected
 /// and (b) labeled as random in the InputPort, this method will add a
 /// new RandomSource system of the appropriate type and connect it to the
 /// subsystem input port.
 ///
+/// @tparam_nonsymbolic_scalar
 /// @param sampling_interval_sec interval to be used for all new sources.
 /// @returns the total number of RandomSource systems added.
 ///
 /// @see @ref stochastic_systems
+template <typename T>
 int AddRandomInputs(double sampling_interval_sec,
-                    DiagramBuilder<double>* builder);
+                    DiagramBuilder<T>* builder);
 
 }  // namespace systems
 }  // namespace drake
```

jwnimmer-tri, 2021-05-19:
We can start working on the code to add the template, and add the new port, (and I'd suggest making a separate PR for each of those changes, in that order), but I think the plan for gradients is not yet clear to me.

> When RandomSource is templated with ADS, that density port also carries the gradient of the density with the sample, namely d p(x) / dx where x is the sample.

I understand that outcome, and it might be a valid one, but I don't yet understand the mechanism you'd want to use for it.

The selection of the autodiff's partials' indices is established by the _application code that populates the context_.  It's not always true that the partial of the RandomSource's 0th output should be in the 0th row of the derivatives vector.  Perhaps the user is taking partials with respect to more than just the source's density.

I wonder if the source should never set any partials, but instead the _downstream system_ that is shifting or scaling the distribution should use Parameters for those factor(s), and thus we could introduce the partials at that point, instead?

jwnimmer-tri, 2021-05-19:
Hmm, introducing the partials only in the downstream system works for the random value, but not really conveniently for the density.  I have a few ideas for how to resolve this, but I need more information about where `d p(x) / dx` needs to end up before I could recommend one.

hongkai-dai, 2021-05-19:
> The selection of the autodiff's partials' indices is established by the application code that populates the context. It's not always true that the partial of the RandomSource's 0th output should be [...]

Sorry I wasn't clear, and I completely agree with you that the user can set the partial gradient w.r.t whatever variable they want, not necessarily the sample value. 

How about we have 3 output ports in this RandomSource system
1. One output port for a sample
2. One output for the density (pdf) of that same sample from port 1.
3. One output for the gradient of the density w.r.t the sample.

The output of each port templated on double and AutoDiffScalar. But the AutoDiffScalar version is just casting the double value to the AutoDiffScalar type with empty gradient.

jwnimmer-tri, 2021-05-19:
I think that's a clear and unambiguous specification.  I wonder how awkward (or not) it is to use that formulation in your application's code.  Maybe we can look at your diagram / application offline.

hongkai-dai, 2021-05-19:
Thinking more carefully about my use case, I think I made a mistake in my statement above.

The very common use case is that I will have a parameterized probability density function `p(x;θ)`, where θ is the parameter of this distribution, `x` is a sample drawn from this distribution. One example is that for a normal distribution x∼ N(μ, σ), where μ is the mean of this distribution, and σ is the variance of this distribution. Here [μ, σ] is the parameter θ of this distribution. I draw a sample `x` from this normal distribution, and I want to know how the probability of drawing `x` would change, if the mean/variance of this distribution changes (intuitively, if I move the mean `μ` closer to `x`, then the probability of drawing `x` would be larger). Such knowledge can be obtained by computing the gradient `dp(x;θ)/dθ`. This use case is not covered in the discussion above. 

Here is a use case in robotics:
I have a linear controller `u = K * x + w`, where `w` is a noise term, drawn from a Gaussian distribution `w∼N(0, 1)` (a zero-mean, unit variance Gaussian distribution. I can build this controller with the following diagram
```
LinearController  K * x   ------->
                                         Adder --->
RandomSource generates w     ------->
```
The output port of the adder gives me the controller `u = K * x + w`. I know this output `u` is also going to be a normal distribution, with pdf N(K*x, 1), namely its mean is `K*x`, with a unit variance. Note that now the control gain `K` is a parameter of this normal distribution. Now I want to know if I change the control gain `K` a little bit, how does it change the probability of `u`. Namely I want to compute the gradient `d p(u) / dK`. I don't think this gradient can be obtained from the proposal above, as currently `RandomSource` only knows the information about the zero-mean Gaussian distribution N(0, 1). It knows nothing about the linear control part `K * x`.

I would think that the solution for this particular problem is to create a leaf system that contains both the linear controller and the noise distribution. It has one output port for computing `u = K * x + w`, and another function (but not an output port) to compute the density.
```
template <typename T>
class NoisyLinearController : LeafSystem<T> {
// Evaluate u = K * x + w.
VectorX<T> output(const Context<T>& context) const;

// Evaluate the density p(u)
T density(const VectorX<T>& u) const;
};
```
When computing this system with AutoDiffScalar, and if we want to evaluate the gradient `d p(u) / dK`, we first evaluate `u` whose derivative stores the gradient `du/dK`, and then the density function can easily compute the gradient `d p(u) / dK` through chain rule.

RussTedrake, 2021-05-20:
So I actually thought quite a bit about this when I first wrote `RandomSource`; and am remembering now why I decided not to implement the density output port.  It gets tricky fast!

Just to recall the original design decisions:
1) any random signals in a diagram originate at a `RandomSource`
2) `RandomSource` blocks output random variables in the trivial scaling (e.g. N(0,I)), but those outputs can be scaled with `Gain` blocks, etc.  
3) A system can mark an input port as expecting a `RandomSource` of a particular type (`kUniform`, `kGaussian`, etc).
The reasoning for (2-3) was that we didn't want to have to handle every possible parameter in every possible algorithm -- for instance in the Kalman filter code we know from the input port of a system that we want a Gaussian random variable to be wired up, and we know from the gradients of e.g. the state update relative to that input port what the covariance matrix is.

But, of course, probability densities are more complicated.  If the random variable is pushed through a linear transform, then the density needs to be [scaled appropriately] (https://www.math.hkust.edu.hk/~maykwok/courses/ma246/246t2dnew.pdf).

If you follow this line of reasoning, then it becomes very tempting to try to make a scalar type for a random variable.  But a random variable is not a scalar type (the computation needs to track conditional dependencies between different variables, otherwise simple operations like `x+x` could give very wrong answers).  That route leads to a full probabilistic programming language.  After talking about it a bunch, even with some colleagues who live and breath inference, we decided to just not offer the densities... because in practice it's very rare for people to do useful work with complex transformations of random variables anyhow.  It's not worth the hassle.

But you are right about the particular use case you have being an important one.  Specifically, I think that taking gradients of the density with respect to the mean and covariance are basically the only gradients that people can meaningfully take for Gaussians.  We should probably not support more general/complicated functions of that variable.

So here is one proposal: 
a) add a second output port to `RandomSource` for p(x).  Possibly document it with a warning about how challenging it is to propagate densities properly.
b) create a new primitive system `LinearTransformDensity` that has two input ports (one for the random variable and a second for the density ) and two output ports ( the random variable and the density ).  It would declare parameters for A and b, and the `AutoDiffXd` version of this system would know how to take gradients of both outputs with respect to those parameters.  One could then consume the outputs of that system in any policy... not only linear policies.  The `LinearTransformDensity` would take an argument in it's constructor for `RandomDistribution` so that it could properly declare the input port.

N.B. The alternative design for stochastic systems that I considered initially did not have systems mark their input ports as random explicitly, but instead pass a diagram into e.g. the Kalman filter algorithm with a `RandomSource` block already wired up.  That algorithm would then have to dig through a diagram and find all included sources of randomness.  We didn't like that idea before for a few subtle reasons, which I would want to reconstruct before we tried to revisit it.  But in that pattern, we could accomplish your goals by just adding mean and covariance parameters to the `GaussianRandomSource` system and avoid the new `LinearTransformGaussian`.

I think the proposal above it better.  I'll keep thinking on it, too.

hongkai-dai, 2021-05-20:
I like the `LinearTransformDensity` idea! I think it well captures my use case.

Just one question, I am not sure if the linear transformation matrix `A` and `b` should be parameters. As far as I know, the parameters are fixed during the simulation. But in that linear controller example `u = K*x + w`, `A = I` and `b=K*x`, since the state `x` will change in time, now the matrix `b` will also change during simulation.

I agree that it is daunting to try to capture all the probablistic programming features, for example, if we have two random variables `x1` and `x2`, computing the density of `x1 + x2` requires the convolution of the density p(x1) and p(x2), which doesn't have the closed form solution in general.

RussTedrake, 2021-05-20:
A and b could certainly be inputs to the system, instead of parameters.  But are you actually changing them during a single roll-out?  Are the things that you are going to modify using the gradient information (e.g. controller parameters) also `const` during a simulation?

Also, I realized (as I slept on it!): `RandomSource` does not need to output the density after all.  Since the `LinearTransformDensity` class knows that it's e.g. a mean-zero unit-covariance Gaussian coming in, it has everything it needs to know with just the sample value.  (Would that change the name, then, to e.g. just `ProbabilityDensity`?). That simplifies the proposal considerably.

But following this logic, any system with a random input port could also implement this, if they so desired, because they've labeled their input port to have a known distribution.  So we could also offer this feature as just a function, available inside any system method w/o needing input/output ports?

ggould-tri, 2021-05-20:
This issue had no assigned team, so I am assigning Hongkai/manipulation; Hongkai, you can reassign either of those as you like.

jwnimmer-tri, 2021-05-20:
This is what I came up with while reflecting on the above.

I buy Hongkai's case that there's a plausible reason for the `a` and `b` terms to be inputs, for example, because `b` might depend on `x`, and `x` or `Kx` might be most conveniently found on an output port, instead of the user writing code to repeatedly copy it into a system parameter.  I suspect it would also more easily pave the way for convenient initialization of the original partials on `K` when computing `∂/∂K`, by simply `initializeAutoDiffTuple` on `K`, which would be a declared system parameter of the controller block.

I think it does make sense to have static functions available to implement the density functions without any Systems around, but I think those would live in `drake/common/random.h` for anyone to use.  We don't need the user to depend in `//systems/primitives` in order to use those.

A proposal for `drake/primitves/scaled_random_source.h` ...
```cxx
/** A block that applies linear scaling to the output of a RandomSource, and
also makes available the probability density of the output.

  w_out = a * w_in + b
  density = p(w_out)

where w_in an input is supplied by a RandomSource (with trivial scaling), and
`a` and `b` are additional input ports.  (See FixInput or ConstantVectorSource
for how to set `a` or `b` to constants.)

Refer to RandomSource for its discussion of zero-order hold, band-limited
approximation of continuous white noise, and related.

@system
name: ScaledRandomSource
input_ports:
- w_in
- a
- b
output_ports:
- w_out
- density
@endsystem

The `a` port can remain disconnected, in which case it defaults to identity.
The `b` port can remain disconnected, in which case it defaults to zero.

@see @ref stochastic_systems

@ingroup primitive_systems
*/
template <typename T>
class ScaledRandomSource final : public LeafSystem<T> {
  // ...
};
```

It might be clearer to say `w_trivial` and `w_scaled` instead of `w_in` and `w_out`?
We could also say `gain` and `bias` instead of `a` and `b`?

RussTedrake, 2021-05-20:
Looks mostly good to me.  To be clear, I think `A` should be a matrix (I'm ok with `Gain`) and `b` a vector, because the off-diagonal terms are important for capturing covariance.

It's slightly annoying to take that matrix through an input port (AbstractValue? vectorized matrix?), I guess.  

I'm afraid don't love `ScaledRandomSource`, since it does more than scaling and since it's not actually the source.

jwnimmer-tri, 2021-05-20:
Yes, I had `A` as a matrix and `b` as a vector in my head, but I didn't connect the dots that it meant `A` would come in on a vector input port.  That is annoying.  (We will probably need it to be a numeric port, not abstract, because we want double <=> autodiff conversions to operate correctly.)  This is another use case for multi-dimensional numeric ports (and parameters, and ...).

We could offer a `ScaledRandomSource::FixInput(context)` helper that took in `A` as a `Matrix` and reshaped it correctly when writing to the context.

I don't love the class name either.  I liked "Source" only because, from the point of a user adding it to their Diagram, it is (in a sense) the actual source -- the `AddRandomInputs` for simulation (or analysis logic, for design) is what actually plugs in the clocked seed to feed this system, but the user should not be messing with that themselves.  I do see how it can be misleading, though.  I didn't like any other of the names upthread, though, either.  Requires more thinking, I guess.

hongkai-dai, 2021-05-24:
I agree that `ScaledRandomSource` might not be the best name, first we do more than scaling, but a linear transformation with potentially an offset term `b`. Second I think `Source` implies this system doesn't have an input.

How about `LinearTransformRandomSignal`? As Russ mentioned above, the system doesn't have to contain an output port for density (although I think that density output port would be convenient). If we only have a density input, but not density output, while we have random signal as both input and output, maybe we should use `RandomSignal` in the class name, instead of `density`?

RussTedrake, 2021-05-24:
I like that name.  Fwiw, if you just want the density from the original distribution you could still use this block (with the identity transform).

hongkai-dai, 2021-05-25:
I am trying to add the feature in a two-step process:
1. Modify `RandomSource` class. As Jeremy mentioned in https://github.com/RobotLocomotion/drake/issues/15061#issuecomment-844317029, we can templatize `RandomSource` class. Also I think we need to add a `density` output port, that reports the pdf of that particular sample.
2. Add a templated `LinearTransformRandomSignal` class, whose input ports include
    - A sample of the random signal `x`
    - The density of the input signal `x`.
    - (optional) The linear transformation matrix `A`.
    - (optional) The linear transformation matrix `b`.
    Its output include
    - The output random signal `A*x+b`
    - The density of the output signal.

@RussTedrake @jwnimmer-tri did I miss anything?

jwnimmer-tri, 2021-05-25:
I thought we agreed that `LinearTransformRandomSignal` does not need to receive the density on an input port ([_RandomSource does not need to output the density after all_](https://github.com/RobotLocomotion/drake/issues/15061#issuecomment-844900340))?  It knows which `RandomDistribution` is being used (via constructor argument), and knows `x` from the input port.  Those alone are sufficient to compute the transformed density.

If we added another output port to `RandomSource`, the wiring becomes much more complicated.  The `AddRandomInputs` logic would become more difficult -- you'd need some kind of label or heuristic for which density ports should be connected when it adds the `RandomSource`.

In any case, for the templating part (adding `T=AutoDiffXd` support to `RandomSource`), please do that in a standalone PR, before adding any new features onto `RandomSource` or elsewhere.

Your plan seems okay, otherwise.

The matrix-valued `A` is annoying as input.  We'll want to provide sugar to fix a `MatrixX<T>` into the context to set it (and we might as well do `b` while we're at it)?

hongkai-dai, 2021-05-25:
> I thought we agreed that LinearTransformRandomSignal does not need to receive the density on an input port (RandomSource does not need to output the density after all)? It knows which [...]

Ah you are right, I somehow forgot that part. I have updated the plan in https://github.com/RobotLocomotion/drake/issues/15061#issuecomment-848339289

hongkai-dai, 2021-05-26:
@jwnimmer-tri In the patch, I don't quite get why we put the new class `RandomSourceT` in the internal namespace. It seems to me that there is no harm to put it in the `systems` namespace? If I recall correctly, we don't add python bindings for object in the `internal` namespace, but I do want to bind `RandomSourceT<AutoDiffXd>`

jwnimmer-tri, 2021-05-26:
The classname we want in the future will be `RandomSource<T>`.  If we introduced `RandomSourceT<T>` in a public namespace now, we'd have to deprecate it again a few months.  We _could_ certainly do that, but since _nobody should ever create this class by hand_, there is no point.  Everyone should use `AddRandomInputs`, which will support autodiff)

hongkai-dai, 2021-06-02:
@RussTedrake when we compute the density of `w_out` as `w_out = A * w_in + b`, I think it is challenging if `A` is not invertible. In general to apply the idea in https://www.math.hkust.edu.hk/~maykwok/courses/ma246/246t2dnew.pdf I think `A` need to be an invertible matrix. For special case when `w_in` is from a Gaussian distribution, even if `A` is non-invertible, we know that `w_out` must also be a Gaussian and we can compute its distribution, but I think in general if `w_in` is not Gaussian, then computing the density of `w_out` is hard given non-invertible `A`.

My plan is that in `LinearTransformDensity::Density(const VectorX<T> w_out_sample)` function, we compute the density when the input is from a Gaussian, or when `A` is invertible. Otherwise I will throw a runtime error. Does this sound good to you?

RussTedrake, 2021-06-03:
I took a few minutes to look, and yes, I agree with that assessment.  Nice.  Thanks!

RussTedrake, 2021-07-30:
@hongkai-dai -- should this be closed?

hongkai-dai, 2021-07-30:
Yes, it has been resolved.

closed 2021-07-30 by hongkai-dai