RobotLocomotion/drake #15062: Missing (linking) reference for RotationMatrix constructed with certain AutoDiff types

edrumwri, 2021-05-20:
We're using getting a linking error when attempting to implement some code that computes a Hessian matrix, FYI, using drake::math::hessian(.). 

Seems to be a SFINAE failure (substitution failure is not an error, except when it is...)

Smallest example showing the failure is this diff to the rotation matrix test in 90af862:

```
diff --git a/math/test/rotation_matrix_test.cc b/math/test/rotation_matrix_test.cc
index bc29bf914..6a4f6aab8 100644
--- a/math/test/rotation_matrix_test.cc
+++ b/math/test/rotation_matrix_test.cc
@@ -628,6 +628,11 @@ GTEST_TEST(RotationMatrix, ProjectToRotationMatrix) {
   EXPECT_TRUE(R.inverse().IsNearlyEqualTo(R.transpose(), 8 * kEpsilon));
 }
 
+GTEST_TEST(RotationMatrix, ConstructRot) {
+ using T = Eigen::AutoDiffScalar<drake::Vector2<Eigen::AutoDiffScalar<drake::Vector2<double>>>>;
+ Matrix3<T> m;
+ RotationMatrix<T> R(m); 
+}
 
 // Test RotationMatrix cast method from double to AutoDiffXd.
 GTEST_TEST(RotationMatrix, CastFromDoubleToAutoDiffXd) {
```

Building yields this error:
```
└─ $ ▶ bazel build -c dbg math:rotation_matrix_test
INFO: Analyzed target //math:rotation_matrix_test (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
ERROR: /tmp/drake/math/BUILD.bazel:488:20: Linking of rule '//math:rotation_matrix_test' failed (Exit 1): gcc failed: error executing command /usr/bin/gcc @bazel-out/k8-dbg/bin/math/rotation_matrix_test-2.params

Use --sandbox_debug to see verbose messages from the sandbox gcc failed: error executing command /usr/bin/gcc @bazel-out/k8-dbg/bin/math/rotation_matrix_test-2.params

Use --sandbox_debug to see verbose messages from the sandbox
bazel-out/k8-dbg/bin/math/_virtual_includes/geometric_transform/drake/math/rotation_matrix.h:352: error: undefined reference to 'drake::math::RotationMatrix<Eigen::AutoDiffScalar<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, 2, 1, 0, 2, 1> >, 2, 1, 0, 2, 1> > >::ThrowIfNotValid(Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, 2, 1, 0, 2, 1> >, 2, 1, 0, 2, 1> >, 3, 3, 0, 3, 3> const&)'
collect2: error: ld returned 1 exit status
Target //math:rotation_matrix_test failed to build
```

SeanCurtis-TRI, 2021-05-20:
You've run face first into Drake's very *designed*, very limited support of scalar types. More `RotationMatrix` code has moved from the .h file into the .cc file. That movement further cements the policy of which scalar types are supported. The function your linker can't find moved in #14860. Depending on how you exercise `RotationMatrix`, there would be other functions your linker would bark about. 

That does leave you out in the code as you are trying to use compile-time sized derivatives. That will need to be a larger conversation. You could resolve this locally, by applying a patch to your local Drake instance and move the implementations back into the header. Presumably, this scalar type has limited scope, otherwise you'd run into linker errors all over Drake.

jwnimmer-tri, 2021-05-20:
Here's the doc citations:

https://drake.mit.edu/doxygen_cxx/classdrake_1_1math_1_1_rotation_matrix.html:

> Template Parameters
> T | The scalar type, which must be one of the default scalars.

https://drake.mit.edu/doxygen_cxx/group__default__scalars.html:

> Most scalar-templated classes in Drake only support a small, fixed set of scalar types:
>    double (always)
>    drake::AutoDiffXd (almost always)
>    drake::symbolic::Expression (sometimes)

The discussion at #12814 is related.

In #2619 and #10991 we looked into adding a fixed-size autodiff (or sbo-optimized autodiff) to the list of default scalars, but never crossed the finish line on proving it was worth the complexity via benchmarking.

edrumwri, 2021-05-20:
Ha! I see that I was advocating this change in support of faster compile times in #12814!

We have a high performance application where we don't want the memory allocation, so I just solved the problem by using Matrix3 in place of RotationMatrix.

closed 2021-05-20 by edrumwri