RobotLocomotion/drake #15089: [geometry] Modifiable collision filters

SeanCurtis-TRI, 2021-05-25:
## The problem

Drake has the concept of "collision filters". In short, a Drake user can declare that a pair of geometries (A, B) can be ignored in various pair-wise proximity queries (e.g., penetration, signed distance, etc.)

Currently, Drake can only *accumulate* collision filters. That is to say, once a pair has been declared as "filtered", it cannot be reversed. Furthermore, there are various practical applications in which *temporary* modifications to collision filtering would be convenient.

## The proposal

The current API consists of a family of methods: `ExcludeCollisions{Within|Between}` which add filtered pairs to the overall filter configuration. This current API and is part of the *persistent* API (see below) We need to add two things:

  - A parallel persistent `AllowCollisions{Within|Between}` which can reverse the effect of `ExcludeCollisions`.
  - A transient API for adding/removing collision filters.

### Transient filter modifications

We will facilitate transient filter modifications (changes that can be applied and then easily undone) using the concept of a `FilterDeclaration` -- a mechanism for declaring what pairs should be filtered and what pairs should not. The collision filter configuration is defined by a persistent base state with an arbitrary number of ordered `FilterDeclaration`s applied to it. The current API would modify the persistent base but have no notion of the `FilterDeclaration`.

The `FilterDeclaration` is the basis for the transient filter modification. Each `FilterDeclaration` will have a unique identifier associated with it. `FilterDeclaration`s can be removed from the sequence and the final configuration will be the persistent base state with the *remaining* ordered declarations reapplied (although that is not necessarily how it will be implemented).

There are some further implications of this design:

  - Include an API to take the current configuration and "bake" it into the base persistent state (invalidating all `FilterDeclaration` ids that had previously been generated).
  - Only the *final* `FilterDeclaration` in the sequence is mutable. Once a declaration is followed by another, it can no longer be modified.
  - We provide introspection queries on `FilterDeclaration` ids: is the given id valid? Is the given id the last one?
  - Modifications to the persistent base cannot be made while there is a "live" declaration sequence.

There are some interesting implications in using this paradigm that must be explicitly called out to users in the documentation (this list is not necessarily complete and I expect it will be edited to include more nuances as they occur to me):

  - When making a filter declaration, the user should *ignore* what the current configuration is and unambiguosly state what result they want. Defining a declaration relative to the configuration at time of declaration can become invalid if a previous declaration in the sequence gets subsequently removed.
  - There is *no* harm in making redundant declarations. When in doubt, declare.

## The plan

- [ ] - Move the collision filter API off of `SceneGraph` and onto a dedicated class `CollisionFilterManager` (deprecating the old API). All new APIs will be placed on this new manager.
- [ ] - Replace the underlying "legacy" filter implementation with a new implementation that will extend toward modifications better.
- [ ] - Extend the API to *permanently* remove filters.
- [ ] - Extend the API to facilitate *temporary* modifications of collision filters.

sherm1, 2021-05-26:
SGTM!

EricCousineau-TRI, 2021-06-17:
FYI I unchecked the box for #15115, given that it was reverted in #15139.
Was confusing when writing release notes (#15207).

PR #15492 (2021-08-20): [geometry] Introduce "transient" collision filter modifications
This introduces the concept of a history of collision filter modifications and the ability to edit that history.

 - Introduce new id type: `FilterId`.
 - `CollisionFilterManager` introduces the concept and provides an API for exercising it.
 - `CollisionFilter` implements the plumbing.
 - Python bindings

resolves #15089

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15492)
closed 2021-08-20 by jwnimmer-tri