RobotLocomotion/drake #15124: solvers: Rename AddL2NormCost

RussTedrake, 2021-06-03:
since the cost that gets added, |Ax-b|^2, is not the L2 norm.  It's the L2-norm squared?

RussTedrake, 2021-06-03:
having an actual L2 norm cost would also be useful, but a bit more subtle (e.g. do we always want the conic version with a slack variable?  or would we just give the nonlinear version to e.g. snopt?).  Also, we probably need to deprecate this name so that people don't have the implementation changed out from under them.

shensquared, 2021-06-03:
Fwiw, in convex land, 2-norm is actually numerically preferred by solvers than quadratic forms (I think exactly due to that second-order cone reformulation). So e.g., http://cvxr.com/cvx/doc/advanced.html#eliminating-quadratic-forms encourages user to explicitly feed in norms whereas some other interfaces would take care of the conversion behind the scene. Not sure if the norm still has a win over the quadratic in e.g., snopt.

hongkai-dai, 2021-06-04:
I plan to do this in two steps
1. Rename `AddL2NormCost` function to `AddL2NormSquaredCost`. Also change our tutorials. Deprecate the old `AddL2NormCost` function.
2. After 3 month of deprecation notice, add back `AddL2NormCost`, which adds actually the L2 norm in the conic form (with slack variable and second order cone constraint).

@jwnimmer-tri , what do you think? Specifically, is it OK to change the functionality of `AddL2NormCost`, by first deprecating it, and then add it back?

jwnimmer-tri, 2021-06-04:
That's acceptable.

The new `AddL2NormSquaredCost` name, and deprecating the old name, are perfect.

However, it would be somewhat better if the forthcoming non-squared cost was added under a distinct signature.  That way (1) even a user who upgrades across a >3 month span would receive a diagnostic, and (2) you could add and experiment with the slack variable formulation right away, without waiting months.

So for example, maybe the non-squared name could be `AddL2NormCostWithSlack`, which then also leaves room for alternative formulations of L2 norm (as noted by Russ).

Alternatively, you could add a new `AddL2NormCost` overload with required enum argument that selects between slack variable vs nonlinear (snopt).  That overload could be added today, and would not be ambiguous with the existing, deprecated overload.

hongkai-dai, 2021-06-04:
I see, that makes sense.

Here is the new plan
1. Rename the old `AddL2NormCost` to `Add2NormSquaredCost`. People use "l2 norm" and "2-norm" interchangeably. Deprecate the old `AddL2NormCost` function.
2. Add a new method `Add2NormCost` for the alternative formulation.

EricCousineau-TRI, 2021-06-07:
My attempt to summarize point here:
https://github.com/robotlocomotion/drake/pull/15134#pullrequestreview-677797782

We don't want to continue using `AddL2NormCost`, even if it were deprecated, removed, then restored. For that reason, we choose `Add2NormCost` for the future, and thus the intermediate `Add2NormSquaredCost`.

Beyond legacy and migration purposes within Drake, there is no defect with the usage of `Add2Norm*` vs. `AddL2Norm*`

PR #15134 (2021-06-07): Rename AddL2NormCost to Add2NormSquaredCost
Resolves #15124

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15134)
closed 2021-06-07 by EricCousineau-TRI