RobotLocomotion/drake #15125: "Illegal instruction (core dumped)" on older laptop

vincekurtz, 2021-06-03:
I'm trying to install drake on an older laptop (ThinkPad E420, Ubuntu 20.04, Intel i5-2430M CPU) and running into an `Illegal instruction (core dumped)` error. This occurs both with a binary installation and when building from source. 

## What I've Tried

Installing Drake from binaries per [these instructions](https://drake.mit.edu/python_bindings.html#installation). I was then able to run the `drake-visualizer` and import `pydrake` successfully. But hit the "Illegal instruction" error when trying to run `RegisterCollisionGeometry`.

I then figured the precompiled binaries might use some CPU instructions that this older machine doesn't support, so I tried compiling from source ([using cmake](https://drake.mit.edu/python_bindings.html#building-the-python-bindings), since I want to use the python bindings). This had the same result (drake-visualizer works, can import pydrake, create a MultibodyPlant, solve some simple MathematicalPrograms, but RegisterCollisionGeometry fails). 

From [this issue](https://github.com/RobotLocomotion/drake/issues/7740) I thought the problem might be reliance on AVX2 and FMA instructions, which this CPU doesn't seem to support (see below). So I recompiled with the line
```
build -c opt --copt=-march=native -- cxxopt=-march=native
```
in `~/.bazelrc` in the hopes that this would force the compiler to only use available CPU instructions, but this gave the same result.

## Steps to reproduce

This simple python script illustrates the problem:
```
from pydrake.all import *

builder = DiagramBuilder()
scene_graph = builder.AddSystem(SceneGraph())
plant = builder.AddSystem(MultibodyPlant(1.0))
plant.RegisterAsSourceForSceneGraph(scene_graph)

X = RigidTransform()
H = HalfSpace()
C = CoulombFriction()

# Everything so far is fine, but the next line will fail
plant.RegisterCollisionGeometry(plant.world_body(), X, H, "test", C)
```

The same error also crops up in various other precompiled binaries. For example,
```
[drake-build]/install/share/drake/examples/kuka_iiwa_arm/kuka_simulation
```
throws the same error.

## Debug info

System info:
- Ubuntu 20.04
- ThinkPad E420
- Intel i5-2430M CPU
- 4GB RAM

Environment variables
- `LD_LIBRARY_PATH=` (empty)
- `PYTHONPATH=[HOME]/builds/drake/drake-build/install/lib/python3.8/site-packages`

Output of `cat /proc/cpuinfo`: [cpuinfo.txt](https://github.com/RobotLocomotion/drake/files/6592261/cpuinfo.txt)

vincekurtz, 2021-06-03:
Quick update: older versions of the Drake binaries (up to and including v.0.29.0) seem to work perfectly fine. The more recent binaries (corresponding to v.0.30.0 and onward) seem to have this issue.

sherm1, 2021-06-03:
I agree this is likely due to the recently-introduced (by me) dependence on AVX2 & FMA instructions. `-march=native` build from source should fix that, but is likely being overridden by the Bazel instructions [here](https://github.com/RobotLocomotion/drake/blob/master/math/BUILD.bazel#L283) where the use of those instructions depends only on whether we are building for Apple machines:
```
drake_cc_library(
name = "fast_pose_composition_functions", 
srcs = ["fast_pose_composition_functions.cc"], 
hdrs = ["fast_pose_composition_functions.h"],
copts = select({
  "//tools/cc_toolchain:apple": [], 
  "//conditions:default": [
  "-march=broadwell",], }
)
...
```
I don't have an older machine to test this on, but I believe if you remove the `copts` line here you'll get a build that works. Please give that a try and post here whether that fixes the problem.

vincekurtz, 2021-06-03:
Thanks for the quick reply! I'll give that a shot and will report back when it's finished (which may be as late as tomorrow, since the compilation process is pretty slow on this machine).

vincekurtz, 2021-06-04:
Yes, commenting out the `copts` line in `math/BUILD.bazel` and compiling with `-march=native` fixes the issue. I was able to compile successfully and everything seems to be working normally now. Thanks!

sherm1, 2021-06-04:
I am leaving this open with a new title as a call for accessible documentation on how to get Drake compiled on older Intel machines.

jwnimmer-tri, 2021-06-04:
Probably the ["Supported Configuration" documentation](https://github.com/RobotLocomotion/drake/blob/8d48ff944f116112e4823d48481352f547b0364c/doc/_pages/developers.md#supported-configurations) should mention that we require AVX2 and FMA when running on Ubuntu.  We seem to have missed that in the original batch of changes, though we did mention it in the [release notes](https://drake.mit.edu/release_notes/v0.30.0.html).

PR #15165 (2021-06-16): Document Drake's dependence on AVX2 and FMA instructions
Document that Drake builds (since v0.30.0) require AVX2 and FMA instruction support.

This does not include detailed instructions for building on pre-2013 machines, but provides enough information for doing so (perhaps with some help from stackoverflow if necessary).

Resolves #15125

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15165)
closed 2021-06-16 by jamiesnape