RobotLocomotion/drake #15170: Memory aliasing problem in MeshDeformer::set_mesh()

DamrongGuoy, 2021-06-10:
<h1>Problem</h1>
After I reviewed PR #15117 and approved its merge into master, I played with unit tests to educate myself about memory management and found that the following function `geometry::internal::MeshDeformer::set_mesh()` is copying the **content** of `*mesh_M` not its **address** to the alias `mesh_`.

```C++
  /* Sets the mesh for this deformer. */
  void set_mesh(MeshType* mesh_M) {
    DRAKE_DEMAND(mesh_M != nullptr);
    mesh_ = *mesh_M;
  }
...
  MeshType& mesh_;
```

This is because `MeshType& mesh_` still refers to the original memory address since the time of constructor. Unlike a pointer, `mesh_` is a reference, so it cannot change to refer to a different object after constructed, I think. (Is that right?)

I confirmed my hypothesis by adding the following unit test, and it failed as expected:
```C++
  void TestSetMesh() {
    MeshType<T> another_mesh = BoxMaker<T, MeshType>().make(Box(4, 5, 6));
    deformer_.set_mesh(&another_mesh);
    EXPECT_EQ(&deformer_.mesh(), &another_mesh);
  }
...
TYPED_TEST(VolumeMeshDeformerTest, SetMesh) {
  this->TestSetMesh();
}
```

<h1>Proposed Solution</h1>
I think a simple solution is to remove `set_mesh()` function and also remove any calls to `set_mesh()`.  In PR #15117, we have calls like this:

```C++
  DeformableVolumeMesh& operator=(DeformableVolumeMesh&& other) {
    mesh_ = std::move(other.mesh_);
    deformer_.set_mesh(&mesh_);
    return *this;
  }
  ...
  VolumeMesh<T> mesh_;
  MeshDeformer<VolumeMesh<T>> deformer_;  
```
I think `deformer_.set_mesh(&mesh_);` is not needed.  I believe the line `mesh_ = std::move(other.mesh_)` only change the content of `mesh_`, but its address doesn't change.  Therefore, we can safely change the above function to:
```C++
  DeformableVolumeMesh& operator=(DeformableVolumeMesh&& other) {
    mesh_ = std::move(other.mesh_);
    return *this;
  }
```

<h1> Test the idea </h1>
I have WIP PR #15169 to test my idea.  This is what I saw on my Puget:

1. It passes `deformable_volume_mesh_test` with the removal of calls to `set_mesh()`.
2. It fails `mesh_deformer_test` by design, and will pass when we remove the definition of `set_mesh()` and its unit test. (Perhaps we can call this "Test-Driven Development".  It will keep failing until we finish development.)

CC: @SeanCurtis-TRI and @soonho-tri for comments, since I'm still not sure that I understand memory management in C++ well.

@SeanCurtis-TRI does this solution play well with your development plan?

SeanCurtis-TRI, 2021-06-10:
This is some good initial analysis. It's very subtle, but I'd suggest it's even a bit more subtle.

I completely agree that `MeshDeformer` has some lying semantics and those lying semantics are bad. But, accidentally, the runtime code doesn't actually *do* anything "bad". Because when we call the mesh copy assignment, we're copying a mesh *onto itself*. One would presume that the default-defined assignment operators have the early out so no actual copying takes place.

So, the code isn't broken in the sense that it does something "bad", but it is broken in the sense that it's lying. Your suggestion of simply removing the `set_mesh` semantics and relying on `DeformableVolumeMesh`'s *use* of the `MeshDeformer` such that, upon construction, it points to the member and never needs to change what it points to is correct. However, in implementing that change, we'd need to provide some clear documentation explaining the mechanism because it will be *incredibly* non-intuitive. It's one of the reasons why we passed the code as it is currently written: reasonable tests passed and the code was of a familiar pattern.

I'd suggest we close #15169 and I'll submit a fix (with the necessary documentation) when I come back from vacation. As is, the code isn't hurting CI and nothing is using it, so there's no fire.

amcastro-tri, 2021-06-10:
Maybe a dumb question, but I use this pattern a lot and I'd like to learn what the good solution is. Why `Mesh& mesh_` instead of `Mesh* mesh_`? I tend to use the latter given that in my mind the `->` operator in the source communicates we are holding a reference.

DamrongGuoy, 2021-06-10:
For @amcastro-tri , `Mesh& mesh_` and `Mesh* mesh_` are slightly different.

The `Mesh& mesh_` will refer to the same `Mesh` object to the end, so MeshDeformer will always deform the same mesh. Once a MeshDeformer is constructed, it is glued to the same `Mesh` object to the end of its lifetime. 

If we use `Mesh* mesh_`, the pointer can point to a different Mesh object later, so the MeshDeformer can deform another mesh if we choose to.

I hope this helps.  I'm not sure that I explain well.

amcastro-tri, 2021-06-10:
That makes sense @DamrongGuoy, thank you.

PR #15221 (2021-07-01): [geometry] Introduce BvhUpdater and add it to DeformableVolumeMesh
The `BvhUpdater` is a "modifier" for a `Bvh`. Given a mesh and its corresponding `Bvh`, it causes the `Bvh` to update given the current position of the mesh's vertices -- it assumes no topological changes.

This initial implementation is simple.
  - It only updates `Bvh`s built on `Aabb`.
  - It only refits bounding volumes -- it doesn't change the topology at all.

This also adds a `Bvh` and `BvhUpdater` to the `DeformableVolumeMesh`. This keeps the deformable mesh in sync.

(Incidentally cleans up some redundant semantics in  [... cut]
closed 2021-07-01 by SeanCurtis-TRI